arendst/Tasmota #5256: What is Generic (0) vs Generic(18)

wongnam, 2019-02-17:
I am confused to use Generic (0) and Generic (18).
and i am confused the template is activated or not if i also chose the Generic (18) and assigned GPIO pin in Module parameters.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ :  NodeMCU
- [x] _Development/Compiler/Upload tools used_ : 6.4.1.17 core 2.4.3, VSC + PlatformIO IDE
- [x] _If a pre-compiled release or development binary was used, which one?_ : Self-complied
- [x] _You have tried latest release or development binaries?_ : Y

netpok, 2019-02-17:
The template is only active when 0 is selected.

wongnam, 2019-02-17:
I just wonder why don't create a module name is Template (0)?

arendst, 2019-02-17:
It just copies the default template name if the NAME field was empty. You are supposed to change it to any value you like as long as it's just 14 characters.

netpok, 2019-02-17:
Maybe that would be a good idea to not copy the name since it should be changed anyways.

arendst, 2019-02-17:
Positive: My,my,my how long have I been thinking about a solution for this.

wongnam, 2019-02-17:
It will be easy to understand if you use the name like "User Define (0)" for default.

arendst, 2019-02-17:
Module 0 is always user defined so it's name could even be Generic when there are no differences between module 18, being the original Generic module.

The logic to change the name to User Defined would be initiated by any user change in Based on. It currently leaves the name alone as soon as there is any name.

For now I leave it that way.

wongnam, 2019-02-17:
@arendst As i can save the template to change the name and used it normally, but i also saw the "Activate" check box option in "Configure Other", even check and uncheck is the same, it does not save.
```
23:51:30 CMD: status  0
23:51:30 MQT: stat/smartplug/STATUS = {"Status":{"Module":0,"FriendlyName":["Smart Plug"],"Topic":"smartplug","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:51:30 MQT: stat/smartplug/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:53","StartupUTC":"2019-02-17T16:50:37","Sleep":50,"CfgHolder":4617,"BootCount":6,"SaveCount":15,"SaveAddress":"F5000"}}
23:51:30 MQT: stat/smartplug/STATUS2 = {"StatusFWR":{"Version":"6.4.1.17(sonoff)","BuildDateTime":"2019-02-17T21:57:23","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
23:51:30 MQT: stat/smartplug/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"192.168.12.155","LogPort":514,"SSId":["iot-2","Wong"],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
23:51:30 MQT: stat/smartplug/STATUS4 = {"StatusMEM":{"ProgramSize":469,"Free":532,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0F082794","240003A0","0000009E","000013C0"]}}
23:51:30 MQT: stat/smartplug/STATUS5 = {"StatusNET":{"Hostname":"smartplug-1737","IPAddress":"192.168.12.111","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"210.245.31.220","Mac":"2C:3A:E8:0F:86:C9","Webserver":2,"WifiConfig":4}}
23:51:30 MQT: stat/smartplug/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.155","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0F86C9","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:51:30 MQT: stat/smartplug/STATUS7 = {"StatusTIM":{"UTC":"Sun Feb 17 16:51:30 2019","Local":"Sun Feb 17 23:51:30 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+07:00","Sunrise":"06:12","Sunset":"18:01"}}
23:51:30 MQT: stat/smartplug/STATUS10 = {"StatusSNS":{"Time":"2019-02-17T23:51:30","AM2301":{"Temperature":30.0,"Humidity":73.8},"TempUnit":"C"}}
23:51:30 MQT: stat/smartplug/STATUS11 = {"StatusSTS":{"Time":"2019-02-17T23:51:30","Uptime":"0T00:00:53","Vcc":3.449,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"iot-2","BSSId":"38:17:C3:F1:90:01","Channel":1,"RSSI":60}}}
```
[image]

localhost61, 2019-02-17:
@arendst 
Regarding the new Template additions, in the Configuration menu wouldn't it be more intuitive to shift the button `Configure Other` in last position, after `Configure Template`?

wongnam, 2019-02-18:
@arendst it's just a proposal only, 
1- Default the WebUI drop-down list is chosen to configuration.
2- Apply a custom template when it was chosen "Activate"
[image]

arendst, 2019-02-18:
@wongnam regarding Configure Other Template functions. I cannot reproduce:

1 - Having selected module Generic as active module opening the Other Configuration and selecting checkbox Activate will restart the device using the template as active module.
2 - Same situation and changing the name of the template, then selecting activate and restart also changes the name of the module as expected.
3 - Same situation clearing the template field and selecting activate does a restart but does not load the template as expected too. Nothing wrong with users wanting to clear the template field if they feel they need to.
4 - Checkbox might be disabled when template is already enabled. That's a feature change I will implement.

The proposed GUI changes have the following drawback(s)

1 - I see configuring a template as a very basic tool which is probably hardly used by new users. It's also very low level and you need to know what you are doing as it directly relates to the module configuration. That's why it is the last menu option.
2 - Loading a community supplied template is an easier task and therefore is available in the Other Configuration menu.
3 - Having both loading template and editing template in one screen confuses users as the bottom part of the screen has no relation with the changes made in the top part end reverse. Making this relation available adds a lot of coding without adding functionality.

So for now it stays this way.

wongnam, 2019-02-18:
> @wongnam regarding Configure Other Template functions. I cannot reproduce:
> 
> 1 - Having selected module Generic as active module opening the Other Configuration and selecting checkbox Activate will restart the device using the template as active module.
- I agree. and tried.
> 2 - Same situation and changing the name of the template, then selecting activate and restart also changes the name of the module as expected.
- I agree. make sense.
> 3 - Same situation clearing the template field and selecting activate does a restart but does not load the template as expected too. Nothing wrong with users wanting to clear the template field if they feel they need to.
- Got it.
> 4 - Checkbox might be disabled when template is already enabled. That's a feature change I will implement.
- I agree. Thanks.
> The proposed GUI changes have the following drawback(s)
>
> 1 - I see configuring a template as a very basic tool which is probably hardly used by new users. It's also very low level and you need to [...]
- Ok
> 2 - Loading a community supplied template is an easier task and therefore is available in the Other Configuration menu.
- It's up to you decide. 
> 3 - Having both loading template and editing template in one screen confuses users as the bottom part of the screen has no relation with the changes made in the top part end reverse. Making this relation available adds a lot of coding without adding functionality.
- No. In fact, I still see that it still has a mutual relationship after I save it and it restarts itself. as is, it's good and it should stays this way.

> So for now it stays this way.

wongnam, 2019-02-18:
> 3 - Having both loading template and editing template in one screen confuses users as the bottom part of the screen has no relation with the changes made in the top part end reverse. Making this relation available adds a lot of coding without adding functionality.

Oh, I understand now if both screens in one page that it  will cause a conflict each other if user have changed the same time with different parameters.
OK, Thank you for your explain @arendst . I agree all you said. Thanks.

closed 2019-02-18 by wongnam