arendst/Tasmota #5257: sonoff.bin from http://thehackbox.org/tasmota/020300/ is not OK

mmatesic01, 2019-02-17:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
sonoff.bin from http://thehackbox.org/tasmota/020300/ is not OK. Firmware upload to sonoff device from webpage or clean FW upload does not "boot" after upload.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

andrethomas, 2019-02-17:
Will check, thanks for notifying

andrethomas, 2019-02-17:
Seems ok - it does the intermediate sonoff-minimal.bin and then to sonoff.bin

```
16:40:16 RSL: stat/sonoff/RESULT = {"OtaUrl":"http://thehackbox.org/tasmota/020300/sonoff.bin"}
16:40:17 RSL: stat/sonoff/RESULT = {"Upgrade":"Version 6.4.1.17 from http://thehackbox.org/tasmota/020300/sonoff.bin"}
16:40:19 HTP: Web server stopped
16:40:19 HTP: Web server active on sonoff-3673 with IP address 192.168.42.200
16:40:21 HTP: Web server stopped
16:40:48 HTP: Web server active on sonoff-3673 with IP address 192.168.42.200
16:40:49 RSL: stat/sonoff/UPGRADE = Successful. Restarting
16:40:50 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
@cp:0
ld
⸮
00:00:00 Project sonoff Sonoff Version 6.4.1.17(f33c592-minimal)-2_3_0
00:00:00 WARNING This version does not support persistent settings
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as sonoff-3673...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on sonoff-3673 with IP address 192.168.42.200
16:41:06 RSL: tele/sonoff/INFO1 = {"Module":"Generic","Version":"6.4.1.17(f33c592-minimal)","FallbackTopic":"cmnd/DVES_0E2E59_fb/","GroupTopic":"sonoffs"}
16:41:06 RSL: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-3673","IPAddress":"192.168.42.200"}
16:41:06 RSL: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
16:41:08 HTP: Web server stopped
16:41:42 HTP: Web server active on sonoff-3673 with IP address 192.168.42.200
16:41:43 RSL: stat/sonoff/UPGRADE = Successful. Restarting
16:41:44 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
@cp:0
ld
⸮
00:00:00 Project sonoff Sonoff Version 6.4.1.17(f33c592-sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as sonoff-3673...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on sonoff-3673 with IP address 192.168.42.200
```

mmatesic01, 2019-02-17:
> Poslano: Sunday, February 17, 2019 4:43:21 PM
> Za: arendst/Sonoff-Tasmota
> Kp: mmatesic01; Author

Hi,
I have uploaded sonoff.bin 6.4.17_2.3.0  (tried with flashing tool and through web page) on Sonoff S20 and Sonof TH device. Sonoff_minimal.bin 6.4.17_2.3.0 uploads OK, but sonoff.bin 6.4.17_2.3.0  just does not make serial connection awaliable. It acts like it is bricked.
S20 after sucessful flash GREEN LED does not emmit.

Sonoff_basic.bin 6.4.17_2.3.0  also worked, but when I changed setting from Generic to Sonoff S2X it also »bricked«.

It is all OK, if I reflash back with 6.4.1 based on 2.3.0.
In this case S20 after sucessful flash GREEN LED does emmit.

Marko Matešič

Jason2866, 2019-02-17:
Cant reproduce. Tried on a S20 with OTA with core 2.3.0, 2.4.2 and 2.5.0
All loaded without a issue and device works as expected.

mmatesic01, 2019-02-17:
Hi,

USing sonoff-classic.bin and sonoff-basic.bin, they are good from an upload point with clean flash and green LED flashes so I can set wifi parameters thought serial.

With Sonof.bin, not. No green LED flashing.

Marko Matešič

andrethomas, 2019-02-17:
Also tested loading 6.4.1 and then doing OTA to latest dev bin but found no problems. Best guess is something just went wrong during the flash rewrite.

mmatesic01, 2019-02-17:
Latest Sonof.bin 4d0ab94 works fine 😊

Marko Matešič

________________________________
Od: Marko Matešić <mmatesic@live.com>
Poslano: Sunday, February 17, 2019 5:04:17 PM
Za: arendst/Sonoff-Tasmota
Zadeva: RE: [arendst/Sonoff-Tasmota] sonoff.bin from http://thehackbox.org/tasmota/020300/ is not OK (#5257)

Hi,

USing sonoff-classic.bin and sonoff-basic.bin, they are good from an upload point with clean flash and green LED flashes so I can set wifi parameters thought serial.

With Sonof.bin, not. No green LED flashing.

Marko Matešič

Jason2866, 2019-02-17:
Have you copied the URL (for example http://thehackbox.org/tasmota/020500/sonoff.bin)
in OTA box? Leave the device alone, it does all needed steps by itself (OTA magic) See picture.
[image]

mmatesic01, 2019-02-17:
I am using http://thehackbox.org/tasmota/020300/ becuase of WiFi satablility. Latest build of sonof.bin with checksum 4d0ab94 is OK.

Marko Matešič

Poslano: Sunday, February 17, 2019 5:10:47 PM

[screenshot_20190217-170902]<https://user-images.githubusercontent.com/24528715/52915641-efdae100-32d6-11e9-8c03-4ab3ea071ab8.png>

andrethomas, 2019-02-17:
> Latest Sonof.bin [4d0ab94](https://github.com/arendst/Sonoff-Tasmota/commit/4d0ab9448d736bb14672f88cd556c75607f03509) works fine

Good stuff - I looked for problems during a recompile but did not see any - obviously this caused an update to the web server hosting the binaries... so I guess we'll never know what went wrong :/

Jason2866, 2019-02-17:
Please close this issue
Thx

andrethomas2, 2019-02-17:
Closing this issue as it seems to be resolved.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-17 by andrethomas2