arendst/Tasmota #5259: Webcommand ; must also be encoded

dirkhe, 2019-02-17:
If I  use the Web command like 
http://3dprinter/cm?cmnd=BacklogDelay%20100;Power1%20off the ; must also be encoded wit %3B.

So I read, that this is not under controll from Tasmota, it would be usefull, if you could change your documentation.

Regards
Dirk

ascillato, 2019-02-17:
Hi,

Which core version are you using? 

Characters supports had changed if you use Arduino core 2.5.0. 

We should add to the wiki if that is required for cores 2.3.0, 2.4.2, 2.5.0

Are you able to test it? 

Remember that you have access to the wiki for editing.

dirkhe, 2019-02-17:
HI,

I didn' know, that I could edit the wiki. 
I checked core Version 2.3 and 2.5. With 2.3 encoding is not nessesary, but it works also. For core 2.5 it is nessesary. I've no 2.4 core available, but I assume, that this will also work with %3B. 
So I changed the ; to %3B and made a comment.

closed 2019-02-17 by dirkhe

ascillato, 2019-02-17:
Thank you very much

LuluBulu, 2019-12-11:
Hello! I'm new to Tasmota. I used to have command with ; and space (instead of %20) and it was working fine for 6.6.0/2.3.0. Now with 7.1.1/2.6.1 space is working but not ;. So I had to change web commands.

ascillato, 2019-12-11:
The new arduino core now handles this according the http standard. Not related to Tasmota code.

LuluBulu, 2019-12-11:
> The new arduino core now handles this according the http standard. Not related to Tasmota code.

Ok, but what about spaces? Don't get me wrong, I really like your work from all the lads that help with every scenario. It's not that much of work for me to change commands, either. But I'm curious as spaces works ok without the need for encoding to %20.

ascillato, 2019-12-11:
In that case is not Tasmota the one translating spaces, is your browser as the http standard