arendst/Tasmota #5263: MQTT Socket Errors

JacoFourie, 2019-02-17:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
MQTT socket errors result in reconnects on Mosquitto. I have 4 sonoffs and they all do this. I also have some other ESP8266 and ESP32 devices that I wrote the code for myself and they stay connected all the time without the socket errors from Mosquitto.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : Sonoff-Basic and Touch
- [ ] _Development/Compiler/Upload tools used_ : OTA
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [X] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
The sonoff should stay connected as the WiFi signal is above 80%. Why does it loose MQTT connection ?

**Screenshots**

[image]

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-02-17:
Hi, please share the status 0 output

ascillato, 2019-02-17:
The issue is a known problem from core 2.4.2 and 2.5.0 with some brands of routers.

If you use core 2.3.0 you won't have that issue.

Anyway, which core and which mqtt library do you use in your own firmwares?

ascillato, 2019-02-17:
Tasmota precompiled bins on core 2.3.0:

http://thehackbox.org/tasmota/020300/

JacoFourie, 2019-02-17:
Hi. I am running 6.4.1. Should I replace 6.4.1 with 2.3.0 ?

ascillato, 2019-02-17:
Please, use latest Tasmota with core 2.3.0: http://thehackbox.org/tasmota/020300/

Which arduino core do you use in your own firmwares that don't disconnects.

JacoFourie, 2019-02-17:
I am using the ESP32 core v 1 not 1.0.1

https://github.com/espressif/arduino-esp32

ascillato, 2019-02-17:
Ok, but ESP32 is another chip and core not related to Tasmota nor Arduino esp8266 core. 

The issue is just with any firmware compiled with Arduino core 2.4.2 or 2.5.0 and just with about the 3% of the brands of routers.

You should test the latest Tasmota precompiled with Arduino core 2.3.0

The issue is outside Tasmota software. Is in the Arduino core. Specifically in the SDK libraries for the esp8266 chip

JacoFourie, 2019-02-17:
How do I know if the basic or touch is 4 or 1 M. What should I use ?

ascillato, 2019-02-17:
Use all the time 1M and you will be ok.

ascillato, 2019-02-17:
Remember to erase all flash with esptool.py as explained in the wiki and flash again with the precompiled bins from development so as to avoid any issue of corrupt data left in the flash.

JacoFourie, 2019-02-17:
So can I not do it over the air ? The devices are in the roof

JacoFourie, 2019-02-17:
Can I use reset 5 after I flash with over the air ?

ascillato, 2019-02-17:
Yes, you can. Erasing all flash is a recommendation to avoid further issues. 

If you can't do the serial flashing, you can do the OTA. But, if after you experiment some weird issues, just do the reset 5. If that don't solve your issue, do a full erase and flash.

JacoFourie, 2019-02-17:
OK and I will give it a go. Thanks. I will close the issue once I tested it.

ascillato, 2019-02-17:
In your first post you said that you have your own firmware in an esp8266 that doesn't disconnects from mqtt. That firmware you have compiled using core 2.3.0? And which mqtt library did you use?

JacoFourie, 2019-02-17:
The one that you configure in the Arduino IDE. 2.3.0 is the one I used I think when I complied the code.
I am not using any new 8266. I am now using ESP32. So I have not updated the code running on the older 8266 chips.

ascillato, 2019-02-17:
Ok. That explain why it worked in your case.
Core 2.3.0 have no issues with mqtt sockets.
Which brand is your wifi router?

JacoFourie, 2019-02-17:
I have 3 different AP in my house. I have Microtik , Ubiquiti and Tenda.
How do I know what core is running on the Sonoff. I can see the version of Tasmota but how do I know the core that was used ?

ascillato, 2019-02-17:
Go to information menu. There it is.

Also you can go to the Tasmota console and type `status 0`

That will give you all the config of Tasmota.

JacoFourie, 2019-02-17:
Aaa OK cool. Must say this software rocks. The amount of stuff you have put into such a small chip is amazing. Well done to all involved !!!

ascillato, 2019-02-17:
Microtik has proven to have zero issues with Tasmota. Ubiquiti has some issues with core 2.4.2. I don't know if with latest core 2.5.0 it has issues. May be not.

But anyway with core 2.3.0 you will be fine.

JacoFourie, 2019-02-17:
OK I flashed 3 of them with the 2.3.0 core. Will let it run. Thanks so much for the help !!

closed 2019-02-17 by JacoFourie

mmatesic01, 2019-02-18:
If I use 2.3.0 version 6.4.1 from http://thehackbox.org/tasmota/release/020300/, then I have no such problems as described in the beginning of this thread.
But if I use 2.3.0 version 6.4.17 from http://thehackbox.org/tasmota/020300/, then I have the exact problems.

All the sonoff devices which were upgraded to 6.4.17 started having such problems. The rest which stayed on 6.4.1, were stable.

JacoFourie, 2019-02-18:
Is 6.4.1 not the latest? Where did you get 6.4.17 from ?

ascillato, 2019-02-18:
@JacoFourie

6.4.1 is the latest master release
6.4.1.17 is the latest development that will become the next master release.

The differences are new features and bug fixing that are being tested before making a new master release. The changelog is available. See readme for the link.

ascillato, 2019-02-18:
@mmatesic01

There are no wifi changes between 6.4.1 core 2.3.0 and 6.4.1.17 core 2.3.0

So, if you are having wifi issues may be it is related to corrupt data in your flash. Try reset 5 and if that don't solve your issue do a full erase with esptool.py as explained in the wiki and flash again with the precompiled bins from development branch. Thanks

JacoFourie, 2019-02-18:
OK. All seems fine with my 3 units after the 2.3.0 version.