arendst/Tasmota #5267: Sonoff basic controlling WS2812b string - basic questions

nst197, 2019-02-18:
I have a Sonoff RF R2 POWER V1.0 which I've flashed with Tasmota v6.4.0. I'm trying to control a string of WS2182b leds and haviing some issues I was hoping for some advice with.

The issue:

When I turn on the PSU to the leds all the leds stay off, I assume this is expected. When I click the second on/off toggle in the Tasmota gui the led string lights up, all leds white. Again I assume this is expected so all good so far.
From this point none of the commands appear to do anything. If I click the on/off toggle again all leds stay on. If i try to change color by typing "Color Red" in the console nothing happens. If I slide the dimmer nothing happens.
Can anyone offer any suggestions as to what I might have done wrong or advice as to how to troubleshoot it further?

The setup:

In Tasmota config I've set IO2 to WS2812 which gives me the extra on/off toggle and brightness slider in the gui.
240V Input to the PSU connected to the output of the sonoff basic so I can control it also. This appears to be working fine. 
5v Output of the PSU to 5v and Gnd on the LED string.
I've run a wire from the IO2 pin on the sonoff which is directly connected to the data wire on the LED string. I've seen reference to using a resistor here but I've haven't currently got one. Is that the issue?
I've also run a wire from the GND pin on the Sonoff to the GND on the led string.

ascillato, 2019-02-18:
Please, complete the troubleshooting template in order to have information so as to properly help you. 

Remember that it is available the Tasmota Support Chat for addressing this type of issues. Github issues are for bug reporting. Thanks.

ascillato2, 2019-02-18:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

toto79, 2019-02-18:
Pls check, the minimum voltage of "Din" is PSU V (5.x V )* 0.7 = 3.5 V. At 3.3V unexpected behavior like yours is to be expected.
A small level shifter will help.

ascillato2, 2019-02-19:
Closing this issue as there is no feedback. When you have time please complete the troubleshooting template. Thanks.

closed 2019-02-19 by ascillato2

shaun-sr, 2019-03-16:
Hello, looking for some help with WS2812 LED strip. I'm having same problem as @nst197 

When I hit the toggle button on Tasmota-MainMenu, LED strip turns White and after that stays like that no matter what I do; I've tried console, slider and toggle button but nothing seems to work, even tried using different pins. 

I am using Tasmota on NodeMCU with WS2812; I've already tested my setup with Adafruit NeoPixel and it works as expected. 

I would really appreciate some help. Thanks!

(I did try the chat option, but got no reply)

nst197, 2019-03-16:
The default method that tasmota uses is bit banging which gave that effect on my ws2812 stripe. I had to compile my own version using the uart method to get it working.

toto79, 2019-03-16:
@shaun-sr : how long is your led strip?

something you could also try:   
    _a level shifter for the data pin (3.3V is not recommended)_
    _another powersupply with max 5V output (not 5.x V)_
    _short wire for data can reduce noise_

shaun-sr, 2019-03-18:
@nst197 Thanks for the quick reply
Can you please point me towards some 'how to' regarding the same; or if you could send me your version of Tasmota, that would be awesome. 
viditrathore@gmail.com

shaun-sr, 2019-03-18:
@toto79 thanks for replying
My led strip is 1m long with 30 individually addressable LED's. 
I did try level shifter, another power supply, another led strip, another esp8266 (D1 mini) and small cable but still the same result. 
I'm sure that it is not a hardware problem as 'Adafruit NeoPixel' library works fine on my setup.

nst197, 2019-03-18:
@shaun-sr 

I tested my setup with the NeoPixelBus test sketch first to find out which methods worked for me.

To edit and compile my own version I used Arduino IDE and followed the guide in the wiki mostly:
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

For Uart method change line 45 of sonoff/xplg_ws2812.ino to read:
    typedef NeoEsp8266Uart800KbpsMethod selectedNeoSpeedType;
Note this method will ignore the pin specification and will use GPIO2. So you'll have to wire it up on that pin. 

You could also try the DMA method which uses GPIO3. To do that uncomment line 396 in sonoff/my_user_config.h which reads '//  #define USE_WS2812_DMA' . This will obviously cause it to ignore any changes above so try these methods separately.

shaun-sr, 2019-03-19:
Thank you so much @nst197 .  I tried the Uart method you suggested and it is working. I'm so happy, I really appreciate you helping me out. 

After turning my setup into permanent, I realised that the on board LED is connected to GPIO2; Now the onboard LED stays on. 
 
I really hope this WS2812 Issue gets fixed soon. 

Once again thanks for the help.

mumin50, 2019-03-22:
Can someone share how to use DMA after uncomment  // #define USE_WS2812_DMA
just step by step ... do i have to put something in web menu?

I managed to run 140 leds with bitbang metod but it causes wifi to freeze any idea why

nst197, 2019-03-22:
After you uncomment that line it just uses DMA for ws2812 devices. Is it not working? Are you using gpio3?

mumin50, 2019-03-22:
I tried multiple metods now. Using core 2.5
I ping it constant. when i turn flip on switch in menu it just freze... Module is connected to WiFi but pings Time out... Some pings go but most time out. I was thinking it is because of metod but now i realise that all do same.
How i do. I uncomented Line, flashed. Set it up. Then chosing gpio3 in web menui select ws2812.  Because without i can not send commands. Then just fliping switch and it freze... It is no problem with power. I use just d1mini without leds. Same isue

xrydm, 2020-02-22:
@nst197 
> You could also try the DMA method which uses GPIO3. To do that uncomment line 396 in sonoff/my_user_config.h which reads '// #define USE_WS2812_DMA' . This will obviously cause it to ignore any changes above so try these methods separately.

I have docker installed on my server so i tried the [tasmocompiler](https://github.com/benzino77/tasmocompiler) and compiled my own binary with WS2812_DMA. 
In Custom parameters i have defined:
`#ifdef USE_WS2812_DMA`
  `  #undef USE_WS2812_DMA`
`#endif`
`#define USE_WS2812_DMA`

I run it on sonoff basic with level shifter between sonoff and 5V adapter. Finaly after two evenings of trial and error it runns smoothly and without flicker. Thank You!