arendst/Tasmota #5269: Timer switches correct on the time set but toggles 5 seconds later.

HaViGit, 2019-02-18:
Using Tasmota with my Sonoff Basic everything seems okay and I can toggle as expected. Setting the timer (time on and time off scertain time) I discover a problem with my setup. After searching the community I did not find a solution, maybe someone can help.

When I program  the timer to switch of at, let’s say, 14:02 it does but 5 seconds later it switches on again:

14:02:00 MQT: stat/Sonoff-200/POWER = OFF
14:02:00 MQT: tele/Sonoff-200/UPTIME = {"Time":"2019-02-18T14:02:00","Uptime":"0T00:02:18"}
14:02:05 MQT: stat/Sonoff-200/RESULT = {"POWER":"ON"}
14:02:05 MQT: stat/Sonoff-200/POWER = ON

No matter if I emulate ‘nothing’ or ‘Wemo' and it makes no difference if I check or uncheck MQTT in the menu. Can someone help me with the right settings or any or suggestion to solve this?

Firmware version: 6.4.1(sonoff)-2_4_2

ascillato, 2019-02-18:
Please, use command weblog 4 for more debug information and try again. With that you will see the SRC (source) of that.

HaViGit, 2019-02-18:
Hi ascillato,

Thanks for fast responding! Short after posting above message I now must confess that I have made a mistake... On the background my MQTT broker with Node-RED was running and it toggles the Sonoff 5 seconds after receiving the command from the timer. So my 'issue' is solved ;-)

Greetings, Hans.

closed 2019-02-18 by HaViGit