arendst/Tasmota #5276: double socket

detafun, 2019-02-19:
Hello,

I have a [double socket](https://images-na.ssl-images-amazon.com/images/I/71jdIYuDPhL._SL1500_.jpg). The two relays are switched via D2 (2) and D6 (1). A blue LED over D1, a green LED over D6. There is still a red LED, this should be switched on and off via the button.

Problem is, if I use the relays and LEDs, the relays switch correctly, but the corresponding LED does not. The only thing that lights up better is the blue LED (D1) for both switches in the web interface, not just the blue (D1) LED for relay D6, but also for the green LED D0.
How can I manage to switch the blue LED (D1) only with relay (D6) and switch the green LED (D0) to relay (D2)?

cu Detafun

PS: Sorry for the first entry in the wrong section. Should not be spam, but a real problem where I can not get any further.

netpok, 2019-02-19:
As far as I know you can't do this. But you can achieve what you want by setting up the leds as relays an using rules to make them follow the physical relays. For example `on power1#state do power3 %value% endon` will make relay3 (the led in this case) follow relay1.

detafun, 2019-02-19:
and how do I have to adjust something? At the moment it is like this.
[image]

netpok, 2019-02-19:
In your case you want to set up D1 as Relay3i and D0 as Relay4i an use the following rules: `on power1#state do power3 %value% endon on power2#state do power4 %value% endon`.

If this solves your problem don't forget to close the issue.

detafun, 2019-02-19:
ok, and where do i have to set the rules? Enter?

netpok, 2019-02-19:
Please check the [appropriate wiki page](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules) for more information next time, but for now you need to enter the following commands to the console:
```
rule1 on power1#state do power3 %value% endon on power2#state do power4 %value% endon
rule1 1
```

Also we do have support chat on Discord for cases like this.

detafun, 2019-02-19:
A clearly again learned something. Are the rules saved? Even after an update or reboot?

So now the LEDs go ON but not off.

netpok, 2019-02-19:
As long as you didn't disable the saving of settings they should be saved. But of course you should just test it by doing a power cycle.

If this solves your problem don't forget to close the issue.

detafun, 2019-02-19:
Switching on with LED ON goes.
Switching off with LED OFF does not work.

netpok, 2019-02-19:
Since this is not a support chat, please close this issue and find me on [discord](https://discordapp.com/invite/Ks2Kzd4)

netpok, 2019-02-19:
Resolved through discord, also fixed the wiki about disabling one-shot detection.

detafun, 2019-02-19:
ok, thanks to netpok, it was still missing the rule1 4.
It works now.

closed 2019-02-19 by detafun