arendst/Tasmota #5277: Ghost-Switching on Shelly 2

ZioTitanok, 2019-02-19:
I've got several Shelly2 flashed with Esptool with Tasmota (6.4.1, core 2_4_2/2.2.1).
I'm experiencing ghost-switching on two Shelly2, both on channel 2. They're placed in the same place, in the electrical cabinet of the floor, as I've done for some other Shelly on a different floor with no issues at all. One of them was before on another floor, and it worked very well for a couple of months.

Both of them are linked with momentary switch and I'm using them as relay (switchmodex 3). I've tried to swap the circuit on channel 1 and 2 but nothing changed: channel 2 keeps ghost switching on both device. Of course I've checked (5 timed) all the connections and the cables. WiFi signal is very good, main router is just 1 meter away.

If I set button/switch debounce on 500 or more, the problem seems to be resolved for a while (a couple of ghost switching for day).

Below a weblog 4, button/switch debounce 50, ghost switching log:

09:16:49 MQT: tele/Shelly2P2/STATE = {"Time":"2019-02-19T09:16:49","Uptime":"1T10:36:35","Vcc":3.461,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Vodafone-Titanok's House","BSSId":"E4:8F:34:46:E8:5C","Channel":8,"RSSI":84}}
09:16:49 MQT: tele/Shelly2P2/SENSOR = {"Time":"2019-02-19T09:16:49","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2019-01-26T14:32:05","Total":1.946,"Yesterday":0.285,"Today":0.134,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}
09:16:49 SER: Ricevuto 06 19 AA 00 00 00 00 09 85 00 00 00 07 00 00 00 47 01 00 00 F1 CB F6 C3 1B
09:16:50 SER: Ricevuto 06 19 A2 00 00 00 01 09 84 00 00 00 09 00 00 00 47 01 00 00 55 CC 28 C4 AD
09:16:51 SRC: Switch
09:16:51 MQT: stat/Shelly2P2/RESULT = {"POWER2":"ON"}
09:16:51 MQT: stat/Shelly2P2/POWER2 = ON
09:16:51 SER: Ricevuto 06 19 0F 07 00 00 FD 08 D4 09 00 00 34 04 00 00 20 10 00 00 FD B1 2C C4 1D
09:16:51 CFG: Salvato nella flash in F6, Conteggio 627, Bytes 3584
09:16:52 SER: Ricevuto 06 19 FE 06 00 00 FD 08 A8 09 00 00 24 04 00 00 DF 0F 00 00 20 B2 38 C4 BD
09:16:53 MQT: tele/Shelly2P2/SENSOR = {"Time":"2019-02-19T09:16:53","ENERGY":{"TotalStartTime":"2019-01-26T14:32:05","Total":1.946,"Yesterday":0.285,"Today":0.134,"Power":25,"ApparentPower":41,"ReactivePower":33,"Factor":0.60,"Frequency":50,"Voltage":230,"Current":0.179}}
09:16:53 SER: Ricevuto 06 19 E6 06 00 00 00 09 80 09 00 00 15 04 00 00 DF 0F 00 00 62 B3 32 C4 AF
09:16:54 SER: Ricevuto 06 19 E6 06 00 00 FF 08 8C 09 00 00 19 04 00 00 DF 0F 00 00 02 B3 02 C4 2D

Below a weblog 2, button/switch debounce 50, ghost switching log:

09:18:19 MQT: stat/Shelly2P2/RESULT = {"WebLog":2}
09:18:37 MQT: stat/Shelly2P2/RESULT = {"POWER2":"OFF"}
09:18:37 MQT: stat/Shelly2P2/POWER2 = OFF
09:18:40 MQT: tele/Shelly2P2/SENSOR = {"Time":"2019-02-19T09:18:40","ENERGY":{"TotalStartTime":"2019-01-26T14:32:05","Total":1.947,"Yesterday":0.285,"Today":0.135,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}
09:20:07 MQT: stat/Shelly2P2/RESULT = {"POWER2":"ON"}
09:20:07 MQT: stat/Shelly2P2/POWER2 = ON
09:20:10 MQT: tele/Shelly2P2/SENSOR = {"Time":"2019-02-19T09:20:10","ENERGY":{"TotalStartTime":"2019-01-26T14:32:05","Total":1.947,"Yesterday":0.285,"Today":0.135,"Power":26,"ApparentPower":43,"ReactivePower":34,"Factor":0.62,"Frequency":50,"Voltage":232,"Current":0.185}}
09:20:32 MQT: stat/Shelly2P2/RESULT = {"POWER2":"OFF"}
09:20:32 MQT: stat/Shelly2P2/POWER2 = OFF
09:20:35 MQT: tele/Shelly2P2/SENSOR = {"Time":"2019-02-19T09:20:35","ENERGY":{"TotalStartTime":"2019-01-26T14:32:05","Total":1.947,"Yesterday":0.285,"Today":0.135,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}
09:20:42 CMD: buttondebounce 500
09:20:42 MQT: stat/Shelly2P2/RESULT = {"ButtonDebounce":500}
09:20:47 CMD: switchdebounce 500
09:20:47 MQT: stat/Shelly2P2/RESULT = {"SwitchDebounce":500}

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Esptool on RarpberryPi
- [x] Hardware used : Shelly2, RaspberryPI
- [x] If a pre-compiled release or development binary was used, which one? : 6.4.1 (2_4_2/2.2.1)
- [x] You have tried latest release or development binaries? : Yes
- [x] Provide the output of command ``status 0`` :

09:35:59 CMD: status 0
09:35:59 SRC: WebConsole from 192.168.1.135
09:35:59 RSL: Topic Ricevuto /status, Dimensione Dati 1, Dati 0
09:35:59 RSL: Gruppo 0, Indice 1, Comando STATUS, Dati 0
09:35:59 MQT: stat/Shelly2P2/STATUS = {"Status":{"Module":47,"FriendlyName":["Shelly2P2","Shelly2Padronale2"],"Topic":"Shelly2P2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,3,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:35:59 MQT: stat/Shelly2P2/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-IT.bin","RestartReason":"Power on","Uptime":"1T10:55:45","StartupUTC":"2019-02-17T21:40:14","Sleep":50,"BootCount":59,"SaveCount":634,"SaveAddress":"F7000"}}
09:35:59 MQT: stat/Shelly2P2/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T15:05:55","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
09:35:59 MQT: stat/Shelly2P2/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Vodafone-Titanok's House",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
09:35:59 MQT: stat/Shelly2P2/STATUS4 = {"StatusMEM":{"ProgramSize":522,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000410","0FDAE794","000383A0","23B617CE","00003BC0"]}}
09:35:59 MQT: stat/Shelly2P2/STATUS5 = {"StatusNET":{"Hostname":"Shelly2P2","IPAddress":"192.168.1.143","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"B4:E6:2D:55:C7:94","Webserver":2,"WifiConfig":4}}
09:35:59 MQT: stat/Shelly2P2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100","MqttPort":1883,"MqttClientMask":"Shelly2P2","MqttClient":"Shelly2P2","MqttUser":"hamqtt","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:35:59 MQT: stat/Shelly2P2/STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 19 08:35:59 2019","Local":"Tue Feb 19 09:35:59 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:51","Sunset":"18:16"}}
09:36:00 MQT: stat/Shelly2P2/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
09:36:00 MQT: stat/Shelly2P2/STATUS10 = {"StatusSNS":{"Time":"2019-02-19T09:36:00","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2019-01-26T14:32:05","Total":1.947,"Yesterday":0.285,"Today":0.135,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}}
09:36:00 MQT: stat/Shelly2P2/STATUS11 = {"StatusSTS":{"Time":"2019-02-19T09:36:00","Uptime":"1T10:55:46","Vcc":3.459,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Vodafone-Titanok's House","BSSId":"E4:8F:34:46:E8:5C","Channel":8,"RSSI":88}}}

netpok, 2019-02-19:
This issue shuld be fixed in the latest development builds, so please try the following ota links:
`http://thehackbox.org/tasmota/sonoff.bin`  for core 2.4 or
`http://thehackbox.org/tasmota/020500/sonoff.bin` for core 2.5 if you also have wifi stability issues

Don't forget to close the issue if this solved your problem

ZioTitanok, 2019-02-19:
> `http://thehackbox.org/tasmota/020500/sonoff.bin` for core 2.5 if you also have wifi stability issues

I'll try this one, I'll let you know.

ascillato2, 2019-02-19:
Hi, 

Sorry, duplicated issue: https://github.com/arendst/Sonoff-Tasmota/issues/4665

Solution is update to latest Tasmota version from development branch as @netpok has explained very well. 

Thanks.

closed 2019-02-19 by ascillato2

HobbytronicsPK, 2019-05-13:
Have a look on to [Ghost / Random Switching on Sonoff Devices (MQTT Retained)](https://blog.hobbytronics.pk/sonoff-tasmota-tips-tricks/#Ghost-Random-Switching)