arendst/Tasmota #5278: Sonoff T1 3 Gang ESP8285 wont flash (From Ebay, seems FAKE)

Evarra, 2019-02-19:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

Evarra, 2019-02-19:
Hello Guy's!

Today I tried to update my ebay SONOFF T1-3 smart switch to TASMOTA.
Unfortunately, when I opened the device, I see that it's not a know version of SONOFF T1.

No Sonoff logo or any of information about sonoff product .. only simmilar thing was eWE-LINK APP description in manual.

Outside:
[image]
[image]
[image]

Inside: (Wires was soldered by me)

[image]
[image]
[image]

So on the backside of PCB are RX,TX,GND,3,3V and SW pad's for firmware upgrade.
On front side has two pad's RESET and Ground.

I trace the PAD to ESP8285 with multimeter and they are connected to ESP8285.
RX - ESP8285 PIN 25, TX- ESP8285 PIN 26 and GIPO0 - ESP8285  PIN 15.

First everything seemed okay, on power up I connect GIPO0 with GND and WIFI light stop blinking, button light are dimmed, indicate that ESP8285 is in programming mode.

When I try to upgrade Tasmota.. I get no success.. RX light blinking but no TX ..
[image]
[image]

I've been upgraded before SONOFF device's with ESP8285 and ESP8266 so my setup must be  okey ..

I try many loading options Arduino, ESP.Easy.Flasher and sonOTA for firmware loading
Change wireing RX-RX TX-TX and RX-TX TX-RX .. but no success!
It seems like the ESP chip is closed somehow or it isn't in progaming mode.

Allso I try all know way's to set device on programming mode 
1. Double reset
2. Hold button 1 and 4 x reset 
3. GIPO0 to GND .. but still same result.

Many hours of research, I couldn't find the same problem on internet with someone else.
That's why I posted it here, maybe someone can still help ?

The device works correctly with eWElink software. Pairig find it secound generation device.
Firmware version 2.7.1

Any questions are welcome,
Thank you!

Evar

gajotnt, 2019-02-19:
I have two of them, the PCB is a little different, got them about 1year ago. Dont remember how i flashed them but they work with tasmota.
When i get home will take some photos and if i have time will try to remenber how i got it flashed.

netpok, 2019-02-19:
The GPIO0 to GND should do the trick and as far as I know you can't write protect an ESP. Also you should check again your connections and that there isn't anything else connected to the RX and TX pins which could interfere with the programming.

The good thing about the ESP being dirt cheap, that they are not worth making fakes (although there were some esp8266 modules with lower flash)

Evarra, 2019-02-19:
Thank you for answering!

All ready 100 times check :D also change USB-TTL interface but still no luck..
I Discover this, when I short GND + RESET device send this to console (74880:8,N,1) : 

ets Jan  8 2013,rst cause:2, boot mode:(3,3)

load 0x40100000, len 2592, room 16 
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8 
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4 
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : DOUT
SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000

rf cal sector: 251
freq trace enable 0
rf[112] :€€€xxžxĆxžųxą€xųxž€xž€x€€xxžxĆxžųxą€xųxž€xž€x€€xxžxĆxžųxą€xųxž€xž€x€€xxžxĆxžųxą€xųxž€xž€x€€xxžxĆxžųxą€xųxž€xž€x€€xxžxĆxžųxą€xųxž€xž€x€€xxžxĆxžųxą€xųxž€xž€x€€xxžxĆxžųxą€xųxž€xž€x€€xxžxĆxžųxą€xųxž€xž€x€€xxžxĆxžųxą€xųxž€xž€x

Maybe this is somhow helpful.

Evar

netpok, 2019-02-19:
What happens when you do a reset with IO0 grounded? Does this device has RF functionality?

Evarra, 2019-02-19:
Sorry No, I think ther is no chip for RF.. the CHIP spot is empty.. only RF antenna on PCB
[image]

> What happens when you do a reset with IO0 grounded? Does this device has RF functionality?
> 
> Also dont forget to use code blocks `` ``` `` for logs.

Evarra, 2019-02-19:
Uhh .. I'm new here.. I somehow close this thread.... :)

I try do reset when IO0 was grounded ... only button lights was dimmed a little...

When I try to ground GIPO0 after Powe on, button no one will turn red as it was press..

So I think Button 1 and GIPO0 is somehow connected

closed 2019-02-19 by Evarra

netpok, 2019-02-19:
try to flash it with GPIO0 connected to GND through the whole time of flashing

Evarra, 2019-02-19:
I thik there is something wrong with my ESP, It will go into programming mode put no success on flashing.
All the wires are correct and check many times .. TTL RX goes ESP TX and TTL TX goes ESP RX. As see the boot message come's from ESP side.. but flashing message won't go through. Maybe ther is some issiue with UART TX port.. how knows ...

netpok, 2019-02-19:
The problem is that that message should not appear with gpio0 pulled to gnd.

lalo-uy, 2019-02-19:
Are you wiring GPIO0 to ground?

Evarra, 2019-02-19:
> The problem is that that message should not appear with gpio0 pulled to gnd.

Yes you are right, No message on RESET when GIPO0 is Grounded.

netpok, 2019-02-19:
You should try reading the current firmware or run chip_id with esptool

Evarra, 2019-02-19:
> You should try reading the current firmware or run chip_id with esptool

No succes ...

> .\IoT\Python\ESPTool>esptool.py
 --port COM8 chip_id
esptool.py v2.7-dev
Serial port COM8
Connecting........_____....._____....._____....._____....._____....._____....._____
A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

Tried both modes programming(GIPO0 Grounded all the way) and normal power up mode. Result was same .. RX blinking on my FDTI USB TTL and thats it.. no more action..

netpok, 2019-02-19:
I'm officially out of ideas... sorry...

Evarra, 2019-02-19:
> I'm officially out of ideas... sorry...

No problem, I think it's just broken .. Thank you very much!

Evarra, 2019-02-20:
So I get this issue resolved... 1000x check I found that ESP8285 PIN 15. GIPO0 was connected with ESP8285 PIN 14.. bad soldering from manufacture side.

I disconnect PIN 14 and PIN 15 .. now everything works :)

Thank you everbody!

netpok, 2019-02-20:
Please make an article about this at the [wiki page of the product](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK,-EU,-and-US), preferably with one or two images for future reference for other users.

Thanks for your help.

Evarra, 2019-02-20:
> Please make an article about this at the [wiki page of the product](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK,-EU,-and-US), preferably with one or two images for future reference for other users.
> 
> Thanks for your help.

No problem, I add some lines and picture of new PCB.