arendst/Tasmota #5281: Dynamic Sleep confusion

oskarn97, 2019-02-19:
Hello!

I am not sure, if the Dynamic Sleep settings are affecting the MQTT Command response time.
If I want the Sonoff Basic to react as quickly as possible to MQTT Commands and do not care about Power Consumption, should I disable sleep completely? Which would be the optimal settings?

Thanks

ascillato, 2019-02-19:
The optimal settings are the defaults. But if you don't want power consumption reduction you can use the command sleep 0 in the console to disable it.

The response time for mqtt is not noticiable between sleep 0 and sleep 50 (50 miliseconds max) of dynamic sleep.

The dynamic sleep will fluctuate the idle time of free cpu cycles depending on usage. If the device is idle, it will use less cpu, so less power. If the device is receiving mqtt messages, press buttons, etc etc, the time the cpu is free will be less so, the actual sleep will be very little using dynamic.

ascillato, 2019-02-19:
For further information there is a specific chapter about this in the wiki.

Also, there is the Tasmota Support Chat.

closed 2019-02-19 by oskarn97