arendst/Tasmota #5283: Sonoff Pow R2 does not get time from ntpserver

Gwenselah, 2019-02-19:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-02-19:
Hi,

Sorry, can't reproduce your issue. My pows are working fine. 

Please, check your network. This issue had happened before and most of the time is just a misconfiguration on your router or firewall. Just once was that ntp servers were down for a couple of hours.

jziolkowski, 2019-02-20:
@Gwenselah make sure your DHCP server sends correct NTP option as well. Tasmota is based on Arduino core, which takes this info and makes it primary (which causes problem with DHCP servers sending bad or empty options for NTP)

Gwenselah, 2019-02-20:
Hello,
thank you for your replies.
At the moment i have 4 Sonof Pow R2 in the same wlan. All have the 6.4.1 firmware. 2 of them get the correct time, the other 2 not.
Now I upgraded one to version 6.4.1.17 with core 2.5.0. this didn't solved the issue.
Are there some debugging commands to trace the getting time process?

Kind regards.

jziolkowski, 2019-02-21:
Come and join us at our discord. There were users with this problem. Might provide more help.
https://discord.gg/Crj7tCJ

ascillato2, 2019-02-22:
Hi,

Any news on this issue?

Gwenselah, 2019-02-23:
No, I setup a ntp server on my raspberry pi, but nothing changed. I increased the debug log, but there is nothing in it, which would help.
I will go over to discord now

Gwenselah, 2019-02-23:
Now I think i got it.
If you use DHCP on the sonoff and your DHCP server sends no or incorrect ntp option, then you can not use "ntpserver1 <IP>" to set an internal ntp server. You can set it, but it will not work. You have to set a satic ip on the sonoff addiotionally.
Version  6.4.1.17

ascillato, 2019-02-23:
Very interesting! Thanks for sharing this.
That explains why I couldn’t reproduce the issue.

Thanks, now we have a workaround for people with this NTP issue for the DHCP in their router until something can be done in the code.

andrethomas, 2019-02-25:
Its a SDK issue - if NTP server exists (configured correctly or not) on the same IP as the DHCP server (no setting for NTP server necessary on DHCP settings) then the SDK automatically uses the NTP service it assumes to be good on the same IP of the DHCP server... or at least this has been my observation.

arendst, 2019-02-25:
So as it is SDK related you will have to use a workaround by adding one or two additional NTP servers using command ``ntpserver2`` and ``ntpserver3``.

jziolkowski, 2019-02-26:
@arendst it doesn't change anything. We've tried it with some discord users some time ago. The solution is to use DHCP server with proper NTP option configured in the lease.

@Gwenselah you've discovered what I basically told you in my first response.

Gwenselah, 2019-02-26:
@jziolkowski 
Please consider that many people only have dhpc "server" running on routers like Fritzbox and yes, may be I missunderstood you.
For those people my workaround it is easier then setup an run a real DHCP Server.
So, please stay polite.

ascillato2, 2019-02-28:
Closing this issue as it is has been answered. https://github.com/arendst/Sonoff-Tasmota/issues/5283#issuecomment-465630518

closed 2019-02-28 by ascillato2