arendst/Tasmota #5284: Tuya dimmer - Trying to set dimming level without turning on light

ericdes, 2019-02-20:
Hi, first I'd like to thank you for this excellent work that has improved a lot over time!

I understand there is a memory feature in the Tuya dimmer since the light turns backs on at its previous brightness level.

My experience so far was to emit an MQTT brightness command to a light being off, **which results in turning it on** at the specified brightness level. 

I just would like to change the value of the brightness memory in the Tuya dimmer while not changing the power state -- i.e. keep the light off.

Jason2866, 2019-02-20:
Tried?
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#SetOption20

ericdes, 2019-02-21:
Indeed setting option 20 did the trick for me.

Thank you!

 

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#SetOption20

ascillato2, 2019-02-21:
Closing issue as it has been solved.

closed 2019-02-21 by ascillato2