arendst/Tasmota #5285: Cant enter via Phone with hostname to UI

hwarang098tkd, 2019-02-20:
So i have alot of switches sonoff Touch
I CANT enter via my phone with hostname but only with IP adr
From my desktop i can.

```
11:30:19 CMD: hostname
11:30:19 MQT: stat/sonoff_t_9/RESULT = {"Hostname":"livingroom_relay"}
11:35:39 CMD: status 5
11:35:39 MQT: stat/sonoff_t_9/STATUS5 = {"StatusNET":{"Hostname":"livingroom_relay","IPAddress":"192.168.1.2","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"60:01:94:84:CE:82","Webserver":2,"WifiConfig":5}}
```

In my phones wifi settings

[image]
 
i tried http://livingroom_relay and http://livingroom_relay.local

What i have to do?
thnx in advance

netpok, 2019-02-20:
To use it via local DNS resolution, you have to use your router as DNS server, based on your status message it is `192.168.1.254`:
`...255.0","DNSServer":"192.168.1.254","Mac":"60:0...`

To use it via mDNS resolution, you have to enable mDNS on the device by executing `SetOption55 1` in the console as documented on the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#Setoption55).

hwarang098tkd, 2019-02-20:
it is enabled.still cant by my phone.
[image]

netpok, 2019-02-20:
You should try a device restart first, and use the .local address.

curzon01, 2019-02-20:
Your Phone is using the internet public Google DNS 8.8.8.8 and this public DNS does not know anything about the names of your local network. Use 192.168.1.254 as DNS also for your Phone.

netpok, 2019-02-20:
Yes, but mDNS should work for him with those settings, as it has nothing to do with the DNS server.

ascillato2, 2019-02-20:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-20 by ascillato2

hwarang098tkd, 2019-02-21:
> Wireless Client Isolation is _not_ enabled on your router.

cant find it to mu router Technicolor TG589vn v3

> If it's not working you can try to run `avahi-browse --all` on your computer or it's windows/mac counterpart to check if it shows up or not

with Bonjour Browser i can see the hostnames "_http._tcp.      livingroom_relay, IP Addresses 192.168.1.3:80"

> Use 192.168.1.254 as DNS also for your Phone.

did it and still nothing and also with .local

netpok, 2019-02-21:
Really weird, I don't have any more tips.

On the other hand you should really disable WPS, because it's a [big security hole](https://www.howtogeek.com/176124/wi-fi-protected-setup-wps-is-insecure-heres-why-you-should-disable-it/).

hwarang098tkd, 2019-02-21:
From laptop with wifi i can access, i tried another phone and the problem is the same. Iphone and android. So i cant via phone with wifi

netpok, 2019-02-21:
You can try to check it with a bonjour browser on the phone like [this](https://play.google.com/store/apps/details?id=com.druk.servicebrowser&hl=en_US) (never used it myself), but probably the best way to manage your devices is via OpenHAB or Home Assistant, those have nice mobile apps too.

hwarang098tkd, 2019-02-21:
OK with this app i can see all this devices with their hostnames and Ips on my phone. It is  something
I have openhab 2 years now and everything runs from there. But sometimes i want to configure details via ui (tasmota). So i wanted it to have access with hostname from phone in my house not only desktop

netpok, 2019-02-21:
If you see them then the problem lies somewhere else, my last guess: Is there a chance that you are using adblocker on the devices like AdGuard? Those are blocking the requests.