arendst/Tasmota #5286: tasmota reset without button

elzershark, 2019-02-20:
tasmota reset without button.
a led Bulb (Tuya) have not a Button.
can the software reset without botton? by bulbs resett is 5x 
without electricity by tuya.
can you install that?

netpok, 2019-02-20:
I don't really understand your message sorry, but:
wrong wifi ssid and/or password: if you know the ssid or password you entered just set that temporarily in router
wrong admin password: if you have mqtt configured you can reset it via that

Also since tasmota uses GET request for forms you should check your history, maybe it's there look for entries with name of `Save Configuration`.

for example the following link
`http://sp10/co?t1=%7B%22NAME%22%3A%22Generic%22%2C%22GPIO%22%3A%5B23%2C22%2C24%2C17%2C134%2C132%2C0%2C0%2C131%2C52%2C21%2C0%2C0%5D%2C%22FLAG%22%3A0%2C%22BASE%22%3A67%7D&p1=SecretPassword&b1=on&a1=Sonoff&a2=Sonoff2&a3=Sonoff3&a4=Sonoff4&b2=0&save=`
contains the password (`SecretPassword` in this case) for web interface in the `p1` parameter.

Any other cases you are out of luck and you probably have to open the case (or find a bug in the code, but then please report it).

elzershark, 2019-02-20:
no, you can not send a mqtt command, when you not have the password, ssid, wlan ore what ever 😊
For lamps, there should be a possibility that reset over "de-energized". for example Switch 5x (de-energized)

ascillato, 2019-02-20:
Hi

At this moment, Tasmota has a bootloop protection routine that if Tasmota starts and resets (or loose power) in less than 2 seconds, and this happens several times in a row, Tasmota will start changing settings to default, one by one. I think that 10 cycles in a row put mostly settings to default. But wifi and mqtt settings will remain.

May be, can be added an extra configuration key at compiling time to make Tasmota full reset if it is performed 11 cycles of less than 2 seconds of power ON for each cycle.

What do you think?

netpok, 2019-02-20:
Actually no, there really shouldn't be an option for this as it is a huge security hole, if you want you can (but really shouldn't) set `WifiMode` to `2` then if you start the device and it can't connect to wifi it starts up in AP mode, but that doesn't solve the wrong web password.

You're better off writing down your passwords, of course only in a secure way like using [pass](https://www.passwordstore.org/).

If you still want it against all my explanation, all I can do for you is implement `WifiMode 6` as WifiManager without password since it would take up probably less then 10byte of code space and there are enough already reserved eeprom for it.

netpok, 2019-02-20:
@ascillato I haven't checked that code, but will later today as it should do a full reset on all eeprom pages because of security reasons.

Reset count is stored in the RTC eeprom as far as I know and that's not retained after a power loss, also most devices cannot be resetted with that frequency, because of power supply capacitors.

ascillato, 2019-02-20:
That was just an idea to be reviewed. Just simple as 11 power on/off of the bulb energy in less than 2 seconds each ON (as some original firmware do for these type of devices) and just perform a reset 1 command after that.

netpok, 2019-02-20:
After checking the code it is possible in a secure way without exposing the config.

Also this wasn't meant against your idea I wrote the first comment without seeing yours.

ascillato, 2019-02-20:
>Also this wasn't meant against your idea I wrote the first comment without seeing yours.

Hehehe, don't worry about that. An idea polished by several people is always a better idea.

andrethomas, 2019-02-20:
I have reset devices this way... sometimes intentionally and other times unintentionally :)

netpok, 2019-02-20:
I tried it with the SP10 smart socket I couldn't get it to remember the boot count after power loss, also there is no option to reset the device this way since the last case is:
```
      if (RtcReboot.fast_reboot_count > Settings.param[P_BOOT_LOOP_OFFSET] +4) {  // Restarted 6 times
        Settings.module = SONOFF_BASIC;             // Reset module to Sonoff Basic
  //      Settings.last_module = SONOFF_BASIC;
      }
```

@andrethomas is it possible that you were talking about the buttonpress reset method?

andrethomas, 2019-02-20:
Yes, the button method.

The fast reboot counter will not work as it requires the rtc memory to be active (powered) so power cycling a device will not cause a revert to factory defaults... it can work if you have a reset pin connected... i.e. being able to reboot the device without it having lost power.

netpok, 2019-02-20:
Okay, just the original problem was that the device doesn't have buttons :)

andrethomas, 2019-02-20:
Yes, but I'm not totally sure whether this is a question to a current problem or a feature request...

DavinKD, 2019-02-20:
Just want to throw my .02 in.  It would be nice to be able to reset bulbs with no buttons.  The ones I have were able to do so when flashed with the Tuya stuff, so there should be a way to do it.  This is not a high priority for me, though, as all my bulbs are working fine.

netpok, 2019-02-20:
My best idea currently is to implement a WifiConfig 2 like interface, but only with a big reset button

andrethomas, 2019-02-20:
Will need to think about how this can be implemented in flash without wearing the flash out too much...

netpok, 2019-02-20:
That's why I suggested the new WifiConfig as it doesn't need flash writes opposed to boot counting.

andrethomas, 2019-02-20:
Yes, correct wificonfig will solve it - not sure how the tuya flash thing works yet and how that initial binary is configured.

elzershark, 2019-02-21:
The reset without a button should be feasible. There are too many password problems at the moment. There have already been some lights "broken" because z.b. in the password "* #" was present and thus had no access to the light bulb.

netpok, 2019-02-21:
If `*` was present in the password field, it did not get saved in firmwares before 6.4.17 (it is fixed now), so it should wait for an open network. Please try to set your AP into Open authentication mode and try to connect to the device.

elzershark, 2019-02-21:
fact that * causes the lamp to become unusable. It is automatically generated in some browsers at the webpage setting where mam can create a password. If somebody presses "save" without first deleting the password, that was the light source

netpok, 2019-02-21:
Did you try to recover it with an open network as I mentioned above?

elzershark, 2019-02-21:
wenn man ein passwort aus versehen vergibt bei der webseiten passwort erstellung., hat man kein zugriff mehr auf die lampe. weder per mqtt noch sonstwie. ich rede nicht vom wlan passwort :)

netpok, 2019-02-21:
In older firmwares it was handled with:
`if(p1 == "********") p1="";`
So it looks for an open network like this:
[image]

If you set that up on your router you will be able to recover it, also we are looking into possible solutions to implement device reset as you requested.

elzershark, 2019-02-21:
https://www.youtube.com/watch?v=cNdgpF3N8Ao

elzershark, 2019-02-21:
is synonymous only 1 * in the password, it was that. tasmota can not recognize * in the password

netpok, 2019-02-21:
With web password I don't know what you can do, but if the wifi password was bad you should try the open network because it checked with `strchr(tmp,'*')`, so it was a contains check instead of an equality check.

elzershark, 2019-02-21:
ich hoffe mein problem hat man verstanden. Die lampe hat ein passwort bekommen, was nicht bekannt ist, für die web- oberfäche. Die lampe ist im wlan. aber man kann nicht über mqtt oder sonstwie zugriff zu ihr bekommen. man kann auch nicht über mqtt befehl die lampe resetten. weils das resetten nur über ein button funktioniert, ist diese lampe unbrauchbar geworden. das nur zum verständnis. deshalb muss es für lampen eine möglichkeit geben sie zu resetten ohne ein button.
😁

andrethomas, 2019-02-21:
If the device is reachable via mqtt you can send a `reset 1` via mqtt to the device.

elzershark, 2019-02-21:
The device kasnn is not known, it is not known. Once a web password is present, the lamp is unusable.

elzershark, 2019-02-21:
ich gebe es auf. man versteht es nicht. Wenn ein web-passwort existiert.. egal ob es bekannt ist oder nicht, hat man kein zugriff über mqtt auf die lampe. Wenn das web-passwort nicht bekannt ist, kann man mit der lampe nie wieder etwas steuern. ausser man macht die lampe kaputt und flasht sie neu über die hardware. unbekanntest web- passwort = kein mqtt = kein sonstwas. es ist dann eine tote lampe

netpok, 2019-02-21:
Yes, if you haven't set up the mqtt, the only options are flashing via serial connection, or looking for the password in your history. I made an implamentation for what you want, but that will be only available in future versions (if it is accepted). So you're probably out of luck wit those lights.

andrethomas, 2019-02-21:
Have you tried to get it to glitch by powering on and off in 1-second intervals?

elzershark, 2019-02-21:
das web-passwort wurde nicht "vergessen". es wird IMMER von tasmota selber erstellt. wie im video zu sehen ist. wer aus versehen, einfach nur auf "speichern" klickt, hat damit seine lampe zerstört. es wird dadurch das passwort ********.. Also 8x der Strn, Sternchen wird gespeichert. Diese Sternchen kann aber Tasmota nicht verarbeiten. Wenn man als passowrt danach 8x *`eingibt, wird es nicht anerkannt. weil Tasmota keine * verarbeiten kann. es wird nicht als passwort anerkannt. Also.. wer auf "speichern" klickt, ohne vorher das Feld "Passwort" zu löschen, zerstört seine Lampe

andrethomas, 2019-02-21:
If you did not specify a password (i.e. leaving it as ********) then remove the password from your router so that it is an open network without a password - then the device should connect.

netpok, 2019-02-21:
His problem is the web admin password, but at least for the last 4 months it is handled with the following line:
`strlcpy(Settings.web_password, (!strlen(tmp)) ? "" : (strchr(tmp,'*')) ? Settings.web_password : tmp, sizeof(Settings.web_password));`
If the password contains a `*`, it keeps the old password.

elzershark, 2019-02-21:
> If you did not specify a password (i.e. leaving it as ********) then remove the password from your router so that it is an open network without a password - then the device should connect.

Kein wlan.. wiese denken alle über wlan?.. ich schreib WEB.. WEB ist kein Wlan.. Web ist WEB... WEB- Passwort... not wifi.. WEB - Passwort...

netpok, 2019-02-21:
If you have setup an mqtt connection just send it in with a client, like
`mosquitto_pub -h <broker> -t cmnd/<device-topic>/reset -m 1`

elzershark, 2019-02-21:
> If you have setup an mqtt connection just send it in with a client, like
> `mosquitto_pub -h <broker> -t cmnd/<device-topic>/reset -m 1`

NO

andrethomas, 2019-02-21:
which version of Tasmota is on the device?

elzershark, 2019-02-21:
Das hat nichts mit der version zu tun... egal.. ich gebe es auf..

andrethomas, 2019-02-21:
> Das hat nichts mit der version zu tun... egal.. ich gebe es auf..

Does it not occur to you that I ask this question intentionally?

netpok, 2019-02-21:
A specific version could have a bug which you can use to access your device.

elzershark, 2019-02-21:
i have all versionen.. the new version an 7 before. in all versionen is "standard "*********"

andrethomas, 2019-02-21:
Ok, I'm out :)

elzershark, 2019-02-21:
und ein anderes problem. aber das muss ich erst noch richtig testen. 1 lampe kann man flashen. da funktioniert alles so wie es sein soll. in iobroker, Sonoff Adapter. Aber sobald amn mehrere lampen geflasht hat, fangen alle nach einer gewissen zeit an zu blinken, weil sie sich gegenseitig die IP Adressen austauschen. aber das wird noch dauern. daher kann ich nur jeden rate, wer lampen in iobroker oder mit mqtt bedienen möchte, das erst mal zu lassen.

netpok, 2019-02-21:
If you have a star in the password field it will not be saved no matter if its 1 or 11.

Also I don't get your last message, IP addresses shouldn't matter, as each device should have a different topic.

elzershark, 2019-02-21:
> If you have a star in the password field it will not be saved no matter if its 1 or 11.
> 
> Also I don't get your last message, IP addresses shouldn't matter, as each device should have a different topic.

Doch, wird gespeichert. im Flash sind die 8 sterne vorhanden..

netpok, 2019-02-21:
No it doesn't you will always get 8*-s as its masking out the password.

Also I'm following andrethomas too...

elzershark, 2019-02-21:
https://www.youtube.com/watch?v=cNdgpF3N8Ao

Wenn ich da speichern drücken würde, würde ich das gerät kaputt machen

netpok, 2019-02-21:
Yes and that won't be saved because the code above prevents it, feel free to check out the code.

elzershark, 2019-02-21:
hier Probier dein Glück. Passwort ist ********
mqtt = 
Sonoff 
pw=
casio
http://badlampe.​r1cwmfe0usjdikdc.​myfritz.​net

meingraham, 2019-02-21:
Ich denke, @netpok bedeutet "check out the code" nicht darin zu suchen, was der browser rendert, um das eigentliche passwort zu maskieren, sondern vielmehr, was das programmmodul im GitHub-Code-Repository zum Speichern und Verwenden des tatsächlichen macht Kennwort übertragen, aber nicht übertragen oder auf einer Remote-Schnittstelle anzeigen, was die Sicherheit beeinträchtigen könnte, die Sie zu gewährleisten versuchen.

I was hoping that replying (as provided by Google Translate) in Deutsche, a language that is not used by those providing support on this forum but rather using the your native language which you insist on using will help convey to you what I mean. After all, a dialog using languages which we don't mutually share will surely enhance communication manifestly; yes?

netpok, 2019-02-22:
@elzershark This just got merged into development, you can test it. You
just need to activate it with `WifiConfig 7` in the console.

elzershark, 2019-02-22:
👍

ascillato2, 2019-02-22:
Closing this issue as the feature has been added.

Thanks everyone for sharing theirs ideas. :+1:

closed 2019-02-22 by ascillato2

RyanEwen, 2019-06-30:
Hey guys I have some Globe RGB bulbs which I flashed with 6.4.1 (and one with 6.5.0) that I am needing to reset after updating the hostnames caused them to drop off the network and never come back.

The firmware had my SSID and MQTT settings hardcoded, so resetting them should get them back online.

However I can't figure out how to reset them. Stock f/w is 6 power cycles quickly. I see in here the idea of doing it 10 or 11 times. The light is just flashing white/green/white/green indefinitely.

Is this reset feature in 6.4.1 or 6.5.0 and how exactly does it work? Didn't see it in the wiki anywhere.

meingraham, 2019-06-30:
Booting and then booting again within 10 seconds each time for six consecutive cycles should successively reset settings on the device. This reset scheme was implemented in the 6.2.1 timeframe.

RyanEwen, 2019-06-30:
I tried that but it wasn't working for me.

However I noticed on a working device that when I was changing hostnames, my Wifi SSID was being changed. My SSID has an apostrophe in it. It was being cutoff from that point on.

So I setup my phone as a hotspot with the truncated name and all my messed up bulbs connected to it and I was able to reconfigure them :)

meingraham, 2019-06-30:
Ah... special characters - https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#ssid

**`... do not use special characters or white spaces in the SSID...`**

[13 more replies]