arendst/Tasmota #5287: wall switch1 control relay2

megamarco833, 2019-02-20:
hi, my configuration is generic module 18 (nodemcu)
i'm using last version of tasmota
i create the device configuration as:
GPIO2 button1
gpio4 button2
gpio5 button3

gpio12 relayi1
gpio13 relayi2
gpio14 relayi3

now with button1 i toggle relayi1
with button2 i toggle relayi2
with button3 i toggle relayi3

what i would like to implement is that using button1 i toggle relayi2 and no more the relayi3
the others button2 and 3 still maintain their behaviours

second step is:
with longpress of button1 toggle relay1
with shortpress of button1 toggle relay2

how i can perform that?
i see the rules, and try, but without success...can some one help me?
thanks

ascillato, 2019-02-20:
Hi,

Yes, it is possible. Please, check the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#16-using-an-external-button-with-single-press---double-press-and-hold

If you need further assistant please find us in the Tasmota Support Chat. Github issues are for bug reporting with the template completed. Thanks

ascillato2, 2019-02-20:
Closing this issue as it is duplicated. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-20 by ascillato2