arendst/Tasmota #5290: Add new Drivers and Sensors

nosaj66au, 2019-02-21:
Hi All, 
  Just got back to my little RF bind interrogation project.
 
**Have you look for this feature in other issues and in the wiki?**
Yes
**Is your feature request related to a problem? Please describe.**
I would like to contribute by adding a driver or 2 but have limited time to pull apart the code to insure the correct integration.

**Describe the solution you'd like**
A document / tutorial  on how to add sensors and drivers and/or details around the system architecture.
explaining the module interconnection.

netpok, 2019-02-21:
Hi

The sensors are pretty much self contained, you should check out the other sensors code, look for the `xsns_*` files. Aside from this you should check out the whole codebase for coding style and general overview of code, as this probably cannot be replaced by any existing or non-existing documentation.

ascillato2, 2019-02-21:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-21 by ascillato2

nosaj66au, 2019-02-22:
Thanks Guy's. However I dis-agree. Please don't take me the wrong way, I love this piece of software and uses it entirely for my home automation. I'm a professional software engineer for Industrial automation and IOT and if you are going to use the codebase as your document reference you need a hell of a lot more embedded comments. At the moment we don't even have a 1 page dataflow diagram. If you want people to contribute the project you need to give then the information. 
It now seems like this project is more of a product then a Open source project.  I was very disappoint that is topic was "Closed" so quickly. @netpok great job mate this is what open source is all about. Its a shame nobody else will see my post to add there comments. Can we please leave RFI posts open for other to participate this is the third time one of my posts has been shutdown.

ascillato, 2019-02-22:
Hi,

Sorry, do not take it personal. Your issue was duplicated. These questions has been addressed before.

Please, check issues: https://github.com/arendst/Sonoff-Tasmota/issues/2138, https://github.com/arendst/Sonoff-Tasmota/issues/1828

And please, check the wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-API

If you need further information, please chat with us in the Tasmota Support Chat.