arendst/Tasmota #5292: Blitzwolf SHP2 LED continuously flashes red and blue

TalZiv, 2019-02-21:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : esptool
- [X] Hardware used : CP2102-MICRO-USB-to-UART-TTL
- [X] If a pre-compiled release or development binary was used, which one? : classic
- [X ] You have tried latest release or development binaries? : from the UI updated to 6.4.1
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**
Hi,
I got a weird problem with SHP2, when connected just to serial and not 220v, it connects to Wifi and I can get to the device web UI I also updated to 6.4.1.

when connected to 220v the LED continuously flashes red and blue and does not connect to Wifi.

I would like to do keep the serial connectors on when connecting the device to 220v but I am afraid :)

what may cause this issue?

netpok, 2019-02-21:
You should try doing a `reset 1` or `reset 3` in the console and reconfiguring it from scratch.

ascillato2, 2019-02-22:
Hi, closing this issue as it has been answered and there is no feedback. When you have time please tell us if this works for you. If not, please ask here or in the Tasmota Support Chat. Thanks

closed 2019-02-22 by ascillato2