arendst/Tasmota #5293: Dev Firmware keeps bootlooping/restarting after entering MQTT Server IP

cyablo, 2019-02-21:
**Describe the bug**
Every Dev-Firmware > 6.4.1.15(de5023f) provided by http://thehackbox.org/tasmota/archive/ keeps on bootlooping/restarting every ~20 Seconds.

Im using a bunch of Sonoff S2X and OBI Sockets (Version 1). Every Dev-Firmware above 6.4.1.15(de5023f) keeps on restarting the devices every ~20 Seconds. I already tried a factory reset and configuration from the start.

**The Sockets work fine until I enter the IP Address of my MQTT Server**

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff S2X & OBI Sockets V1
- [ ] _Development/Compiler/Upload tools used_ :
- [x] _If a pre-compiled release or development binary was used, which one?_ : > 6.4.1.15(de5023f)
- [x] _You have tried latest release or development binaries?_ : Yes, also bootloops
- [x] _Provide the output of command_``status 0`` :
```
14:22:14 CMD: status 0
14:22:14 MQT: stat/tasmota14/STATUS = {"Status":{"Module":8,"FriendlyName":["tasmota14"],"Topic":"tasmota14","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:22:14 MQT: stat/tasmota14/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software Watchdog","Uptime":"0T00:00:15","StartupUTC":"2019-02-21T13:21:59","Sleep":50,"CfgHolder":4617,"BootCount":40,"SaveCount":52,"SaveAddress":"F8000"}}
14:22:14 MQT: stat/tasmota14/STATUS2 = {"StatusFWR":{"Version":"6.4.1.17(280dd25-sonoff)","BuildDateTime":"2019-02-19T16:01:47","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
14:22:14 MQT: stat/tasmota14/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["cybnet-2-4",""],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
14:22:14 MQT: stat/tasmota14/STATUS4 = {"StatusMEM":{"ProgramSize":534,"Free":468,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","0FDAE794","000783A0","23B617CE","00003BC0"]}}
14:22:14 MQT: stat/tasmota14/STATUS5 = {"StatusNET":{"Hostname":"tasmota14","IPAddress":"192.168.1.93","Gateway":"192.168.1.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.18","Mac":"80:7D:3A:6A:DB:5E","Webserver":2,"WifiConfig":4}}
14:22:14 MQT: stat/tasmota14/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.26","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6ADB5E","MqttUser":"DVES_USER","MqttType":1,"MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:22:14 MQT: stat/tasmota14/STATUS7 = {"StatusTIM":{"UTC":"Thu Feb 21 13:22:14 2019","Local":"Thu Feb 21 14:22:14 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:47","Sunset":"18:19"}}
14:22:14 MQT: stat/tasmota14/STATUS10 = {"StatusSNS":{"Time":"2019-02-21T14:22:14"}}
14:22:14 MQT: stat/tasmota14/STATUS11 = {"StatusSTS":{"Time":"2019-02-21T14:22:14","Uptime":"0T00:00:15","Vcc":3.475,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"cybnet-2-4","BSSId":"64:70:02:67:56:19","Channel":6,"RSSI":62,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

**To Reproduce**
Flash Dev-Firmware > 6.4.1.15(de5023f)

andrethomas, 2019-02-21:
Please try the latest compile (6.4.1.16) - There were some changes made that impact on mqtt.

Please advise if it solves your issue. Thanks.

cyablo, 2019-02-21:
Updated one OBI and one S2X, working fine so far. Seems to be fixed with latest build.

ascillato, 2019-02-21:
Great to know that now it is working fine for you.

Please, close this issue. Thanks

closed 2019-02-21 by ascillato2