arendst/Tasmota #5294: SONOFF T2 EU boot mode

VaL080, 2019-02-21:
Hello All,

I've got a problem to enter my SONOFF T2 EU in boot mode.
When I press BTN 1, press reset , release reset and release BTN1 => Nothing appends.

Below, this is what i have when i turn on the SONOFF T1 EU. 
Have you any idea about how i can go in boot mode?

[image]

Thanks for your help!

ascillato, 2019-02-21:
Hi,

Sorry, the output is not from Tasmota firmware. 

If you need assistant on how to flash a device, please find us in the Tasmota Support Chat. Github issues are for Tasmota software bug reporting. 

Thanks

ascillato2, 2019-02-21:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-21 by ascillato2

xury77, 2019-09-08:
Hi.
Does Tasmota work on versions T2 and T3?
Nowhere is it officially written.

ascillato, 2019-09-08:
Yes