arendst/Tasmota #5298: Wemos D1 Mini & DS18b20

FraatRamski, 2019-02-21:
I'm trying to use DS18b20 on my Wemos D1 Mini with fresh tasmota checkout on 21.02.2019

#define USE_DS18x20 is uncommented 
gerätetyp: 18
red on VCC
black on GND
yellow on D5 GPIO14 Sensor ans also on tested on D2 GPIO4  4k7 pulup to vcc

on the main page i have no temperature info just 
Generic Modul
Sonoff
Einstellungen

there is no line with "Temperature: "

Status 0

21:32:37 CMD: status 0
21:32:37 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:32:37 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:05:53","StartupUTC":"2019-02-21T20:26:44","Sleep":50,"CfgHolder":4617,"BootCount":28,"SaveCount":62,"SaveAddress":"F6000"}}
21:32:37 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.18(sonoff)","BuildDateTime":"2019.02.21 21:06:16","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
21:32:37 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
21:32:37 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":535,"Free":468,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000407","0FDAE794","240783A0","23B617CE","00003BC0"]}}
21:32:37 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3445","IPAddress":"xx.xx.xx.95","Gateway":"10.12.34.1","Subnetmask":"255.255.255.0","DNSServer":"10.12.14.1","Mac":"3C:71:BF:2C:AD:75","Webserver":2,"WifiConfig":4}}
21:32:37 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2CAD75","MqttUser":"DVES_USER","MqttType":1,"MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:32:37 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Feb 21 20:32:37 2019","Local":"Thu Feb 21 21:32:37 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:47","Sunset":"18:19"}}
21:32:37 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-02-21T21:32:37"}}
21:32:37 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-02-21T21:32:37","Uptime":"0T00:05:53","Vcc":2.945,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"xxxxxx","BSSId":"7xxxx31","Channel":11,"RSSI":68,"LinkCount":2,"Downtime":"0T00:00:09"}}}

netpok, 2019-02-21:
Please try out with `sonoff-sensors.bin`.

Also use code block (` ``` `) for easier readability.

FraatRamski, 2019-02-21:
with `sonoff-sensors.bin` and also when I build with env_default = sonoff-sensors I can see the Temperature now but the 1st line is always "Analog0 4". 

Any idea how I can remove this "Analog0 4" line?

Analog0	4
DS18B20-1 Temperature	23.6°C
DS18B20-2 Temperature	23.4°C

23:31:40 CMD: Status 0
23:31:40 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:31:41 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:07:07","StartupUTC":"2019-02-21T22:24:33","Sleep":50,"CfgHolder":1,"BootCount":11,"SaveCount":20,"SaveAddress":"F8000"}}
23:31:41 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.18(sensors)","BuildDateTime":"2019-02-21T23:14:23","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
23:31:41 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["PayPerSer",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
23:31:41 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":567,"Free":436,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FDEE794","2407A3A5","B7FFBFCC","001ABBC0"]}}
23:31:41 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3445","IPAddress":"10.222","Gateway":"122","Subnetmask":"255.255.255.0","DNSServer":"222","Mac":"3C:71222:AD:75","Webserver":2,"WifiConfig":4}}
23:31:41 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"1zzzz2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2CAD75","MqttUser":"MqttUsr","MqttType":1,"MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:31:41 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Feb 21 22:31:41 2019","Local":"Thu Feb 21 23:31:41 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:47","Sunset":"18:19"}}
23:31:41 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-02-21T23:31:41","ANALOG":{"A0":4},"DS18B20-1":{"Id":"021892457D36","Temperature":24.1},"DS18B20-2":{"Id":"021892459EA7","Temperature":23.4},"TempUnit":"C"}}
23:31:41 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-02-21T23:31:41","Uptime":"0T00:07:08","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"gg","BSSId":"7gg31","Channel":11,"RSSI":44,"LinkCount":1,"Downtime":"0T00:00:39"}}}

netpok, 2019-02-21:
Analog 0 is the ADC pin I think, it is generally used for measuring VCC. This issue needs some looking because as far as I know the sensor should be supported. I'm thinking it is related to some kind of a library update, because I have this problem since I cleaned some cache folder if I'm remembering it correctly.

andrethomas, 2019-02-21:
Analog input is enabled by default in sonoff-sensors.bin - My recommendation is just to ignore the mqtt data of that since it does not serve any purpose for you from what I can deduce.

The alternative is to compile your own binary with only the options enabled that you need.

localhost61, 2019-02-22:
To overcome this, isn't it possible to use the template and uncheck ADC0?

andrethomas, 2019-02-22:
The ADC is not included in the template from what I know because it's a special function pin... it's not a GPIO as such.

It may be possible to make this happen in future versions but since @FraatRamski is already self-compiling the logical solution is to just compile for sonoff (the standard one) and enable support for DS18b20 in the my_user_config.h

netpok, 2019-02-22:
It could work, it is a device flag.

@andrethomas This is a bug since it is one of the most basic temperature sensor and it is already enabled for the `sonoff.bin`.

localhost61, 2019-02-22:
Yes it's an option:
[image]

andrethomas, 2019-02-22:
Ah yes, forgot about the device flags after 2AM this morning :)

It should indeed be enabled by default for sonoff.bin - Unfortunately I own any DS18b20's so I cannot confirm.

FraatRamski, 2019-02-22:
Now is all running, thank all for support

closed 2019-02-22 by FraatRamski