arendst/Tasmota #5300: Armtronix is not working with version 6.4.x

jsponz, 2019-02-21:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
After following the instructions, I was able to install Tasmota 6.3.0.6 (several months ago) on an Armtronix single dimmer. 

Yesterday, I updated the Armtronix to latest releas and it does not dim the lights.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : Armtronix single dimmer
- [ ] _Development/Compiler/Upload tools used_ :
- [X] _If a pre-compiled release or development binary was used, which one?_ : 6.4.0 and 6.3.0.6
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_
Update to version 6.4.x and try to dim a light. 

**Expected behavior**
With version 6.3.0.6 it dims and it works as expected. 

**Screenshots**
N/A

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

andrethomas, 2019-02-21:
Please consider trying the latest development binaries from http://thehackbox.org/tasmota/ to see if it rules out your issue and report back. Thanks.

ascillato2, 2019-02-22:
Hi, any news on this? 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

ascillato2, 2019-02-23:
Closing this issue as there is no feedback. Please, when you have time tell us if that have worked for you. Thanks

closed 2019-02-23 by ascillato2