arendst/Tasmota #5301: v6.4.1.18 don't boot on core v2.5.0

localhost61, 2019-02-21:
Did someone test latest commit v6.4.1.18 with core 2.5.0?
It fails by me.
___
- After OTA upgrade of two Sonoff Dual-R2, with this firmware, *both devices don't boot*, no Led blink.
- So I only changed core to 2.4.2 in platformio.ini  
- Then, I erased the flash with esptool.py (because I use a sensor and want now to restore serial logging)
- And ran "Upload and Monitor" from VScode and it runs with no issue 
```
Building .pioenvs\Dual-R2\firmware.bin
Checking size .pioenvs\Dual-R2\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [======    ]  57.1% (used 46788 bytes from 81920 bytes)
PROGRAM: [=====     ]  46.2% (used 473268 bytes from 1023984 bytes)
Configuring upload protocol...
Looking for upload port...
Use manually specified: COM12
Uploading .pioenvs\Dual-R2\firmware.bin
Uploading 477408 bytes from .pioenvs\Dual-R2\firmware.bin to flash at 0x00000000
................................................................................ [ 17% ]
. . .
...................................................................              [ 100% ]
=============== [SUCCESS] Took 84.45 seconds
--- Miniterm on COM12  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
00:00:00 Projet Spare Sonoff Version 6.4.1.18(sonoff)-2_4_2
00:00:00 WIF: Connexion à l'AP1 MyAP en mode 11N comme Spare-0490...
00:00:04 WIF: Connecté
00:00:04 HTP: Serveur web actif sur Spare-0490 avec l'adresse IP 192.168.61.137
22:00:49 RSL: INFO1 = {"Module":"Sonoff Basic","Version":"6.4.1.18(sonoff)","FallbackTopic":"cmnd/DVES_9E81EA_fb/","GroupTopic":"sonoffs"}
22:00:49 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"Spare-0490","IPAddress":"192.168.61.137"}
22:00:49 RSL: INFO3 = {"RestartReason":"Power on"}
```
- Then, I power off the device, put it in download mode, restore core 2.5.0 in platformio.ini,  and run again "Upload and Monitor"and ...
```
Building .pioenvs\Dual-R2\firmware.bin
Checking size .pioenvs\Dual-R2\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=====     ]  54.5% (used 44644 bytes from 81920 bytes)
PROGRAM: [=====     ]  48.4% (used 496064 bytes from 1023984 bytes)
Configuring upload protocol...
Looking for upload port...
Use manually specified: COM12
Uploading .pioenvs\Dual-R2\firmware.bin
Uploading 500208 bytes from .pioenvs\Dual-R2\firmware.bin to flash at 0x00000000
................................................................................ [ 16% ]
. . .
.........                                                         [ 100% ]
=============== [SUCCESS] Took 92.45 seconds 
--- Miniterm on COM12  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:4, boot mode:(1,5)

wdt reset
```
As you can see, I use a moded version of my_user_config.h that helps me to stay easily under 512kB but I make sure that it's content stay synced with current version.

andrethomas, 2019-02-21:
Hi, can't reproduce using the pre-compiled dev bins

```
00:00:00 Project sonoff Sonoff Version 6.4.1.18(c97ea4d-sonoff)-2_5_0
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as sonoff-3673...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on sonoff-3673 with IP address 192.168.42.200
23:05:18 CMD: otaurl
23:05:18 RSL: stat/sonoff/RESULT = {"OtaUrl":"http://thehackbox.org/tasmota/020500/sonoff-4m.bin"}
23:05:24 CMD: status 0
23:05:24 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:05:24 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020500/sonoff-4m.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:21","StartupUTC":"2019-02-21T22:05:03","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":9,"SaveAddress":"3FA000"}}
23:05:24 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.18(c97ea4d-sonoff)","BuildDateTime":"2019-02-21T20:09:24","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
23:05:24 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Wireless",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
23:05:24 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":3512,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","0FDAE794","000783A0","23B617CE","00003BC0"]}}
23:05:24 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3673","IPAddress":"192.168.42.200","Gateway":"192.168.32.1","Subnetmask":"255.255.240.0","DNSServer":"192.168.32.1","Mac":"DE:AD:00:00:BE:EF","Webserver":2,"WifiConfig":4}}
23:05:24 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0E2E5959","MqttUser":"DVES_USER","MqttType":1,"MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:05:24 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Feb 21 22:05:24 2019","Local":"Thu Feb 21 23:05:24 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:47","Sunset":"18:19"}}
23:05:24 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-02-21T23:05:24"}}
23:05:24 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-02-21T23:05:24","Uptime":"0T00:00:21","Vcc":2.966,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless","BSSId":"00:00:00:00:00:00","Channel":3,"RSSI":94,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```

localhost61, 2019-02-21:
- Same behavior with `xdrv_01_webserver.ino` from 2018.02.18_12h19

andrethomas, 2019-02-21:
The dev bins are based on the last merge as at the time of writing (c97ea4d)

[image]

localhost61, 2019-02-21:
@andrethomas 
- On which core?
- I know it compiles, but did you test with core 2.5.0 on device?

localhost61, 2019-02-21:
Oops, didn't see previous message. Sorry ;-)

localhost61, 2019-02-21:
Thank you for your help! Your sonoff-basic firmware works properly.
So I close the issue and go ahead with my investigation to isolate what I did wrong.

andrethomas, 2019-02-21:
No sweat - could be core related on platformio so maybe I should mention @Jason2866 so he can just take a peek to rule that out.

localhost61, 2019-02-21:
Yep :-(
I'm removing step by step all differences I could find.
- git stash           // -> so that everything is even with last build
- Just change the minimum in platformio.ini: USB port and Core version
- Uninstall core 2.5.0 in VScode to force the refresh.
- Keep only arendst files in my workspace
- Restart VScode
- Select [sonoff-basic]/[Upload and Monitor] in VScode
=> Always the same message of the Soft Watchdog that triggers, and no blue LED blink

localhost61, 2019-02-22:
I confirm that the problem is not with core 2.5.0 but with the version pointed by platformio.ini, it works with previous one from Jason2866 repository:

```
; *** Esp8266 core for Arduino version 2.5.0
;platform                  = espressif8266@2.0.0
platform                  = https://github.com/Jason2866/platform-espressif8266.git#Tasmota 
```

arendst, 2019-02-22:
I noticed the same yesterday; the official platformio release 2.0.0 did watchdog reboots on load. Never seen any serial message.

Keep using @Jason2866 version for the time being until platfformio manages to fix their issue.

localhost61, 2019-02-22:
OK
Maybe a comparison of both resulting firmware.map could help???

localhost61, 2019-02-22:
Looking at firmware.map, at first glance it appears that the linker is different.
- On Release 2.0.0 addresses are defined on 64bit when it was 32bit for the beta.
- In data segment, (starting at 0x3ffe84dc) the filling gaps are different between variables, 2.0.0 is a bit more efficient and earn 8 bytes in that game.
In fact there are a lot of differences and I'm not skilled enough to interpret them :-(

Jason2866, 2019-02-22:
I can confirm. The platformio team has something messed up in there release version 2.5.0
But i do have problems with 6.4.1.18 with some wemos minis. It seems 6.4.1.18 draws more current
as the versions before and the weak  3.3 Vreg cant handle this. Same symptoms as before dynamic sleep

andrethomas, 2019-02-22:
Will do some power requirement testing a little later today to see what the differences are between 2.3.0, 2.4.2 and 2.5.0 (The officially released one) and then also between 6.4.1 and the current dev bin (6.4.1.18) and report back

arendst, 2019-02-22:
Also keep in mind the core team is about to release 2.5.1 using the same SDK as core 2.4.2 (=2.2.1) to solve issues with SDK 3.0.0 on core 2.5.0 (Ah love these numbers!!). This may also alleviate the problems currently experiencing with 2.5.0

Jason2866, 2019-02-22:
Core 2.4.2 and Core 2.5.0 with SDK 2.2.1 does hang wemos devices too. It is not core related!

andrethomas, 2019-02-22:
> Ah love these numbers!!

Maybe I should wait to test core 3.0.0 using SDK 2.5.0 ;)

Jason2866, 2019-02-22:
Version numbers like this (homebrew 2.5.1)
[image]
For everyone who wants to try:
`platform                  = https://github.com/Jason2866/platform-espressif8266.git#Stage-Tasmota`

localhost61, 2019-02-22:
@Jason2866
 Your update of platformio.ini works for me with Stage core 2.5.1 and commit [3ce36de](https://github.com/arendst/Sonoff-Tasmota/commit/3ce36defe287a0ceea0a1d0251d841fcc869dca1)
___
[image]
___
[image]

localhost61, 2019-02-22:
So you found something? Because when I tested Stage core yesterday it didn't work. I guess it was `fe8c185`

Jason2866, 2019-02-22:
I am on https://github.com/esp8266/Arduino/commit/044568c220ae3a0ad6f151611ba834f9b3e1402e
with
`platform = https://github.com/Jason2866/platform-espressif8266.git#Stage-Tasmota`

All other didnt worked...

localhost61, 2019-02-22:
I don't understand the relation between esp8266/Arduino and the platform.

Jason2866, 2019-02-22:
There is no relation. I adjusted platformio just with `Stage-Tasmota` to use Arduino @044568c

localhost61, 2019-02-23:
Thanks @Jason2866 both platforms are working now:
- https://github.com/Jason2866/platform-espressif8266.git#Tasmota
- https://github.com/Jason2866/platform-espressif8266.git#Stage-Tasmota

I close the issue

closed 2019-02-23 by localhost61

Jason2866, 2019-02-23:
The error is in platformio.py  You find a working version here: https://github.com/platformio/platform-espressif8266/issues/128#issuecomment-466655071