arendst/Tasmota #5302: TLS&Webserver

baha00, 2019-02-21:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [1 ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ 1] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [1 ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [1 ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

Hi there, its not really and issue but as TLS and Webserver cannot be used at same time, is it possible to create rule that webservers stops once mqtt connection is established (when WIFI connectcion is setup trough webserver)

andrethomas, 2019-02-21:
At the moment this is not feasible due to memory constraints.

As a workaround, you may flash a normal sonoff.bin to the device and complete the configuration.

Once the configuration step has been completed you can OTA upgrade to your desired self compiled binary by storing the binary on a local web server (raspberry pi other other simple web server) and sending a mqtt commands as follows

`cmnd/topic/otaurl` with payload `http://x.x.x.x/mybinary.bin` to set the OTA upgrade URL and then send a second command `cmnd/topic/upgrade` with a payload of `1`

This will cause the device to initiate an OTA upgrade from the url provided without the need for the web ui interface.

baha00, 2019-02-21:
hm, interesting, so Firmware 1 that I already have(disabled all unnecessary sensors)
[image] when connected start OTA with firmware 2. with webserver disabled and TLS enabled...  and in case of errors ,network change etc  hold button to reload firmware 1 with webserver.... ?

andrethomas, 2019-02-21:
If you need to get a web ui back you will need to revert to a firmware that has web ui support using the same method outlined above.

If you just need to send settings you can do everything you need via mqtt just like when using the command console.

Also check out a new development on the horizon by @jziolkowski at https://github.com/jziolkowski/tdm which allows management of Sonoff-Tasmota devices over mqtt instead of http so in future you will be able to perform most of the functions that are in the web ui through the mentioned utility.

baha00, 2019-02-22:
I need webui only for connecting to network as I didn't open device to flash it but did it trough OTA (mockTuya cloud). In this case if I change wifi pass, or network I need to re-flash device by opening it. First I wanted to change my_user_config and enable webserver overthere than disable it with user_config override.. but this also is not possible

andrethomas, 2019-02-22:
I would not recommend running webserver and mqtt tls at the same time due to the reasons I mentioned earlier.

If your mqtt server is on your local network then there is not much reason to use TLS.

So its either the first mentioned option mentioned earlier or running without TLS over your key protected wifi to a local mqtt server.

baha00, 2019-02-22:
I want take risk and enable tls and webserver, leaves me in theory 13k of ram, but this command Webserver0 is not shutting down webserver 
[image]

baha00, 2019-02-22:
[image]

baha00, 2019-02-22:
well, when you are stupid nothing works... webserver 0 works but webserver0 no...
Ill test this with webserver off and tls on and let you know resuts

closed 2019-02-22 by baha00