arendst/Tasmota #5303: SM16716 usage appears to cause runaway condition resulting in thermal trip for certain bulbs

maikthulhu, 2019-02-22:
**Describe the bug**
Some Tuya-compatible RGBW bulbs (ex. Merkury Innovations MI-BW904) appear to overheat and cause thermal trip (bulb resets and continues to do so every 3-10 seconds) only when SM16716 pins are configured.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : MI-BW904
- [x] _Development/Compiler/Upload tools used_ : thehackbox dev releases/also commits compiled locally using pio/Rpi3 with tuya-convert
- [x] _If a pre-compiled release or development binary was used, which one?_ : pre-compiled dev builds 280dd25-sonoff and c97ea4d-sonoff, locally compiled 2c51b8d sonoff
- [x] _You have tried latest release or development binaries?_ : Yes
- [x] _Provide the output of command_``status 0`` : 
```
01:34:01 CMD: status 0
01:34:01 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Boys' lamp"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
01:34:01 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:30","StartupUTC":"2019-02-22T00:33:31","Sleep":50,"CfgHolder":4617,"BootCount":17,"SaveCount":47,"SaveAddress":"F6000"}}
01:34:01 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.18(c97ea4d-sonoff)","BuildDateTime":"2019-02-21T20:01:36","Boot":4,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
01:34:01 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["dalek",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
01:34:01 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":535,"Free":468,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","0FDAE794","000783A0","23B617CE","00003BC0"]}}
01:34:01 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3214","IPAddress":"10.10.0.126","Gateway":"10.10.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.10.0.5","Mac":"84:F3:EB:2E:6C:8E","Webserver":2,"WifiConfig":4}}
01:34:01 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2E6C8E","MqttUser":"DVES_USER","MqttType":1,"MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
01:34:01 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri Feb 22 00:34:01 2019","Local":"Fri Feb 22 01:34:01 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:46","Sunset":"18:21"}}
01:34:01 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-02-22T01:34:01"}}
01:34:01 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-02-22T01:34:01","Uptime":"0T00:00:30","Vcc":0.729,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":35,"POWER":"ON","Dimmer":100,"Color":"FFFFFFFF","HSBColor":"0,0,100","Channel":[100,100,100,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"dalek","BSSId":"0C:9D:92:55:EC:70","Channel":7,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:09"}}}
```

**To Reproduce**
1. Flash the bulb using tuya-convert, upgrade to 6.4.1 binary shipped with tuya-convert as per instructions.
2. Configure GPIO settings as follows.
```
GPIO0: None (0)
GPIO1: None (0)
GPIO2: None (0)
GPIO3: None (0)
GPIO4: SM16716 CLK (140) # Color LEDs
GPIO5: PWM1 (37) # White LED array/brightness
GPIO9: None (0)
GPIO10: None (0)
GPIO12: None (0)
GPIO13: SM16716 PWR (142) # Color LEDs
GPIO14: SM16716 DAT (141) # Color LEDs
GPIO15: None (0)
GPIO16: None (0)
```
3. Set warm white brightness to 100% (PWM1 as configured).
4. Wait 20-30 minutes.
5. Monitor LoadAvg, bulb temp (just feel).
6. Bulb should enter a loop of rebooting.

**Expected behavior**
Bulb to operate normally, not get as hot as it appears to, and not reset after 20-30 minutes.

**Screenshots**
N/A

**Additional context**
I have 5 of these bulbs.  One is flashed with 2c51b8d-sonoff downloaded at the time from thehackbox dev repo.  One is flashed with a locally-compiled checkout of 2c51b8d-sonoff.  Those two have never experienced this overheating/reboot problem (though I've just set the latter up to test to be sure).  Two others continued to experience this problem even with the locally-compiled 2c51b8d-sonoff.  I noticed at the time the LoadAvg was crazy high (200+ for one, 800+ for another).

The 5th one I'm currently testing with is running the latest dev compiled from thehackbox (c97ea4d).  Initially I only configured PWM1 on GPIO5 (all others set to None (0)) and set brightness to 100% and it ran for hours with no issues.  As soon as I configured the SM16716 pins as above, the bulb began rebooting after the 20-30 minute mark.  It appears the bulbs will run forever until SM16716 pins are configured.  Affected bulbs get noticeably hotter to the touch when fully configured which leads me to think it's a thermal trip, especially since the reboots continue every few seconds until the bulbs are powered off long enough to sufficiently cool them.  LoadAvg never appears to get above 100 (it was around 60 last I saw before one rebooted).

I don't know why the one 2c51b8d-sonoff flashed bulb from thehackbox dev repo appears to be bulletproof (could be hardware differences). After some discussion in discord it was revealed that sleep for scheme0 was disabled some time after 2c51b8d.  That's what prompted me to compile 280dd25 locally (the first revision I noticed the problem with) and set `PWM_LIGHTSCHEME0_IGNORE_SLEEP` to re-enable sleep.  This appeared to be successful (LoadAvg settled to 19 on all flashed bulbs) but the reboot loops still occur after 20-30 minutes.

I can't imagine my builds differ from the one that was up on thehackbox (I followed the pio tutorial) but I'm by no means an expert here.

netpok, 2019-02-22:
Does that happens when you only turn on the SM16716 with PWM1 off? Or is the RGB led off during these tests.

maikthulhu, 2019-02-22:
I've only tried PWM1 on, RGB off so far.

maikthulhu, 2019-02-22:
Below is the `status 0` of what I'll call `light 2` (one of the two "good" ones).  I set it to 100% soft white a couple hours ago and it's running at what I'd call a good temp and isn't rebooting.  The name only shows `sonoff` but it's the local build of 2c51b8d.

```
04:45:35 MQT: light2/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["light2"],"Topic":"light2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
04:45:35 MQT: light2/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T08:35:13","StartupUTC":"2019-02-21T19:10:22","Sleep":50,"CfgHolder":4617,"BootCount":21,"SaveCount":112,"SaveAddress":"F4000"}}
04:45:35 MQT: light2/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1.17(sonoff)","BuildDateTime":"2019-02-20T21:20:03","Boot":4,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
04:45:35 MQT: light2/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["dalek",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","280500000100000000000000000000000000","00000200"]}}
04:45:35 MQT: light2/stat/STATUS4 = {"StatusMEM":{"ProgramSize":531,"Free":472,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","0FDAE794","240783A0","23B617CE","00003BC0"]}}
04:45:35 MQT: light2/stat/STATUS5 = {"StatusNET":{"Hostname":"light2-2648","IPAddress":"10.10.0.107","Gateway":"10.10.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.10.0.5","Mac":"84:F3:EB:74:AA:58","Webserver":2,"WifiConfig":4}}
04:45:35 MQT: light2/stat/STATUS6 = {"StatusMQT":{"MqttHost":"hass.home.halfzero.net","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_74AA58","MqttUser":"hass","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
04:45:35 MQT: light2/stat/STATUS7 = {"StatusTIM":{"UTC":"Fri Feb 22 03:45:35 2019","Local":"Fri Feb 22 04:45:35 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:46","Sunset":"18:21"}}
04:45:35 MQT: light2/stat/STATUS10 = {"StatusSNS":{"Time":"2019-02-22T04:45:35"}}
04:45:35 MQT: light2/stat/STATUS11 = {"StatusSTS":{"Time":"2019-02-22T04:45:35","Uptime":"0T08:35:13","Vcc":2.450,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Dimmer":100,"Color":"0,0,0,255","HSBColor":"0,0,0","Channel":[0,0,0,100],"Scheme":0,"Fade":"ON","Speed":2,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"dalek","BSSId":"0C:9D:92:55:EC:70","Channel":7,"RSSI":100}}}
04:45:37 MQT: light2/tele/STATE = {"Time":"2019-02-22T04:45:37","Uptime":"0T08:35:15","Vcc":2.449,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Dimmer":100,"Color":"0,0,0,255","HSBColor":"0,0,0","Channel":[0,0,0,100],"Scheme":0,"Fade":"ON","Speed":2,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"dalek","BSSId":"0C:9D:92:55:EC:70","Channel":7,"RSSI":88}}
```

maikthulhu, 2019-02-22:
On `light 5` (running c97ea4d-sonoff from thehackbox dev repo) I set `color FFFFFF00` (so RGB is on and PWM1 is off).  It's been running for ~2.5 hours with no issues.  Console output is below.

```
00:00:00 Project sonoff lamp5 Version 6.4.1.18(c97ea4d-sonoff)-2_4_2
00:00:00 WIF: Connecting to AP1 dalek in mode 11N as sonoff-3214...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff-3214 with IP address 10.10.0.126
05:02:21 RSL: tele/sonoff/UPTIME = {"Time":"2019-02-22T05:02:21","Uptime":"0T00:00:06"}
05:05:15 CMD: state
05:05:15 RSL: stat/sonoff/RESULT = {"Time":"2019-02-22T05:05:15","Uptime":"0T00:03:00","Vcc":0.988,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":37,"POWER":"ON","Dimmer":100,"Color":"FFFFFFFF","HSBColor":"0,0,100","Channel":[100,100,100,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"dalek","BSSId":"0C:9D:92:55:EC:70","Channel":7,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:05"}}
05:05:27 CMD: color FFFFFF00
05:05:27 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FFFFFF00","HSBColor":"0,0,100","Channel":[100,100,100,0]}
05:05:53 CMD: state
05:05:53 RSL: stat/sonoff/RESULT = {"Time":"2019-02-22T05:05:53","Uptime":"0T00:03:38","Vcc":1.061,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":47,"POWER":"ON","Dimmer":100,"Color":"FFFFFF00","HSBColor":"0,0,100","Channel":[100,100,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"dalek","BSSId":"0C:9D:92:55:EC:70","Channel":7,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:05"}}
05:07:14 RSL: tele/sonoff/STATE = {"Time":"2019-02-22T05:07:14","Uptime":"0T00:04:59","Vcc":1.096,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":51,"POWER":"ON","Dimmer":100,"Color":"FFFFFF00","HSBColor":"0,0,100","Channel":[100,100,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"dalek","BSSId":"0C:9D:92:55:EC:70","Channel":7,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:05"}}
05:12:14 RSL: tele/sonoff/STATE = {"Time":"2019-02-22T05:12:14","Uptime":"0T00:09:59","Vcc":1.069,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":54,"POWER":"ON","Dimmer":100,"Color":"FFFFFF00","HSBColor":"0,0,100","Channel":[100,100,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"dalek","BSSId":"0C:9D:92:55:EC:70","Channel":7,"RSSI":86,"LinkCount":1,"Downtime":"0T00:00:05"}}
05:12:15 APP: Serial logging disabled
05:17:14 RSL: tele/sonoff/STATE = {"Time":"2019-02-22T05:17:14","Uptime":"0T00:14:59","Vcc":1.089,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":57,"POWER":"ON","Dimmer":100,"Color":"FFFFFF00","HSBColor":"0,0,100","Channel":[100,100,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"dalek","BSSId":"0C:9D:92:55:EC:70","Channel":7,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:05"}}
05:22:14 RSL: tele/sonoff/STATE = {"Time":"2019-02-22T05:22:14","Uptime":"0T00:19:59","Vcc":1.100,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":57,"POWER":"ON","Dimmer":100,"Color":"FFFFFF00","HSBColor":"0,0,100","Channel":[100,100,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"dalek","BSSId":"0C:9D:92:55:EC:70","Channel":7,"RSSI":86,"LinkCount":1,"Downtime":"0T00:00:05"}}
05:27:14 RSL: tele/sonoff/STATE = {"Time":"2019-02-22T05:27:14","Uptime":"0T00:24:59","Vcc":1.101,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":57,"POWER":"ON","Dimmer":100,"Color":"FFFFFF00","HSBColor":"0,0,100","Channel":[100,100,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"dalek","BSSId":"0C:9D:92:55:EC:70","Channel":7,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:05"}}
05:32:14 RSL: tele/sonoff/STATE = {"Time":"2019-02-22T05:32:14","Uptime":"0T00:29:59","Vcc":1.105,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":57,"POWER":"ON","Dimmer":100,"Color":"FFFFFF00","HSBColor":"0,0,100","Channel":[100,100,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"dalek","BSSId":"0C:9D:92:55:EC:70","Channel":7,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:05"}}
05:37:14 RSL: tele/sonoff/STATE = {"Time":"2019-02-22T05:37:14","Uptime":"0T00:34:59","Vcc":1.126,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":57,"POWER":"ON","Dimmer":100,"Color":"FFFFFF00","HSBColor":"0,0,100","Channel":[100,100,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"dalek","BSSId":"0C:9D:92:55:EC:70","Channel":7,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:05"}}
05:42:14 RSL: tele/sonoff/STATE = {"Time":"2019-02-22T05:42:14","Uptime":"0T00:39:59","Vcc":1.117,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":58,"POWER":"ON","Dimmer":100,"Color":"FFFFFF00","HSBColor":"0,0,100","Channel":[100,100,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"dalek","BSSId":"0C:9D:92:55:EC:70","Channel":7,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:05"}}
05:47:14 RSL: tele/sonoff/STATE = {"Time":"2019-02-22T05:47:14","Uptime":"0T00:44:59","Vcc":1.106,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":58,"POWER":"ON","Dimmer":100,"Color":"FFFFFF00","HSBColor":"0,0,100","Channel":[100,100,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"dalek","BSSId":"0C:9D:92:55:EC:70","Channel":7,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:05"}}
05:52:14 RSL: tele/sonoff/STATE = {"Time":"2019-02-22T05:52:14","Uptime":"0T00:49:59","Vcc":1.124,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":58,"POWER":"ON","Dimmer":100,"Color":"FFFFFF00","HSBColor":"0,0,100","Channel":[100,100,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"L [... cut ...] ,"BSSId":"0C:9D:92:55:EC:70","Channel":7,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:05"}}
07:32:13 RSL: tele/sonoff/STATE = {"Time":"2019-02-22T07:32:13","Uptime":"0T02:29:58","Vcc":1.113,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":60,"POWER":"ON","Dimmer":100,"Color":"FFFFFF00","HSBColor":"0,0,100","Channel":[100,100,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"dalek","BSSId":"0C:9D:92:55:EC:70","Channel":7,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:05"}}
```

localhost61, 2019-02-22:
Are they normal these Vcc values?
- 2.5V on `light2`
- 1.1V on `lamp5`

maikthulhu, 2019-02-22:
@localhost61 after turning both lamps off last night and having them on again this morning (no RGBW, PWM1 100%) they appear to be running ~2.2V.  The bulletproof original lamp (call it `lamp1`) is also at ~2.2V.

This issue is beginning to feel a little crazy to me now.  `lamp5` was running fine on `color FFFFFF00` for a few hours last night so I set it to `color 000000FF` (no RGB, full warm white), expecting it to run for the 20-30 minutes and then start rebooting.  It never did after about an hour.  I turned it off and went to bed, turned it back on this morning (along with `lamp1` and `light2`) and `lamp5` has been running solid (`color 000000FF`) for 1.5hrs now.  The definition of insanity comes to mind.

Could it be the hardware needed some sort of burn-in period?  Yes I know that kind of sounds crazy, but as hardware goes, stranger things have happened.  `lamp1` is my oldest bulb and had been running on the Tuya infrastructure with stock firmware for over a month.  `light2` was originally set up on the Tuya infrastructure and used briefly, then sat until a couple days ago when I flashed it with tasmota and has never experienced the rebooting problems.

The 3 remaining bulbs (`lamp3`, `lamp4`, and `lamp5`) were all flashed immediately and had never been registered with Tuya.  `lamp3` and `lamp4` have been off (after experiencing the reboot problem) and are sitting on my desk.  I'm considering attempting to induce the reboot problem on one of them a number of times (with proper cooldown since I'm suspecting thermal), then setting it to `color FFFFFF00` for a couple of hours, then setting it to `color 000000FF` to see if it makes the problem go away again. If so I guess that's a good thing, but doesn't get us to root cause (or even being able to turn the problem on and off, just off).

I'm open to any suggestions.  I also have a 6th bulb (same model) brand new in the box available for testing if anyone has any ideas here.

Sorry for being so long-winded here, just wanting to document everything.

localhost61, 2019-02-22:
Sorry, I can't help :-(

andrethomas, 2019-02-22:
There were some citations of this in the original discussion when the PR here: https://github.com/arendst/Sonoff-Tasmota/pull/4991#issuecomment-459696481

It would appear not all of these bulbs are manufactured to sustain 100% lighting for extended periods of time so from one brand to another there may be differences with regard to this in terms of build quality.

This means that if you're using one of the "not so well designed bulbs" then you'd have to limit the maximum light to a level which is acceptable to the particular manufacturers manufactiring quality... i.e. never go over 80% or what ever you determine to be a good threshold for light vs. stability tradeoffs.

Unfortunately there is nothing in the firmware that can know whether or not the particular device you have is of substandard manufacturing quality/specification.

ascillato2, 2019-02-22:
Closing this issue as it is a hardware issue. We are sorry for that. There isn't much that can be done from the software side.

Anyway, if you have further questions / comments, please do not hesitate on sharing them. 

Thanks everyone for sharing theirs ideas.

closed 2019-02-22 by ascillato2

maikthulhu, 2019-02-23:
Totally understandable.  I want to add this though, as crazy as it sounds just bear with me:

I have been running build c97ea4d compiled locally with `PWM_LIGHTSCHEME0_IGNORE_SLEEP` defined (`sonoff/sonoff.h:107` uncommented) on `lamp3`.  After the initial flash with `sonoff-basic.bin` and then `sonoff.bin` I set `color 000000FF` to full white.  `lamp3` started rebooting after 20-30 minutes. I removed power and let it cool off, then powered it up and set `color FFFFFF00` and let it sit for two hours with no adverse effects.  I then set `color 000000FF` again and it's been running for over an hour without issue.

This behavior matches `light5` where the problem just seemed to stop happening after I let the bulb sit in `color FFFFFF00` for a couple hours.  I don't like this answer, but it's worth noting in case anyone has this problem with these devices in the future.

As a side note, every bulb I've flashed with this build has settled to `LoadAvg: 19`.  Previous to that they were all over the map, with most of them increasing well past 100 (and jumping 4-8 points whenever I'd do something like `state` from the console).

That's all I've got for this issue.  Keep up the great work!

joba-1, 2019-02-23:
just an idea: how about a setoption that limits (r+g+b) value?
-- 
iphone so i typo

andrethomas, 2019-02-23:
> just an idea: how about a setoption that limits (r+g+b) value?

Not a bad idea but it basically means modifying the firmware behaviour to compensate for what appears to be bad design from a physics perspective so I would rather go the route of limiting it in the automation software since it is an acceptable workaround for isolated use case scenarios.

The LoadAvg is just a representation of how long the main loop process takes in relation to the target loop duration (50ms for sleep 50, 100ms for sleep 100 etc) - Its just an indication on whether or not the device sleeps enough to maintain a given loop target speed thereby ensuring that the SDK is given enough time to perform the processes necessary to maintain wifi and tcp/ip functionality.

So in short 19 is about common for a sleep setting of 50 at this point in time which means the main loop takes 19/100*50 milliseconds to complete (~9.5ms) - Any delays or runtime added to the main loop function like when interacting via http or mqtt will momentarily increase the main loop duration to more than 9.5 milliseconds resulting in an elevated load average.