arendst/Tasmota #5306: NodeMCU Random Disconnect and Reconnect

alagarath, 2019-02-22:
First off, I love the Tasmota firmware and this is the first one I've had trouble with! I'm leaning toward the 6.4.1 firmware because of the added RFsend and RFreceive functionality. I wish I had more devices with 6.4.1 but it appears that the latest 6.4.1 firmware has random disconnects if there isn't traffic being generated to the device. If I ping the device I can go for 24 hours without a disconnect or "unavailable" status in Home Assistant. Without the traffic the NodeMCU will spit out this output anywhere between 1-10 minutes. Occasionally it will be a couple times in a minute.

```
20:14:34 MQT: stat/sonoff/RESULT = {"RFSend":"Done"}
20:14:37 MQT: stat/sonoff/RESULT = {"RFSend":"Done"}
20:15:43 WIF: Connecting to AP1 SSID in mode 11N as sonoff-1808...
20:15:47 WIF: Connected
20:15:47 DNS: Initialized
20:15:48 MQT: Attempting connection...
20:15:48 MQT: Connected
20:15:48 MQT: tele/sonoff/LWT = Online (retained)
20:15:48 MQT: cmnd/sonoff/POWER = 
20:16:21 DNS: Initialized
20:16:22 MQT: Attempting connection...
20:16:22 MQT: Connected
20:16:22 MQT: tele/sonoff/LWT = Online (retained)
20:16:22 MQT: cmnd/sonoff/POWER = 
```

I'm running about 10 other Tasmota devices on varying firmware levels and I'm not seeing this issue. I'm wondering if it is related to the 2.4.2 Espressif but I haven't tried a recompile with an older version. This NodeMCU was running OpenMQTT Gateway before and I wasn't seeing this behavior. 

Workaround
1. Ping constantly

Tried
1.  reset 5
2. reset 3
3. Blank 4MB bin flash with espeasy and reflash with 6.4.1
4. Tried with nothing plugged in on two different power sources
5. Tried the reflash above without restoring configuration

Haven't tried
1. Different NodeMCU (shouldn't be the case since it was working with OpenMQTT Gateway)
2. Different power cord (shouldn't be it since it works will pinging)
3. Roll back to older Espressif version with source code and flash

ascillato2, 2019-02-22:
Please, check the wiki at troubleshooting. 

Closing this issue as it is duplicated. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-22 by ascillato2

alagarath, 2019-02-22:
I should have given that page a closer look. I had run across it before and didn't dig in. Sorry I didn't go through the proper channels to troubleshoot but thank you for the quick response.

alagarath, 2019-02-22:
If anyone finds this thread this is my router/tasmota information

Cisco Linksys EA6700 flashed with Advanced Tomato 1.28.0000 -3.4-140 K26ARM USB AIO-64K 
Tasmota release version I'm using 6.4.1 core version 2.3 from http://thehackbox.org/tasmota/release/020300/
Devices flashed are a NodeMCU and I also flashed my Shelly1 that had an occasional connection issue that might have been related to the development version that was running on it

I'm 30 minutes in with no drops. That gets a thumbs up for me based on its behavior before.