arendst/Tasmota #5307: counter n pulse one count

FraatRamski, 2019-02-22:
Just started on a new project water counter. 
The issue is that I will get n pulse for 1 liter. 

Where I can implement this?

ascillato, 2019-02-22:
Just define the gpio you want to use as counter and you will see the counter value in the mqtt broker.

For more information please check the wiki at commands
For other examples search in issues
For support, please use the Tasmota Support Chat. Github issues are for bug reporting. Thanks

ascillato2, 2019-02-22:
Sorry. Closing this issue as it is duplicated. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-22 by ascillato2

FraatRamski, 2019-02-22:
@ascillato: I know how to setup a counter. My question is I will get i.e. 238 pulses for 1 liter and I will not do any calkulation in my mqtt broker.

ascillato2, 2019-02-22:
One way to do that could be:

Set teleperiod to 60
Set a rule to reset the counter every teleperiod
Make that rule also to do the calculation and store the value in a VAR
Publish that VAR with the topic you want. 

For more information please check the wiki at commands
For other examples search in issues
For support, please use the Tasmota Support Chat. Github issues are for bug reporting. Thanks