arendst/Tasmota #5309: Esp-relay 01

weed2all, 2019-02-23:
Hi...it is more like a question and not an issue..
i have a bunch of esp8266 relays and I want to know if the tasmota will work with this...the esp relay modules that I have is exactly the ones in image below...if the tasmota will work on this I only need the ESPeasy to flash the firmware or I need some other software?
Thank You!
[image]

ascillato2, 2019-02-23:
Hi

Yes, Tasmota works on that device. You will need esptool.py for flashing as explained in Tasmota wiki. The flashing process will work also with ESPEasy flasher.

In the wiki are explained all the flashing steps and the serial hardware required. Anyway, if you need further assistant please find us in the Tasmota Support Chat. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-23 by ascillato2

zoomx, 2019-03-26:
There are two models of this board.
One has the relay connected to one ESP8266 pin, usually GPIO2
The other one has another small MCU, an ST8 or an 8051 clone. In this model a command is sent by the ESP8266 to the other MCU by serial and the other CPU enable or disable the relay. This model uses an ESP-01 with AT firmware. It seems like the one in photo above, the small MCU cab be under the ESP-01.

lalo-uy, 2019-03-26:
That one seems to be direct connected, no cpu

[http://ae01.alicdn.com/kf/HTB1MOnrSFXXXXbbXFXXq6xXFXXXz.jpg](url)

zoomx, 2019-03-27:
Right, the other one is very similar but has a small 8 pins MCU.