arendst/Tasmota #5310: Need option to change RGB color to BGR (SM16716)

dgranz, 2019-02-23:
The color bytes are in BGR order on the Feit Electric RGB bulbs (which use the SM16716)
Please add an option to select a BGR ordering.

Jason2866, 2019-02-23:
@dgranz  SM16716 implementation is made from @gsimon75 
Maybe he can add this as a option....

dgranz, 2019-02-24:
Around line 435 of xdrv_04_light.ino
changed to:

  // send 24-bit rgb data
  SM16716_SendByte(duty_b);
  SM16716_SendByte(duty_g);
  SM16716_SendByte(duty_r);

makes the Feit Electric bulb colors work.

Unfortunately I don't have the time to figure out how to make it a configurable option.

daninfuchs, 2019-02-24:
I suspect you could use Generic or the new Template feature (which I've just learned of) to swap PWM pin order to self correct this - I've been doing similarly (Generic, that is) for a set of unsupported WoMo Smart LED controllers for a while now.

dgranz, 2019-02-24:
> I suspect you could use Generic or the new Template feature (which I've just learned of) to swap PWM pin order to self correct this - I've been doing similarly (Generic, that is) for a set of unsupported WoMo Smart LED controllers for a while now.

Unfortunately those don't work as the color order is hardwired within the SM16716 code.

gsimon75, 2019-02-24:
Right, sm16716 doesn't use pwms, it's a completely different controller.
I've added a config option (like `#define USE_SM16716_RGB_ORDER     SM16716_RGB` in `my_user_config.h`), but I'd like to test it before submitting the pull request :).

DavinKD, 2019-02-24:
MY9231 has the same issue. It would be good if these were handled as well. I have some Lohas bulbs that have two like the AILights, but in a different order for color channels.

arendst, 2019-02-24:
Discussed with @gsimon75 (#5322) who will provide a SetOption version.

gsimon75-temp, 2019-02-25:
There is a problem with the color mapping code recently merged: when a domoticz server contacts the bulb, for some reason the `color_remap[]` array gets overwritten.

Not a solution, but a workaround is to move its declaration from below kFixedColdWarm to below light_signal_color:

```
diff --git a/sonoff/xdrv_04_light.ino b/sonoff/xdrv_04_light.ino
index c606425..d765a9f 100644
--- a/sonoff/xdrv_04_light.ino
+++ b/sonoff/xdrv_04_light.ino
@@ -86,8 +86,6 @@ struct LCwColor {
 #define MAX_FIXED_COLD_WARM  4
 const LCwColor kFixedColdWarm[MAX_FIXED_COLD_WARM] PROGMEM = { 0,0, 255,0, 0,255, 128,128 };
 
-uint8_t color_remap[5];
-
 uint8_t ledTable[] = {
   0,  1,  1,  1,  1,  1,  1,  1,  1,  1,  1,  1,  1,  1,  1,  1,
   1,  1,  1,  1,  1,  1,  1,  1,  1,  1,  1,  1,  1,  1,  1,  1,
@@ -112,6 +110,8 @@ uint8_t light_new_color[5];
 uint8_t light_last_color[5];
 uint8_t light_signal_color[5];
 
+uint8_t color_remap[5];
+
 uint8_t light_wheel = 0;
 uint8_t light_subtype = 0;
 uint8_t light_device = 0;
```

I don't yet know the root cause, but in the current form it is broken.

Unfortunately my account (gsimon75) has been flagged, so I can't commit until they lift the ban, so someone please take a look at this.

arendst, 2019-02-25:
That's a strange problem. I'll move the declaration to your suggested (and mine preferred) location anyway.

gsimon75-temp, 2019-02-25:
Mystery solved, it's a buffer overflow :)

In LightColorEntry() the array `light_entry_color[]` can be over-addressed [here](https://github.com/arendst/Sonoff-Tasmota/blob/9f00978a799e3e286be339ab6344da3c307a87a7/sonoff/xdrv_04_light.ino#L1257)

It turned out that my Domoticz v4.9700 sends the color value string not just up to the trailing zero, but beyond that:
```
20:39:43 LightCommand() #1 XdrvMailbox.data='1101040000' data_len=400
20:39:43 light_entry_color[0] = 17
20:39:43 light_entry_color[1] = 1 
20:39:43 light_entry_color[2] = 4 
20:39:43 light_entry_color[3] = 0 
...
20:39:43 light_entry_color[71] = 0
20:39:43 SRC: Light
```

As `buffer_length > 3` is true, the loop runs up to `buffer_length / 2` over-addressing the 5-byte-long `light_entry_color`.

Here is a simple patch for it:
```
diff --git a/sonoff/xdrv_04_light.ino b/sonoff/xdrv_04_light.ino
index c606425..a3db1c0 100644
--- a/sonoff/xdrv_04_light.ino
+++ b/sonoff/xdrv_04_light.ino
@@ -1253,9 +1253,15 @@ bool LightColorEntry(char *buffer, uint8_t buffer_length)
     entry_type = 2;                                 // Decimal
   }
   else if (((2 * light_subtype) == buffer_length) || (buffer_length > 3)) {  // Hexadecimal entry
-    for (uint8_t i = 0; i < buffer_length / 2; i++) {
+    uint8_t limit = buffer_length / 2;
+    if (limit > sizeof(light_entry_color)) {
+      limit = sizeof(light_entry_color);
+    }
+    for (uint8_t i = 0; i < limit; i++) {
       strlcpy(scolor, buffer + (i *2), 3);
       light_entry_color[i] = (uint8_t)strtol(scolor, &p, 16);
+      //snprintf_P(log_data, sizeof(log_data), PSTR("light_entry_color[%d] = %d"), i, light_entry_color[i]);
+      //AddLog(LOG_LEVEL_DEBUG);
     }
     entry_type = 1;                                 // Hexadecimal
   }
```

Taking a look at the symbol table shows that before the appearance of this color_remap (in my case) these were overwritten by this buggy Domoticz string:
```
  3fff0451 g     O .bss   00000005 light_entry_color
  3fff0458 g     O .bss   00000004 energy_command_code
  3fff045c g     O .bss   00000004 ticker_energy
  3fff0460 g     O .bss   00000001 energy_fifth_second
  3fff0462 g     O .bss   00000002 energy_mplw_counter
  3fff0464 g     O .bss   00000002 energy_mplh_counter
  3fff0466 g     O .bss   00000001 energy_mplr_counter
  3fff0467 g     O .bss   00000001 energy_max_energy_state
  3fff0468 g     O .bss   00000001 energy_max_current_flag
  3fff0469 g     O .bss   00000001 energy_min_current_flag
  3fff046a g     O .bss   00000001 energy_max_voltage_flag
  3fff046b g     O .bss   00000001 energy_min_voltage_flag
  3fff046c g     O .bss   00000001 energy_max_power_flag
  3fff046d g     O .bss   00000001 energy_min_power_flag
  3fff046e g     O .bss   00000001 energy_type_dc
  3fff046f g     O .bss   00000001 energy_power_delta
  3fff0470 g     O .bss   0000000c energy_power_last
  3fff047c g     O .bss   00000004 energy_period
  3fff0480 g     O .bss   00000004 energy_kWhtoday
  3fff0484 g     O .bss   00000004 energy_kWhtoday_delta
  3fff0488 g     O .bss   00000004 energy_total
```

If someone has experienced problems with Domoticz and these energy-related things, here's the reason :).

arendst, 2019-02-25:
Thx for reporting. Just merged the fix.

arendst, 2019-02-25:
Oh by the way the initial issue has been solved by the release of PR #5326

closed 2019-02-26 by ascillato2