arendst/Tasmota #5311: DNS: Failed

mpt99, 2019-02-23:
Release 6.4.1 with core 2.5
DNS problem after variable online time.
here is the log:

06:23:29 LOG: SerialLog 0, WebLog 3, SysLog 0, LogHost , LogPort 514, TelePeriod 90
06:23:29 HTP: Configuration
06:23:30 CFG: Saved to flash at 3FB, Count 9, Bytes 3584
06:23:33 HTP: Main Menu
06:23:35 HTP: Console
06:24:36 RSL: STATE = {"Time":"2019-02-23T06:24:36","Uptime":"0T00:01:43","Vcc":3.448,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SID3550","BSSId":"34:81:C4:19:E9:D8","Channel":1,"RSSI":70}}
06:24:36 RSL: SENSOR = {"Time":"2019-02-23T06:24:36","AM2301":{"Temperature":19.8,"Humidity":44.7},"TempUnit":"C"}
06:26:06 RSL: STATE = {"Time":"2019-02-23T06:26:06","Uptime":"0T00:03:13","Vcc":3.447,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SID3550","BSSId":"34:81:C4:19:E9:D8","Channel":1,"RSSI":68}}
06:26:06 RSL: SENSOR = {"Time":"2019-02-23T06:26:06","AM2301":{"Temperature":19.9,"Humidity":44.6},"TempUnit":"C"}
06:27:36 RSL: STATE = {"Time":"2019-02-23T06:27:36","Uptime":"0T00:04:43","Vcc":3.449,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SID3550","BSSId":"34:81:C4:19:E9:D8","Channel":1,"RSSI":84}}
06:27:36 RSL: SENSOR = {"Time":"2019-02-23T06:27:36","AM2301":{"Temperature":19.9,"Humidity":43.4},"TempUnit":"C"}
06:29:06 RSL: STATE = {"Time":"2019-02-23T06:29:06","Uptime":"0T00:06:13","Vcc":3.435,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SID3550","BSSId":"34:81:C4:19:E9:D8","Channel":1,"RSSI":86}}
06:29:06 RSL: SENSOR = {"Time":"2019-02-23T06:29:06","AM2301":{"Temperature":19.8,"Humidity":44.3},"TempUnit":"C"}
06:30:32 UPP: Multicast disabled
06:30:35 DNS: Failed
06:30:35 UPP: Multicast (re)joined
06:30:36 RSL: STATE = {"Time":"2019-02-23T06:30:36","Uptime":"0T00:07:43","Vcc":3.439,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SID3550","BSSId":"34:81:C4:19:E9:D8","Channel":1,"RSSI":84}}
06:30:36 RSL: SENSOR = {"Time":"2019-02-23T06:30:36","AM2301":{"Temperature":19.7,"Humidity":44.5},"TempUnit":"C"}
06:30:36 DNS: Failed
06:30:37 DNS: Failed
06:30:38 DNS: Failed
06:30:39 DNS: Failed
06:30:40 DNS: Failed
06:30:41 DNS: Failed
06:30:42 DNS: Failed
06:30:43 DNS: Failed
06:30:44 DNS: Failed
06:30:45 DNS: Failed

What could be the issue?

Jason2866, 2019-02-23:
Issue template incomplete! Status 0 is missing

mpt99, 2019-02-23:
07:05:16 RSL: STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff2"],"Topic":"Sonoff2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:05:16 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"keine URL fuer Sonoff2 definiert","RestartReason":"Power on","Uptime":"0T00:09:03","StartupUTC":"2019-02-23T05:56:13","Sleep":50,"BootCount":6,"SaveCount":12,"SaveAddress":"3F8000"}}
07:05:16 RSL: STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2019-02-22T20:10:03","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
07:05:16 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SID3550","JD"],"TelePeriod":90,"SetOption":["00008000","55C180C0","00000000"]}}
07:05:16 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":481,"Free":3592,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0F000790","00000000","000000CE","000013C0"]}}
07:05:16 RSL: STATUS5 = {"StatusNET":{"Hostname":"Sonoff2-1217","IPAddress":"10.90.10.135","Gateway":"10.90.10.1","Subnetmask":"255.255.255.0","DNSServer":"10.90.10.1","Mac":"5C:CF:7F:BD:64:C1","Webserver":2,"WifiConfig":4}}
07:05:16 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sat Feb 23 06:05:16 2019","Local":"Sat Feb 23 07:05:16 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:20","Sunset":"17:58"}}
07:05:16 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-02-23T07:05:16","AM2301":{"Temperature":20.9,"Humidity":43.4},"TempUnit":"C"}}
07:05:16 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-02-23T07:05:16","Uptime":"0T00:09:03","Vcc":3.435,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SID3550","BSSId":"34:81:C4:19:E9:D8","Channel":1,"RSSI":68}}}

Jason2866, 2019-02-23:
Try this version http://thehackbox.org/tasmota/020300/sonoff.bin

mpt99, 2019-02-23:
test in progress ;-)

ascillato2, 2019-02-24:
Hi, any news on this?

If it works for you, please close this issue. Thanks.

mpt99, 2019-02-24:
Hello ascillato2,
In Germany we sleep at night ;-)
Core 2.3 works fine, Core 2.5 does not.
Core 2.4x has not been tested.
I have some Sonoffs on the run and want to change because of the wifi gap.
So I can not close here.
I will continue testing with a precompiled version now.
Maybe it's because of my IDE.
greetings
mpt

mpt99, 2019-02-24:
sorry, wrong button :-(

curzon01, 2019-02-25:
I don't think Tasmota can fix issue having with different SDK core versions - see [Troubleshooting Wiki Arduino Core](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#arduino-core-differences).

Even your wish is to use a newer core and this does not work within your environment, there is no other way than using a working core (in your case 2.3.0).
Personaly I have the same with 10% of my Tasmota devices. Using 4 different AVM repater a few Tasmota devices are not stable using core > 2.3.0 connected to exact one of the 4 repeater (it's an older one).

markusb68, 2019-02-25:
Disable mDNS in console: SetOption55 0 
then it's gone

kodibrain, 2019-02-26:
I have the same issue..
06:24:01 DNS: Failed
06:24:02 DNS: Failed
06:24:03 DNS: Failed
06:24:04 DNS: Failed
06:24:05 DNS: Failed
06:24:06 DNS: Failed
06:24:07 DNS: Failed
06:24:08 DNS: Failed
06:24:09 DNS: Failed
06:24:10 DNS: Failed
06:24:11 DNS: Failed
06:28:37 MQT: stat/HofStrahler/STATUS4 = {"StatusMEM":{"ProgramSize":571,"Free":432,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
06:28:37 MQT: stat/HofStrahler/STATUS5 = {"StatusNET":{"Hostname":"HofStrahler-3582","IPAddress":"192.168.2.51","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.50","Mac":"84:F3:EB:B0:ED:FE","Webserver":2,"WifiConfig":4}}
06:28:37 MQT: stat/HofStrahler/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B0EDFEFE","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
06:28:37 MQT: stat/HofStrahler/STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 26 05:28:37 2019","Local":"Tue Feb 26 06:28:37 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:38","Sunset":"18:28"}}
06:28:37 MQT: stat/HofStrahler/STATUS10 = {"StatusSNS"

kodibrain, 2019-02-26:
I have the same issue

curzon01, 2019-02-26:
@markusb68 - unfortunately not, SetOption55 is/was 0 on all these devices - there exists some/few combinations on AP/Core > 2.3.0 which are not stable in my case

mpt99, 2019-02-26:
update:

6.4.1 with core 2.5 does not  work.
6.4.1 with core 2.5 (precompiled) does not  work. 
6.4.1.18 with core 2_5_0/3.0.0-dev (precompiled) works.
6.4.1.19 with core 2_5_0/3.0.0-dev works.

SetOption55 0/1 in test

grettings

ascillato, 2019-02-26:
@mpt99

Hi,

Great to know that it is working for you using latest version. 

Please, close this issue. Thanks.

closed 2019-02-26 by mpt99

kodibrain, 2019-02-27:
thanks for testing @mpt99 

I flashed all my sonoffs with 6.4.1.19 and it works for me.