arendst/Tasmota #5312: Add retain: true to home assistant config

timstanley1985, 2019-02-23:
Hi, I have until now been manually configuring my tasmota switches in home assistant but have been experimenting with home assistant discovery this weekend.

It works great and I love the fact that you also get a sensor for each switch that include RSSI and version etc.! But I configure my switches to use a retain flag on the command topic.  This means if home assistant issues a command while the switch is either out of WiFi range of offline for any reason it picks up the command change as a retauned message when it comes back on line.  Can you considering adding this as an option please?

You would just need to add "Ret":'true" to the config payload.

Jason2866, 2019-02-23:
HA is misconfigured for Tasmota if you use Retain ON.
See Wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant---Manual-Config
There will be no change in Tasmota....

curzon01, 2019-02-23:
If link above can not be opened (result is "Create new page") on browser, use
https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant-%E2%80%90-Manual-Config

ascillato2, 2019-02-23:
Please, to avoid other problem, don't use retained messages. Use instead the automations explained in the wiki at home assistant chapter. Thanks.

closed 2019-02-23 by ascillato2