arendst/Tasmota #5314: Core 2.5.0

smartboart, 2019-02-23:
When do we get a new release  bin file with a core version which will wotk stable in wifi networks.. It is not really fun with actual tasmota release with activated  core...

Jason2866, 2019-02-23:
You have the choice to choose between core 2.3.0, 2.4.2 and 2.5.0.

smartboart, 2019-02-23:
Yes I know, but I do not want to remove all the hardware to flash it again with atom via ftdi... I need a minimal and normal bin file to flash them over the air.. Why do you guys have the core 2.4.2 as Standart when you know it makes trouble.. I have read the wiki and the trouble shooting chapter... I mean I do not understand why the bad core is in the official bin files..

ascillato, 2019-02-23:
Stage is latest core 2.5.0 with more fixes.

Soon they will release core 2.5.1

netpok, 2019-02-23:
There is a compiled version for every core, compatible with ota:
2.3.0: http://thehackbox.org/tasmota/020300/
2.4.2: http://thehackbox.org/tasmota/
2.5.0: http://thehackbox.org/tasmota/020500/

ascillato2, 2019-02-24:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-24 by ascillato2

sir106, 2019-05-29:
2.5.1: http://thehackbox.org/tasmota/020501/