arendst/Tasmota #5315: Portisch firmware Trying to listen to codes

Thebuz, 2019-02-23:
I have flash the sonoff bridge with Tasmota and Portisch firmware.

I am trying get a "B1" code to send by MQTT, it detects it rfraw 177 as 
{"RfRaw":{"Data":"AA B1 03 019A 0398 0424 280809 55"}}

This is button for a light, I need it to listen for the code a send it via MQTT so I can trigger something in home assistant (like how it works normally), I have been trying all sorts of things for the past 4 hours and now i am stressed and tired :(

it is possible to do what i am asking?

Jason2866, 2019-02-23:
This is github issue for errors, not a forum to ask for individual setup or questions.
For this visit us in Discord Chat

ascillato2, 2019-02-24:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-24 by ascillato2

solidminds, 2019-03-08:
> I have flash the sonoff bridge with Tasmota and Portisch firmware.
>
> I am trying get a "B1" code to send by MQTT, it detects it rfraw 177 as
> {"RfRaw":{"Data":"AA B1 03 019A 0398 0424 280809 55"}} [...]

TheBuz, dit you find what you were looking for?

Thebuz, 2019-03-08:
No, and i have tried so much :(

I have tried Tasmota and Portisch firmware, esphome, and a the direct sonoff bridge RF hack (by soldering / bypassing the controller chip) with and nothing works.

Some rf remotes are fine, but others only show when using rtf raw177 and I can't do anything with them from there.

Thanks for asking but I'm still stuck right now :(

solidminds, 2019-03-09:
I only get codes when in rtfraw177 mode. I cant get any codes when in normal rfraw1 mode, which is weird i think.
Or does the bridge alwayts have to be in rfraw177 mode to function?

bakman2, 2019-04-16:
> Or does the bridge alwayts have to be in rfraw177 mode to function?
I have the same question. How does it survive a restart ?