arendst/Tasmota #5316: Wemos D1 doesn't operate in Generic mode

MaiMariarti, 2019-02-23:
I've flashed tasmota v6.4.1 (sonoff.bin latest as of today) on a wemos d1 using flash_download_tools_v3.6.5 for windows, and configured it to read a DHT11 on GPIO2 in the configuration menu of Sonoff Basic Module. 
The correct DHT values show up and I can toggle an LED connected to D6 of the wemos, without any additional change in the configuration screen. Only the DHT11 was assigned. The other 4 entries are left as None(00).
If I change the module type from "Basic" to "Generic (18)" from the drop down list to get access to more I/O pins, I no longer see a toggle switch in the main screen, no matter what I change.
Although the DHT11 values are correct. The LED keeps flashing at ~ 1Hz at any port I assign to it.
What am I doing wrong? I thought Generic was supposed to work with Wemos R1 series!
Thanks in advance for any guidance.

ascillato, 2019-02-24:
To have a toggle button you need to define a relay.
Led flashing means no mqtt connected. 

Please, for support use the Tasmota Support Chat. Github issues are for bug reporting with the template completed. Thanks

MaiMariarti, 2019-02-24:
You didn't understand my problem. 
It is a bug if it works under basic and not under generic. I don't need a relay to turn on an LED. Also, MQTT is deselected (disabled) under "Configure other".
Under basic, it shows the toggle on/off by default. The button disappears if I switch to generic which used to be called Wemos. Wemos has many more I/O pins which are not being used by "basic". I want to use them.
Please read and understand my problem first before dismissing me and referring me to some chat.

ascillato2, 2019-02-24:
Tasmota does not have any button to turn on a led. A web button will be shown only if you have a relay.

Github issues are for bug reporting. 

Please, for support use the Tasmota Support Chat.

ascillato2, 2019-02-24:
Closing this issue as it doesn't have the requested template. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-24 by ascillato2