arendst/Tasmota #5319: Prevent Adrian Scillato from closing legitimate support requests.

MaiMariarti, 2019-02-24:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

Adrian Scillato doesn't seem to understand enough English to understand my writing and keeps telling me "for support use the Tasmota Support Chat"

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

Prevent him from blowing people off and closing legitimate support requests just because he feels like it.

ascillato, 2019-02-24:
Hehe,

Please, for support use the chat. I'm trying to explain the basics of Tasmota.

Github issues are for bug reporting. 

For help it is better a chat. 

Again: If you define a led, you will not have any web button to turn it on. The web buttons will be shown if you define a relay.

Please, let's chat and I can explain Tasmota usage there. 

Thanks

ascillato2, 2019-02-24:
Please, use any of the following entries for support. Thanks 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-24 by ascillato2