arendst/Tasmota #5321: Option to override displayed module name

daninfuchs, 2019-02-24:
**Have you look for this feature in other issues and in the wiki?**
Yes.

**Describe the solution you'd like**
I'd like to see an option, UI or Command, to be able to override the displayed module name ("Generic Module"  "Sonoff Basic"  etc) for more family/user friendly display. For example, unsupported devices using generic or more descriptive (Sonoff 4ch Pro to "Kitchen Controller" or similar) 

**Additional context**
I use both unsupported and supported devices in our home for convenience, and for purely aesthetic reasons feel it would be nice for them to display more useful information than each having the same title. Friendly name is well and good, but still leaves the extra static data.

*(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-02-24:
Hi,

Yes. It is already implemented in latest version 6.4.1.18 from the development branch.

You can now use templates and set the gpios and names as you want. 

Please, update to latest and try the new template feature. You can use the precompiled bins from thehackbox.org/tasmota

If this fulfill your request, please close this issue.

Thanks

daninfuchs, 2019-02-24:
Lol, that's the one. I look smart. :stuck_out_tongue_closed_eyes: 

I saw something about templates in the commands, didn't notice the wiki entry so I didn't understand it. Man you're on top of it, haha. Keep up the great work!

closed 2019-02-24 by daninfuchs