arendst/Tasmota #5323: New hardware revision v2.4 BlitzWolf® BW-SHP2

agsochi, 2019-02-24:
Hi. I have a new BlitzWolf BW-SHP2 V2.4. I managed to flash Tasmota on it. Button and Relay works with "Gosund SP1 v23 (55)" /6.4.1, blue LED is flashing 2sec period. When i push button - relay is on, but no RED led is on. How to fix RED led?

Any help to get it working completely is appreciated!

agsochi, 2019-02-24:
[image]

blakadder, 2019-02-24:
You can create a [template](https://github.com/arendst/Sonoff-Tasmota/wiki/Templates) based on module 45 and try to find out where the led red is connected. Since you have it opened you can trace the GPIOs and see exactly where the LED red is connected. 
If you manage to find the problem it would be helpful to update the wiki with the solution

agsochi, 2019-02-24:
Here is LED (red and blue) trace, but i dont't know pinup of this wifi module.
[image]

agsochi, 2019-02-24:
This is Gosund SP1_v23, Button, Relay and power metering is ok, except led.
```
 { "Gosund SP1_v23",  // https://www.amazon.de/gp/product/B0777BWS1P
 0,    
 GPIO_LED1_INV,    // GPIO01 Serial RXD and LED1 (blue) inv
 0,        
 GPIO_KEY1,        // GPIO03 Serial TXD and Button
 GPIO_HJL_CF,      // GPIO04 BL0937 or HJL-01 CF power
 GPIO_NRG_CF1,     // GPIO05 BL0937 or HJL-01 CF1 current / voltage
 0, 0, 0, 0, 0, 0, // Flash connection
 GPIO_NRG_SEL_INV, // GPIO12 BL0937 or HJL-01 Sel output (0 = Voltage)
 GPIO_LED2_INV,    // GPIO13 LED2 (red) inv
 GPIO_REL1,        // GPIO14 Relay (0 = Off, 1 = On)
 0, 0, 0
 }
```

netpok, 2019-02-24:
Without the type of the ESP module we cant do much, so please attach a photo of it.

agsochi, 2019-02-24:
Here it is:
[image]

netpok, 2019-02-24:
Based on this it should work, please try to set up GPIO13 as a relay instead of a led and try toggling it. Does the led reacts to that?

(The module looks like a tywe2s)

agsochi, 2019-02-24:
Yes, GPIO13 is RED led.

netpok, 2019-02-25:
Then it should work with the Gosund SP1_v23 config. I will check it out after work (just received a batch of TFLAG SP1)

ascillato2, 2019-02-26:
Hi,

Any news on this?

ascillato2, 2019-02-27:
Closing this issue as there is no feedback. When you have time, please tell us if now it is working for you. Thanks

closed 2019-02-27 by ascillato2

agsochi, 2019-03-01:
Strange behavior - i leave device on by night, at the morning red led blinks.
Device loose configuration - wifi settings and device type.
After reconfiguring it starts, i checked all functions - all ok, fnd after 2 mins it loose configuration again.
Red led blink and wifiAP sonoff-4247 is turned on

netpok, 2019-03-01:
Please execute `SetOption1 1` and report back after testing it.

agsochi, 2019-03-01:
Done, i will test it till tomorrow.

agsochi, 2019-03-02:
BlitzWiolf wokrs as expected since i setup it (except red led)

clue404, 2019-03-03:
@agsochi I just received a SP1 C V2.4 and I was wondering if you have a picture with the Pins labeled. It's my first one and I don't want to ruin it :D

clue404, 2019-03-05:
Okay I got it running using this Wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Gosund-SP1

agsochi, 2019-03-20:
FW 6.5.0 - Still no luck with RED LED. Relay is ON, but no RED LED is ON.

agsochi, 2019-03-27:
@hum6ug hi! Did you get a red LED?

hum6ug, 2019-03-27:
Hi, @agsochi. Sorry I was busy. The hardware isn't running right now. I will check later that evening and report to you. But from what I remember everything was working

agsochi, 2019-03-27:
Everything works, but i cant see socket ON/OFF status

hum6ug, 2019-03-29:
Here what my BlitzWolf was running until right now:

Program Version - 6.4.1.19(sonoff)
Build Date & Time - 2019-03-04T15:48:56
Core/SDK Version - 2_4_2/2.2.1(cfd48f3)

I just did a Firware Upgrade via the UI:

Program Version - 6.5.0(release-sonoff)
Build Date & Time - 2019-03-19T12:24:10
Core/SDK Version - 2_3_0/1.5.3(aec24ac9)

I can toggle the ON/OFF via the UI an I here it click in the Hardware.

agsochi, 2019-03-30:
Do you see red led on when relay is on?

SamNitro, 2019-05-25:
if i set "SetOption31 1"
I can see ON/OFF status with RED LED

sorianonet, 2019-06-03:
hi,
i've broken the IO0. is there another way to find it and enter in programmer mode?

agsochi, 2019-06-03:
> if i set "SetOption31 1"
> I can see ON/OFF status with RED LED
Yes, red light shows now ON/OFF status. What "SetOption31 1" mean?

Ic33s, 2019-06-10:
Hi

Just flashed my first smart plug and run into the same problem. 
SetOption31 is described in the wiki => https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#SetOption31

This info leads to the real cause of the blue flashing and the missing on/off indicator led: The blue flashing led indicates WLAN problems (not the case) and MQTT problems (which was the case because I haven't setup anything for this).

If you're not using MQTT at the moment: Browse to your device and go to Configuration/Configure Other => Disable the "MQTT enable" checkbox and save. After that the blue LED stops flashing after connecting to WLAN and the red LED is visible at my smart plug SP1 2.4.

Hope this helps.
Icees

Soe-Koh, 2019-07-11:
> hi,
> i've broken the IO0. is there another way to find it and enter in programmer mode?

same here, have you found a solution?

nygma2004, 2019-10-14:
Hi Guys, I have a v2.4. I flashed Tasmota 6.6.0 a few days ago. I picked Gosund SP1 v23 from the module configuration and it works fine.
The button works, the measurement works and there is a red indicator LED when the relay is ON.
SetOption1 is OFF
SetOption31 is ON

Dakhaasje, 2020-12-18:
In answer to the connections question:
For Blitzwolf SHP2 v2.4 it did not work for me to connect tx and rx to the pads on the ESP module. 
The 3.3v and gnd I connected to the pads on the back of the board, and the TX/RX to the 3rd and 4th connections of the base of the ESP, the hardest to reach side. :-)
At last I soldered a wire to a GND pad on the ESP, so I can manually connect it to io00 when starting. 
No chance of ripping the io00 pad off by accident.
Gosund SP1 v23 works fine as template. Red led and powermeasuring works.
Thank you all for providing the neccessary information!!

Dakhaasje, 2020-12-18:
In answer to the connections question:
For Blitzwolf SHP2 v2.4 it did not work for me to connect tx and rx to the pads on the ESP module. 
The 3.3v and gnd I connected to the pads on the back of the board, and the TX/RX to the 3rd and 4th connections of the base of the ESP, the hardest to reach side. :-)

At last I soldered a wire to a GND pad on the ESP, so I can manually connect it to io00 when starting. 
No chance of ripping the io00 pad off by accident.
Gosund SP1 v23 works fine as template. Red led and powermeasuring works.
Thank you all for providing the neccessary information!!

(afterthought: the 2nd one did not start after flashing. Only when I did solder it like the first one: the hard to reach pads for rx/tx and just before flashing it again using those connections it started...)

peatonet, 2021-02-11:
I tried 3 different templates including the Gosund SP1 v23 [1][2], but I was only able to turn on the red led and there is no 230v on the plug. I also tried configuring all the GPIO pins one by one as relay, but is not working for me.

I'm using this firmware (Tasmota v9.2.0 Julie):

https://github.com/arendst/Tasmota/releases

[1]: https://templates.blakadder.com/gosund_SP1.html
[2]: https://templates.blakadder.com/blitzwolf_SHP2.html

sinasu, 2021-05-13:
working good
tasmota 9.4.0

[image]

skysurf, 2021-12-12:
how do you flash it ... i flash mine like i explain here:
https://forum.cpha.pt/t/problemas-a-flashar-configurar-a-nova-tomada-blitzwolf-bw-shp2-v-2-4/9049

[image]

but when i plug on socket the blue light flash very fast and doesn't do anything more... 
i search for the wirelless tasmota and i can't find nothing and i search on 5ghz rang too ...
what i did wrong ? 
thanks for the help.

barbudor, 2021-12-12:
Hello
Nodemcu PyFlasher is not one of the recommended flashing tool. Not because it will not work but because we don' have the time to test all softwares and support all of them.
There are already 4 flashing tools for Tasmota that are recommended in our documentation including the very easy to use web flasher from https://tasmota.github.io/install/

Secondly you don't give much details about the hardware that you are using. although the flashing may seems to go well, it happen that it didn't worked well. The reason is generally not providing enough power to the device during flashing. This occur most of the time when using USB/serial adapter that don't have a voltage regulator on the board. We have identified a model that is working well (and is among the cheapest ones) in the documentation.
Alternatively using a Nodemcu board is also a good alternative.
Or if you happen to have a good source of 3.3V from a bench top lab power supply that can also help.

H please carefully the Getting Started part of the doc that address all the above points.

If you still have issues or are unsure, please provide more details including photos of your adapter

skysurf, 2021-12-12:
hi i flash the tasmota with arduino uno, instructions in here like i said:

https://forum.cpha.pt/t/problemas-a-flashar-configurar-a-nova-tomada-blitzwolf-bw-shp2-v-2-4/9049

[image]

but when i plug on socket the blue light flash very fast and doesn't do anything more...
i search for the wirelless tasmota and i can't find nothing and i search on 5ghz rang too ...
what i did wrong ?

but i can flash by you tool and with Nodemcu  was well . thanks

sfromis, 2023-06-05:
No, gpio 0 is not accessible other places, unless you remove the metal shield on the other side, to get to the MCU chip. Or *maybe* if you find a place to scrape off the paint on the PCB.

A common issue is using an USB/serial adapter not having enough power to act as a power supply while flashing. Those CPxxxx/PLxxxx are usually not good. The recommended "golden CH340G" board has an extra voltage regulator to ensure this.

skysurf, 2023-06-05:
> I'm happy for you @skysurf that it worked, on my Blitzwolf SHP2 V2.4 unfortunately I removed the I00 connection point during soldering just like @sorianonet I tried flashing it with a CP2102 it failed so many times and I still have PL2303TA and CH340 available if needed. Is there a alternative i00 spot on the device motherboard? Do you or others have any advice what I could do now? [image]

Try to scratch I00, to see if you can get the pad again...

bibkits, 2023-06-17:
[image]
Thanks all for the answers, but where is the pad or trace off GPIO0.
Another problem is playing now, sometimes the device is going on and off. When I plug it into the FTDI than it sometimes apppears in device manager of Windows. Only when i remove 3V3 or GND. Does anybody knows a solution for that problem.