arendst/Tasmota #5329: Basic R2 - WDT RESET - Have to double loop to flash and issue commands to get operational

AEWaller, 2019-02-24:
- Upload tools used : ESP Easy Flasher
- Hardware used :  Sonoff Basic R2 v1.0
- Pre-compiled release 6.4.1
You have tried latest release or development binaries? : No
All work is being done through a serial FTDI interface 
```
I have four units. they are failing after a WDT RESET status message (see logs below).
I have flashed 8 other devices, Original Basic's as well as S26's with the same tools and Tasmota pre-compiled release (so I know this setup works).
The failure is at 1:44 when the device tries to restart after WiFi name is sent, but before WiFi password is input. The error is WDT RESET.
Upon the restart the green LED does not flash and no Serial commands are responded to, **HOWEVER**...
unplug the unit and wait over 5 minutes, plug it back in and Tasmota restarts and green light flashes and button commands set/unset the relay correctly. and Serial interface returns to normal.
Carry out the power off/on before 3-4 minutes and the device remains "dead" (no flash no serial)
After a successful power up of the device I issue a Status 0 command and the units respond as normal, so I issue the WiFi Password command and the device now continues to work (WiFi console successful) 
 for 2 hours without error as normal.

This is confusing, as all the new R2 devices have all acted the same in going 'dead" on a WDT reset.  By double looping the setup cycle I have succeeded in getting two devices up and running.

I'm new to this (and enjoying the investigation) and hope this might help someone else. 

STATUS 0 OUTPUT HERE
`status 0
00:00:47 CMD: status 0
00:00:47 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:00:47 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:46","StartupUTC":"","Sleep":50,"BootCount":8,"SaveCount":10,"SaveAddress":"FA000"}}
00:00:47 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":7,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
00:00:47 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
00:00:47 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
00:00:47 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1130","IPAddress":"0.0.0.0","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:F9:A4:6A","Webserver":2,"WifiConfig":4}}
00:00:47 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F9A46A","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:47 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:47 1970","Local":"Thu Jan 01 00:00:47 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
00:00:47 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:47"}}
00:00:47 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:47","Uptime":"0T00:00:46","Vcc":4.466,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","BSSId":"18:40:A4:CA:7B:35","Channel":1,"RSSI":100}}}
ipaddress1 192.168.0.135
00:01:11 CMD: ipaddress1 192.168.0.135
00:01:11 RSL: stat/sonoff/RESULT = {"IPAddress1":"192.168.0.135 (0.0.0.0)"}
ipaddress2 192.168.0.1
00:01:19 CMD: ipaddress2 192.168.0.1
00:01:19 RSL: stat/sonoff/RESULT = {"IPAddress2":"192.168.0.1"}
ipaddress3 255.255.255.0
00:01:27 CMD: ipaddress3 255.255.255.0
00:01:27 RSL: stat/sonoff/RESULT = {"IPAddress3":"255.255.255.0"}
ipaddress4 192.168.0.1
00:01:35 CMD: ipaddress4 192.168.0.1
00:01:35 RSL: stat/sonoff/RESULT = {"IPAddress4":"192.168.0.1"}
ssid1 PrettyFlyForAWiFi
00:01:42 CMD: ssid1 PrettyFlyForAWiFi
00:01:42 RSL: stat/sonoff/RESULT = {"SSId1":"PrettyFlyForAWiFi"}
00:01:44 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

-------------------
FLASH LOG HERE:
######2019-02-25######
#######0.04.002#######
######FLASH INFO######
BIN file: sonoff.bin
COM port: (COM6) FTDI USB Serial Port (Port_#0002.Hub_#0002)
Baud rate: 115200
######POST FLASH######
No post flash information entered...
######FLASH LOG######
[esptool.exe -vv -cd nodemcu -cb 115200 -cp COM6 -ca 0x00000 -cf "C:\Users\Andrew\Documents\Sonoff Flashing tools\bin\sonoff.bin"]
[25/02/2019 12:44:43 AM] esptool v0.4.12 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
[25/02/2019 12:44:43 AM] 	setting board to nodemcu
[25/02/2019 12:44:43 AM] 	setting baudrate from 115200 to 115200
[... 39 lines cut ...]
[25/02/2019 12:44:43 AM] 	espcomm_send_command: sending command payload
[25/02/2019 12:44:43 AM] 	setting serial port timeouts to 15000 ms
[25/02/2019 12:44:43 AM] 	setting serial port timeouts to 1000 ms
[25/02/2019 12:44:43 AM] 	espcomm_send_command: receiving 2 bytes of data
[25/02/2019 12:44:43 AM] 	writing flash
[25/02/2019 12:44:52 AM] ................................................................................ [ 15% ]
[25/02/2019 12:45:01 AM] ................................................................................ [ 30% ]
[25/02/2019 12:45:10 AM] ................................................................................ [ 45% ]
[25/02/2019 12:45:19 AM] ................................................................................ [ 61% ]
[25/02/2019 12:45:28 AM] ................................................................................ [ 76% ]
[25/02/2019 12:45:37 AM] ................................................................................ [ 91% ]
[25/02/2019 12:45:42 AM] ..........................................                                       [ 100% ]
[25/02/2019 12:45:42 AM] starting app without reboot
[25/02/2019 12:45:42 AM] 	espcomm_send_command: sending command header
[25/02/2019 12:45:42 AM] 	espcomm_send_command: sending command payload
[25/02/2019 12:45:42 AM] 	espcomm_send_command: receiving 2 bytes of data
[25/02/2019 12:45:42 AM] closing bootloader
[25/02/2019 12:45:42 AM] 	flush start
[25/02/2019 12:45:42 AM] 	setting serial port timeouts to 1 ms
[25/02/2019 12:45:42 AM] 	setting serial port timeouts to 1000 ms
[25/02/2019 12:45:42 AM] 	flush complete
[2019-02-25 00:45:42] DONE!
############################
[2019-02-25 00:45:42] The COM port was already in use... the post-flash did not complete.
[2019-02-25 00:45:42] JOB COMPLETED, over and out!

Jason2866, 2019-02-24:
Please erase flash with esptool. Use esptool to flash since your tool didnt successfuly flashed the device.
`[2019-02-25 00:45:42] The COM port was already in use... the post-flash did not complete.`
https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool

Jason2866, 2019-02-26:
Any new info here?

ascillato2, 2019-02-26:
Closing this issue as there is no feedback. When you have time please tell us if reflashing solved your issue. Thanks

closed 2019-02-26 by ascillato2