arendst/Tasmota #5330: Fatal exception when use rules

clientgithub, 2019-02-24:
Tasmota 6.4.1.18, Sonoff Basic module + DHT11 on GPIO14
Rule1 on tele-DHT11#temperature do backlog var1 %value%; sub1 1.2; publish stat/sonoff_basic_01/TEMP %var1% endon

After same time (1 min - 1 day) module reseting with message:
{"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000e1b2 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}

After reset module not connecting to Wi-Fi. After manual reset, connect ok.

Jason2866, 2019-02-24:
Issue template not complete. Please provide `Status 0` Output.

clientgithub, 2019-02-24:
Sorry, i reset module( Next time i'm saving Status 0

netpok, 2019-02-24:
It shouldn't be the reason but your rule will be delayed one tick as the variable replacements happens before execution of the command.

1. tick:
let `%value%` be `1` and the `%var1%` is the default empty:
`backlog var1 1; sub1 1.2; publish stat/sonoff_basic_01/TEMP ` will be executed
2. tick
let `%value%` be `2`:
`backlog var1 2; sub1 1.2; publish stat/sonoff_basic_01/TEMP -0.2` will be executed
3. tick
let `%value%` be `0`:
`backlog var1 0; sub1 1.2; publish stat/sonoff_basic_01/TEMP 0.8` will be executed

clientgithub, 2019-02-24:
netpok, thank you for answer and help. I'm make new rule, maybe now fatal exception goto ...))

closed 2019-02-26 by ascillato2