arendst/Tasmota #5335: Sonoff sleep mode?

Silvrav, 2019-02-24:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
I have 2 sonoff basic's that this morning had a slow 2-3 second blink and did not respond to HA or alexa voice commands. Once I switch the 2 sonoff devices on by pressing the toggle button on the actual unit, the blinking stopped and the LED went solid (as usual when the device relay is on) and then responded to HA and alexa voice commands. It was almost as if the 2 sonoffs went into some kind of sleep mode?

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff basic V1.2
- [x] _Development/Compiler/Upload tools used_ : ESPeasy flash on one and arduinoe IDE on second
- [x] _If a pre-compiled release or development binary was used, which one?_ : v6.4.1 .bin file used
- [x] _You have tried latest release or development binaries?_ : Latest
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
Leave over night

**Expected behavior**
To be connected to AP without blinking and respond to commands from HA or alexa

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-02-24:
There is no sleep mode in Tasmota. Probably something is wrong with your wifi or with your Home Automation solution setup.

Silvrav, 2019-02-24:
> There is no sleep mode in Tasmota. Probably something is wrong with your wifi or with your Home Automation solution setup.

Thank you, will recheck those 2. I flashed 10 new units over the weekend and they all connect fine, its just these 2.

netpok, 2019-02-25:
Please try the latest core 2.5.0 development builds available here: http://thehackbox.org/tasmota/020500/

Core 2.5.0 should be more lenient with wifi than core 2.4.2.

ascillato2, 2019-02-26:
Hi,

Any news on this?

Silvrav, 2019-02-26:
It seems to still be doing it BUT it is now responding to commands (voice from alexa, HA and mqtt). I away for a bit but will pull the full log from the unit itself once I can

ascillato2, 2019-02-27:
Please, share the output of the Tasmota console. Remember to set weblog to 4 for more debug information. Seems that your issue is mqtt disconnections.

Try the precompiled bins from development branch with core 2.3.0 in order to solve your issue 
Thehackbox.org/tasmota/020300

GreyMarvel, 2019-03-01:
I have the same problem and it's not specific to a home automation platform . I updated all my sonoff T1's from V6.3.0 to V 6.4.1 and now all my sonoffs occasionally lose connection and blink 3 times slowly before reconnecting on their own  I tried putting the server to a different home automation server (home assistant to openhab ) and the problem is the same , both are using mosquitto server. I'm not able to provide logs at the moment but I will try in a few days . 

Here is a bit of home assistant log 

1551420261: Client diningroom has exceeded timeout, disconnecting. 1551420261: Socket error on client diningroom, disconnecting. 1551420266: New connection from 192.168.1.194 on port 1883. [INFO] found diningroom on Home Assistant 1551420268: New client connected from 192.168.1.194 as diningroom (c1, k10, u'diningroom'). 1551420278: Socket error on client powersocket, disconnecting. 1551420278: New connection from 192.168.1.135 on port 1883. [INFO] found powersocket on Home Assistant 1551420280: New client connected from 192.168.1.135 as powersocket (c1, k10, u'powersocket').

Silvrav, 2019-03-01:
> Please, share the output of the Tasmota console. Remember to set weblog to 4 for more debug information. Seems that your issue is mqtt disconnections.
> 
> Try the precompiled bins from development branch with core 2.3.0 in order to solve your issue
> Thehackbox.org/tasmota/020300

:33:46 CFG: Saved to flash at F6, Count 190, Bytes 3584
08:33:46 DNS: Failed
08:33:47 DNS: Failed
08:33:48 DNS: Failed
08:33:49 DNS: Failed
08:33:49 HTP: Main Menu
08:33:49 HTP: WeMo basic event
08:33:50 DNS: Failed
08:33:51 DNS: Failed
08:33:52 HTP: Console
08:33:52 WIF: Checking connection...
08:33:52 WIF: Connected
08:33:52 DNS: Failed
08:33:53 DNS: Failed
08:33:54 DNS: Failed
08:33:55 DNS: Failed
08:33:56 DNS: Failed

This is from the most problematic one

GreyMarvel, 2019-03-02:
> Try the precompiled bins from development branch with core 2.3.0 in order to solve your issue
> Thehackbox.org/tasmota/020300

I tried the core 2.3.0 version on one sonoff and this caused it to have fatal errors and restart continually , I had to flash it manually to get it to stop back on 6.4.1 core 2..4.2 and the mqtt connections happen again

Jason2866, 2019-03-03:
Without the required information asked for (in issue template and again from @ascillato2)
no one can help. HA logs have for this issue no relevant info.
So please provide this info or issue will be closed.

ascillato2, 2019-03-06:
Closing this issue as there is no feedback.

Please, when you have time, try the precompiled bins from development branch with core 2.3.0 in order to solve your issue

Thehackbox.org/tasmota/020300
or
Thehackbox.org/tasmota/020402

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-06 by ascillato2