arendst/Tasmota #5337: Sonoff Username and password request

Massy1430, 2019-02-25:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

I flashed a Sonoff 4CH PRO R2 with ESPeasy - sonoff.bin and it went good. Then I accidentally modified the web admin password in Configure Other. Since then, when I try to get access to Sonoff Ip address, I get a message asking me username and password. I tried all kind of words without succeed.
Do I have flash it again? I would like to understand the problem and what's this password for.
Thank you

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used: ESPeasy flash
- [x] Hardware used: Sonoff 4CH PRO R2
- [x] If a pre-compiled release or development binary was used, which one? : sonoff-IT.bin
- [x] You have tried latest release or development binaries? : development binaries
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

netpok, 2019-02-25:
You set up a password for the webinterface, you can login with the username admin and the password you entered.

If you don't remember that password:
Option 1: You have serial connection to the device: Execute `WebPassword 0` through serial

Option 2: You have configured mqtt: Send `0` to `cmnd/<device-topic>/WebPassword`

Option 3 if above not available:
Since tasmota uses GET request for forms you should check your history, maybe it's there look for entries with name of `Sonoff - Save configuration` where `Sonoff` is the configured name of the device.

for example the following link
`http://sp10/co?t1=%7B%22NAME%22%3A%22Generic%22%2C%22GPIO%22%3A%5B23%2C22%2C24%2C17%2C134%2C132%2C0%2C0%2C131%2C52%2C21%2C0%2C0%5D%2C%22FLAG%22%3A0%2C%22BASE%22%3A67%7D&p1=SecretPassword&b1=on&a1=Sonoff&a2=Sonoff2&a3=Sonoff3&a4=Sonoff4&b2=0&save=`
contains the password (`SecretPassword` in this case) for web interface in the `p1` parameter.

Option 4: If you are using `WifiConfig 7` as connection backup tool then you can reset the device when there is no wifi connection available for it.

Option 5: If nothing helps, then you have to erase and reflash the firmware.

Massy1430, 2019-02-25:
Thanks for the answer!
Actually I remember the web admin password, but I don't know the username, I didn't added it anywhere. In "Configure Other" there is just the space for the password. 

Regarding option 2, I have configured mqtt with Home Assistant. From which platform should I send this command? Sorry but I'm a begginner of Tasmota/MQTT.

Thank you

netpok, 2019-02-25:
The username is `admin`

About option 2 you can send it from any mqtt client, you can even use another sonoff device just execute the command `Publish cmnd/<device-topic>/WebPassword 0` in the console.

Massy1430, 2019-02-25:
It worked with the username "admin".

Thanks a lot netpok!!

netpok, 2019-02-25:
Happy to help, please close the issue.

closed 2019-02-25 by Massy1430

meingraham, 2019-02-25:
@netpok I took the liberty of adding this great information to the [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-modified-the-web-admin-password-configure-other-and-now-i-cannot-access-the-web-interface). Please review it to make sure I have captured the information correctly.

Cheers!

Mike

Caswell1000, 2020-01-02:
> You set up a password for the webinterface, you can login with the username admin and the password you entered.
>
> If you don't remember that password:
> **Option 1:** You have serial connection to the [...]

Great answer, managed to get it via history - Option 3