arendst/Tasmota #5338: DHT22 sensor on Wemos D1 mini reports data only until restart

akasma74, 2019-02-25:
**Describe the bug**
I flashed Wemos D1 mini board with Sonoff-Tasmota v.6.4.1 compiled according to [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos%20D1%20Mini)
I get correct readings from DHT22 sensor attached to D4 pin until the board is restarted somehow: Reset button, clicking Restart in Web interface, changing some parameters there and clicking Save changes..
Then it shows Temperature null C, Humidity null%
The only way to get the readings back is to disconnect the board from power and reconnect it back.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Wemos D1 mini, DHT22 temperature/humidity sensor
- [x] _Development/Compiler/Upload tools used_ : pio
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
5:12:24 CMD: status 0
15:12:24 MQT: stat/shower_fan_sensor/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"shower_fan_sensor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:12:24 MQT: stat/shower_fan_sensor/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:13:55","StartupUTC":"2019-02-25T13:58:29","Sleep":50,"BootCount":18,"SaveCount":34,"SaveAddress":"FA000"}}
15:12:24 MQT: stat/shower_fan_sensor/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2019-02-14T11:44:45","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
15:12:24 MQT: stat/shower_fan_sensor/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Level2",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
15:12:24 MQT: stat/shower_fan_sensor/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FDAE794","040383A0","23B617CE","00003BC0"]}}
15:12:24 MQT: stat/shower_fan_sensor/STATUS5 = {"StatusNET":{"Hostname":"shower_fan_sensor-4650","IPAddress":"192.168.0.53","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"194.168.4.100","Mac":"68:C6:3A:E9:F2:2A","Webserver":2,"WifiConfig":4}}
15:12:24 MQT: stat/shower_fan_sensor/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.34","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E9F22A","MqttUser":"mqtt","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:12:24 MQT: stat/shower_fan_sensor/STATUS7 = {"StatusTIM":{"UTC":"Mon Feb 25 14:12:24 2019","Local":"Mon Feb 25 15:12:24 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:40","Sunset":"18:26"}}
15:12:24 MQT: stat/shower_fan_sensor/STATUS10 = {"StatusSNS":{"Time":"2019-02-25T15:12:24","AM2301":{"Temperature":null,"Humidity":null},"TempUnit":"C"}}
15:12:24 MQT: stat/shower_fan_sensor/STATUS11 = {"StatusSTS":{"Time":"2019-02-25T15:12:24","Uptime":"0T00:13:55","Vcc":3.000,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Level2","BSSId":"E8:94:F6:86:4D:96","Channel":11,"RSSI":100}}}
```

**To Reproduce**
1. Flash the board with modified platform.ini as per [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos%20D1%20Mini)
2. Go to device's webpage and configure module: generic(18)
3. After the device reboots, go to configure module and set D4 GPIO2 to AM2301 (2)
4. After the device reboots, you'll see temperature/humidity readings on the main page
5. Click Restart
6. The readings change to nulls

Jason2866, 2019-02-25:
Cant reproduce. Your selfcompiled firmware seems to be faulty.
Try precompiled firmware from http://thehackbox.org/tasmota/release/

wongnam, 2019-02-26:
@akasma74 Try to use D1, D2, D5, D6, D7. don't user D4(it is connected to the onboard LED) as it may be interference by on board LED.

Please read Wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs#usable-pins

akasma74, 2019-02-26:
Thanks @wongnam, I missed that bit, perhaps because it's really confusing when you have pin numbers, GPIO numbers etc.
Switching to D1 does the trick.
Just out of curiosity, I have a circuit that has high potential when something happens, and I want to use it as a button. Is that true that only GPIO 16 (D0) can be used in that case? And is it enough to configure that pin as a button?

closed 2019-02-26 by akasma74