arendst/Tasmota #5339: enabeling mDSN (SetOption55 1) will shout DNS: Error on the console after some minutes.

markusb68, 2019-02-25:
If I issue SetOption55 1 on the console using 6.4.1.18 I get DNS: ERROR in the console every second.
But not directly after sending the command, just after some minutes. Anybody else too?
After sending SetOption55 0 and a reboot the DNS: ERROR is gone again.

localhost61, 2019-02-25:
No issue here, but you give very little details on your setup, to try to reproduce.
Especially:
- What firmware flavor? what core?
- Did you compile yourself, did you try to enable the option in my_user_config.h?
- What is your DNS server? Did you try an other one?

markusb68, 2019-02-26:
Hi, here's the information:
Program Version | 6.4.1.18(sonoff)
Core: 2_5_0/3.0.0-dev(c0f7b44)
Self compiled, have not enabled the option in _my_user_config.h:
#define MDNS_ENABLED           0 
Just in the console.
The DNS is a Pihole behind a Fritzbox Fritz.OS 7.01

localhost61, 2019-02-26:
Same configuration here, but the DNS is the one of my ADSL box.
As you master the DNS you may be able to have a look at it's log

hhalibo, 2019-02-26:
Pls build firmware with  arduino-esp8266-2.4.2 core, I have some sonoff devices, with 2.4.2 is very stable. But use the firmware with 2.5.0-core, running after some minutes, the "DNS: FAILED" showing on the console !

localhost61, 2019-02-27:
Arghh! It happens to me too now!

Was doing several tests yesterday, experimenting mDNS with changing core, looking if mDNS name could be used for MQTT server and/or NTP local server... For the former: Yes on 2.4.2, 2.5.0 and Stage, (provided that .local is appended to the name, here -> PiStash.local ). But for the later: No, I still need to remember the IP address of my Gateway/NTP server.
Tried to compile with mDNS enabled in my_user_config as I suggested above, but as it's not a fresh install, the config was not erased.
Then I let it go one full day a here it is !!!
```
01:17:23 DNS: Échoué
01:17:24 DNS: Échoué
01:17:24 CMD: status 0
01:17:24 MQT: stat/tasmota/RdC/Sjr/003/shutter/STATUS = {"Status":{"Module":39,"FriendlyName":["VR 03 Sjr.2","VR_133_2"],"Topic":"tasmota/RdC/Sjr/003/shutter","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":0,"SwitchTopic":"tasmota/RdC/Sjr/003/shutter","SwitchMode":[2,2,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
01:17:25 MQT: stat/tasmota/RdC/Sjr/003/shutter/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"Pirouet","OtaUrl":"http://192.168.61.129:5000/sonoff-FR.bin","RestartReason":"Software/System restart","Uptime":"0T23:17:40","StartupUTC":"2019-02-26T00:59:44","Sleep":50,"CfgHolder":4617,"BootCount":46,"SaveCount":103,"SaveAddress":"FB000"}}
01:17:25 MQT: stat/tasmota/RdC/Sjr/003/shutter/STATUS2 = {"StatusFWR":{"Version":"6.4.1.19(sonoff)","BuildDateTime":"2019-02-26T01:54:49","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
01:17:25 MQT: stat/tasmota/RdC/Sjr/003/shutter/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"PiStash","LogPort":514,"SSId":["WhoKnows",""],"TelePeriod":300,"Resolution":"418180C0","SetOption":["0400C008","280500000100000000000000000000000000","00000421"]}}
01:17:25 MQT: stat/tasmota/RdC/Sjr/003/shutter/STATUS4 = {"StatusMEM":{"ProgramSize":482,"Free":520,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["0000040C","0F882590","04000001","040000C6","000000C0"]}}
01:17:25 MQT: stat/tasmota/RdC/Sjr/003/shutter/STATUS5 = {"StatusNET":{"Hostname":"Dual-03","IPAddress":"192.168.61.003","Gateway":"192.168.61.128","Subnetmask":"255.255.255.0","DNSServer":"192.168.61.128","Mac":"DC:4F:00:00:00:00","Webserver":2,"WifiConfig":4}}
01:17:25 MQT: stat/tasmota/RdC/Sjr/003/shutter/STATUS6 = {"StatusMQT":{"MqttHost":"PiStash.local","MqttPort":1883,"MqttClientMask":"DVES_%04X","MqttClient":"DVES_8036D3D3","MqttUser":"PiLot","MqttType":1,"MqttCount":91,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
01:17:25 MQT: stat/tasmota/RdC/Sjr/003/shutter/STATUS7 = {"StatusTIM":{"UTC":"Wed Feb 27 00:17:25 2019","Local":"Wed Feb 27 01:17:25 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:52","Sunset":"18:47"}}
01:17:25 MQT: stat/tasmota/RdC/Sjr/003/shutter/STATUS10 = {"StatusSNS":{"Time":"2019-02-27T01:17:25","Switch1":"OFF","Switch2":"OFF","DS18B20":{"Temperature":18.4},"TempUnit":"C"}}
01:17:25 MQT: stat/tasmota/RdC/Sjr/003/shutter/STATUS11 = {"StatusSTS":{"Time":"2019-02-27T01:17:25","Uptime":"0T23:17:41","Vcc":3.465,"SleepMode":"Normal","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"WhoKnows","BSSId":"00:24:00:00:00:00","Channel":9,"RSSI":74,"LinkCount":66,"Downtime":"0T00:03:32"}}}
01:17:25 DNS: Échoué
01:17:26 DNS: Échoué
01:17:27 DNS: Échoué
. . .
01:18:44 DNS: Échoué
01:18:45 CMD: setoption55
01:18:45 MQT: stat/tasmota/RdC/Sjr/003/shutter/RESULT = {"SetOption55":"ON"}
01:18:45 DNS: Échoué
```
So 
- The message happens currently every second
- I doesn't prevent MQTT to work

_Nota: Addresses an devices name are fake_

localhost61, 2019-02-27:
Wait,
- I did setoption55 0 on the console and the device restarted
- I did setoption55 and the device confirms OFF
Then **WHY** do I still connect to my MQTT server on PiStash.local ???

Jason2866, 2019-02-27:
mDNS has -different- issues in all Arduino Cores. This is not under control of Tasmota.
Solution simply dont use it...

Rutaabed, 2019-03-01:
I had this problem when i updated the ESP core to 2.5. I fixed the problem with removing the ESP Library v2.5 and reinstalled the ESP core version 2.4.2 that more stable than the last version. 
Now the module is connected with more than one week. i don't see the problem again ...

localhost61, 2019-03-02:
I just discovered that my Gateway was configured with another name for mDNS, that for DNS and Netbios. It was _Freebox-serveur_ instead of _Freebox_. So I changed it back into _Freebox_ and now I can configure NTP server with _Freebox.local_ and it works too... with `SetOption55 Off` and core 2.5.0.
- Save config / Reset 1 / Restaure config / Check that SetOption55 Off 
-> LinkCount = MqttCount = 1 & Downtime=4s
I'll see what I get after several hours.

ascillato2, 2019-03-06:
Hi,

> The DNS is a Pihole behind a Fritzbox Fritz.OS 7.01

Sorry, Fritzbox is known to have issues with the arduino core. Please, try to use Tasmota with core 2.4.2 or 2.3.0

You can find the precompiled versions at thehackbox.org/tasmota

Thanks and sorry for the inconvenience. It is outside Tasmota and out of our control. See wiki at troubleshooting for the full explanation of the issues of the core and the SDK (closed source)

closed 2019-03-06 by ascillato2

markusb68, 2019-03-06:
Sorry, but the wiki says:
2.5.0: Most Routers of brands Ubiquity and Fritzbox don't produces conflicts or disconnections
and for 2.3 and 2.4.2: Some Routers of brands Ubiquity and Fritzbox produces conflicts and disconnections

So going back to 2.4.2 or 2.3 is not a good idea.
I'm NOT having disconnections! Just the mDSN Problem!

joba-1, 2019-03-06:
to add a piece of related info:

I have 2 fritzbox routers meshed and experience mDNS issues as well.

I have not yet verified this, but it could be related to using WiFiManager
for Wifi setup:
I don't remember any of my (non-tasmota) projects where I setup wifi myself
had problems with mDNS, but one where I now use WiFiManager has.

-- 
iphone so i typo

ascillato, 2019-03-06:
@markusb68 
@joba-1 

Hi, 

Please try the precompiled bin with core 2.3.0 :

http://thehackbox.org/tasmota/020300/sonoff.bin

If that don't solve your issue, there is another option using the stage core (the latest version of the core in the development branch of the Arduino esp8266 core). We have received reports of several issues solved in this edge version.

markusb68, 2019-03-06:
Tried the 2.6 dev. but Project does not compile with it:

```
Arduino: 1.8.1 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Disabled, nodemcu, 26 MHz, 40MHz, DOUT (compatible), 1M (no SPIFFS), 2, nonos-sdk 2.2.1, v2 Higher Bandwidth, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\xx\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:67:0,

                 from C:\Users\xx\Documents\Arduino\sonoff\xplg_ws2812.ino:25:

C:\Users\xx\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoEsp8266DmaMethod.h:51:127: error: declaration of C function 'void rom_i2c_writeReg_Mask(uint32_t, uint32_t, uint32_t, uint32_t, uint32_t, uint32_t)' conflicts with

     void rom_i2c_writeReg_Mask(uint32_t block, uint32_t host_id, uint32_t reg_add, uint32_t Msb, uint32_t Lsb, uint32_t indata);

                                                                                                                               ^

In file included from C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/esp8266_peri.h:25:0,

                 from C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/Arduino.h:38,

                 from sketch\sonoff.ino.cpp:1:

C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/esp8266_undocumented.h:9:13: error: previous declaration 'void rom_i2c_writeReg_Mask(int, int, int, int, int, int)' here

 extern void rom_i2c_writeReg_Mask(int, int, int, int, int, int);

             ^

exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 Module.
```

markusb68, 2019-03-06:
PS: esp8266\2.5.0\ ... is just the folder name, but it's the git clone from today (06.03.19 12:15)

ascillato, 2019-03-06:
You need to delete the folder, download again and then run the script to download all the toolchain. The exact instructions are in the readme of the Arduino esp8266 core repository.

markusb68, 2019-03-06:
I did: Deleded Folder, downloaded again. and ran:
c:\Users\xx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\tools\get.py
It compiles when #define FIRMWARE_BASIC is set and it compiles other projects.
But it doesn't compile when //#define FIRMWARE_BASIC is commented out.

Have you tried the latest github master of 2.6 Dev ?

markusb68, 2019-03-06:
No, it's the one out of the tasmota lib folder.
I have copied all files from
Sonoff-Tasmota-development.zip:
\lib\NeoPixelBus-2.2.9
to:
c:\Users\xx\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\ folder.

Compared, files are identical.
Have you tried the latest github master of 2.6 Dev to build a generic build?

ascillato, 2019-03-06:
Have you tried the latest github master of 2.6 Dev to build a generic build?

yes we did, and we are trying to help you.

ascillato, 2019-03-06:
Can you delete all the Tasmota Folder and Libraries and download it again. If you copy over an old library, some old files (that were renamed) could be still there making conflict.

markusb68, 2019-03-06:
I never copied over an old lib. always remove all old files.
I did it again. Checked that neopixel lib files are the same as in lib folder of Tasmota zip.
But still the same:
```

Arduino: 1.8.1 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Disabled, nodemcu, 26 MHz, 40MHz, DOUT (compatible), 1M (no SPIFFS), 2, nonos-sdk 2.2.1, v2 Higher Bandwidth, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\xx\Documents\Arduino\libraries\NeoPixelBus-2.2.9\src/NeoPixelBus.h:67:0,

                 from C:\Users\xx\Documents\Arduino\sonoff\xplg_ws2812.ino:25:

C:\Users\xx\Documents\Arduino\libraries\NeoPixelBus-2.2.9\src/internal/NeoEsp8266DmaMethod.h:51:127: error: declaration of C function 'void rom_i2c_writeReg_Mask(uint32_t, uint32_t, uint32_t, uint32_t, uint32_t, uint32_t)' conflicts with

     void rom_i2c_writeReg_Mask(uint32_t block, uint32_t host_id, uint32_t reg_add, uint32_t Msb, uint32_t Lsb, uint32_t indata);

                                                                                                                               ^

In file included from C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/esp8266_peri.h:25:0,

                 from C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/Arduino.h:38,

                 from sketch\sonoff.ino.cpp:1:

C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/esp8266_undocumented.h:9:13: error: previous declaration 'void rom_i2c_writeReg_Mask(int, int, int, int, int, int)' here

 extern void rom_i2c_writeReg_Mask(int, int, int, int, int, int);

             ^

exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 Module.

```

arendst, 2019-03-06:
See https://github.com/Makuna/NeoPixelBus/issues/252

markusb68, 2019-03-06:
Thanks. It compiled now.
I know, that it has no relation with mDNS issue.
No I'll test mDNS with latest ESP Core 2.6 DEV.

markusb68, 2019-03-07:
Tested mDNS with latest ESP Core 2.6 DEV:
After SetOption55 1 DNS: Failed is gone.

BUT latest ESP Core 2.6 DEV seems to have severe issues with mDNS.
Using zeroconfServiceBrowser one could discover the ESP's as "WWW HTTP" but there is nothing now.
using older cores the anouncement of mDSN services worked.

markusb68, 2019-03-07:
And Still: DNS: Failed. So I'll give up using mDNS.

KDMcMullan, 2020-01-04:
I don't see any stability in the DNS features. For example, I have v6.7.1 core 2.2.2-dev with setoptin55 on in a Sonoff  Basic and it works perfectly in my airing cupboard. Same firmware in (ostensibly) the same hardware, and it doesn't work in my cloakroom. I suspect they are connecting to two different Access Points. This makes me think the inconsistency is between the operation of various router types.

localhost61, 2020-01-05:
@KDMcMullan as  you don't share your setup we can't help you to find the reason why.
You may try to disable 5GHz band on your AP to improve stability.
All that I can say is that I'm running v8.102 on all my devices, connected to 3 very different Access-Point (the french Freebox ADSL gateway, a Cisco AP and a chinese powerline extender Tenda PA6/PH10) and sharing the same SSID on 3 different channels... 
mDNS works really great with SetOption55 ON with currently 29 devices online.

1) NTPserver "Freebox.local"
2) MQTThost "RPi3B.local"
3) OTAURL "http://RPi3B.local:5000/tasmota.bin"
and Gateway IP, DNS server are obtained through DHCP

KDMcMullan, 2020-01-05:
It was a statement than a request for help. Something for the OP to b aware of.

I simply state that regardless of the hardware or location, sometimes the DNS fail message is present, sometimes it's not.

I'm using a Sagemcom FAST5364 router. I have tried 2 different hardware versions of S20, a T1 (EU version), a Sonoff BASIC, several SOnoff Duals, a NodeMCU. Usually with v6.7.1 software.

It doesn't happen with v7.2 or 8.1, however they have their own problems. It takes ages for them to connect and the connection is very unstable.