arendst/Tasmota #5341: Add command to set MQTT keep alive time

ronschaeffer, 2019-02-25:
Cannot find a mention of keep alive in Commands section of docs. In https://github.com/arendst/Sonoff-Tasmota/issues/3940 it is stated that keep alive tiem is currently 15 seconds.

Please add a command to set MQTT keep alive time to any chosen value.

Jason2866, 2019-02-25:
For what use case is this needed?

arendst, 2019-02-25:
Won't work without a recompile as it is PubSubClient library related.

Jason2866, 2019-02-25:
You can change it to your needs in https://github.com/arendst/Sonoff-Tasmota/blob/development/lib/PubSubClient-EspEasy-2.6.09/src/PubSubClient.h
and compile your version

ronschaeffer, 2019-02-25:
> For what use case is this needed?

Apologies. I should have included the use case.

I using a a number of Sonoff basics attached to the lighting ring to determine the on/off status of room lights which are controlled by 433MHz switches which do not themselves provide status feedback. So, HA is unaware of manual on/off the switch. HA automations watch for ...LWT/Online and ...LWT/Offline messages and to update the status in the HA GUI accordingly. It would be preferable to have the Off (Offline) status reflected in HA sooner than ~15 seconds after the light is switched off at the wall.

ronschaeffer, 2019-02-25:
> You can change it to your needs in https://github.com/arendst/Sonoff-Tasmota/blob/development/lib/PubSubClient-EspEasy-2.6.09/src/PubSubClient.h
> and compile your version

Thanks. I will check that out.

Longer term, could the keep alive setting be made modifiable by a command? Note, for example, that keep alive time is one of the possible settings of the Shelly 1's built-in MQTT support.

Jason2866, 2019-02-25:
No, since this setting is fixed at compile time. Already explained from project owner arendst before
Please close this feature request because it will / cannot be done.

ayavilevich, 2020-11-04:
@Jason2866 @arendst 

Version 2.8 of PubSubClient adds
https://pubsubclient.knolleary.net/api#setKeepAlive
https://github.com/knolleary/pubsubclient/blob/2d228f2f862a95846c65a8518c79f48dfc8f188c/CHANGES.txt

So this can now be implemented.

An important feature, as the current 30s default fits internet based devices but is too long for LAN devices.

Jason2866, 2020-11-04:
Tasmota use a modified tweaked version of PubSubClient  (PubSubClient-EspEasy)
The current version works very stable and reliable. It was a long way to get there...
Since mqtt is the heart of Tasmota and any small change can have unknow side effects
there will no change made for a feature that is seldom needed.

s-hadinger, 2020-11-04:
Can I ask why is 30s keep alive too long?

You can always change the value and recompile your own version with a shorter value but I don't see why you would need it.

ayavilevich, 2020-11-05:
@s-hadinger I want to create an automation that runs as soon as the device becomes unavailable. Currently it detects it as unavailable after 30 seconds due to the keep alive.
Is there another way to do what I want with Tasmota?

I know I can recompile. It is just not scalable.

I guess at some point you will move to PubSubClient 2.8, right. It is at PubSubClient-EspEasy-2.7.12 now. Not too far away version wise. Where does the "PubSubClient-EspEasy" version come from?

buteomont, 2020-12-14:
I too need this feature for similar reasons, using the LWT to establish a power problem.  The built-in value in my case is too short, as I keep getting false indications of a power outage.

ascillato, 2020-12-14:
You should check your mqtt broker logs. The default time in the library is more than enough for the broker. Seems that you are having another issue. May be mqtt or wifi disconnections?

buteomont, 2020-12-14:
My apologies for not knowing this, but where may I find the logs?  I checked the console and only see the last 18 or so messages that were sent.
I also posted a better description of my problem on the Discussions tab.  I hope that's not the wrong place to do it.

buteomont, 2020-12-14:
Sorry, I missed the "broker" part of "broker logs".  I will check those. Thanks

ascillato, 2020-12-14:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

namadori, 2021-03-09:
Hello, in the PubSubClient-EspEasy-2.7.12 documentation it is stated that the KeepAlive is 15 seconds. May I ask why now Tasmota overrides that value with 30 seconds? 
For the typical use cases on a LAN I think 15 seconds will be better for many scenarios that involve the detection of the shutdown of the device power.

https://github.com/arendst/Tasmota/blob/development/lib/default/PubSubClient-EspEasy-2.7.12/README.md

Jason2866, 2021-03-10:
For connecting to extern mqtt brokers with TLS 15 seconds was to short.

buteomont, 2021-03-10:
This is why it should be a configuration option.

namadori, 2021-03-10:
I agree it would be a useful configuration, but as this is an open source project I think we can only "kindly ask" the developers that are already working for free to give us this great software.

I would say, please if you are going to update with the upstream PubSubClient library, take in consideration that exposing the configuration will be a welcome addition.

Thanks.

closed 2021-03-18 by close-issue-app[bot]

namadori, 2021-03-19:
Just saw the new MqttKeepAlive command in 9.3.1.2. A huge thank you to @arendst for this great software. :thumbsup: :smile:

timaseth, 2021-04-08:
Theo and his DevTeam are awesome, and that is the least I can say. The countless days and hours of pleasure and comfort your code has provided us @arendst - God bless!