arendst/Tasmota #5343: Device is loses configuration

stephan1980, 2019-02-25:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Arduino IDE 1.8.5 Mac OSX
- [x] Hardware used : Amica Node MCU
- [x] If a pre-compiled release or development binary was used, which one? : sonoff-sensors.bin
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
 23:52:09 CMD: Status 0
23:52:09 MQT: SmartHome/Wettersensor/STATUS = {"Status":{"Module":18,"FriendlyName":["WettersensorV2"],"Topic":"Wettersensor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:52:09 MQT: SmartHome/Wettersensor/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Blocked Loop","Uptime":"0T13:30:24","StartupUTC":"2019-02-25T09:21:45","Sleep":50,"CfgHolder":4617,"BootCount":36,"SaveCount":67,"SaveAddress":"F9000"}}
23:52:09 MQT: SmartHome/Wettersensor/STATUS2 = {"StatusFWR":{"Version":"6.4.1.17(b4b1ee9-sensors)","BuildDateTime":"2019-02-14T18:04:56","Boot":6,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
23:52:09 MQT: SmartHome/Wettersensor/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Solaris",""],"TelePeriod":120,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
23:52:09 MQT: SmartHome/Wettersensor/STATUS4 = {"StatusMEM":{"ProgramSize":561,"Free":440,"Heap":16,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashMode":2,"Features":["00000809","0FDEE794","0007A3A4","B7FFBFCC","001ABBC0"]}}
23:52:09 MQT: SmartHome/Wettersensor/STATUS5 = {"StatusNET":{"Hostname":"Wettersensor-2952","IPAddress":"192.168.60.155","Gateway":"192.168.60.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.60.1","Mac":"EC:FA:BC:05:8B:88","Webserver":2,"WifiConfig":4}}
23:52:09 MQT: SmartHome/Wettersensor/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.60.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_058B88","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:52:09 MQT: SmartHome/Wettersensor/STATUS7 = {"StatusTIM":{"UTC":"Mon Feb 25 22:52:09 2019","Local":"Mon Feb 25 23:52:09 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:40","Sunset":"18:26"}}
23:52:09 MQT: SmartHome/Wettersensor/STATUS10 = {"StatusSNS":{"Time":"2019-02-25T23:52:09","ANALOG":{"A0":1}}}
23:52:09 MQT: SmartHome/Wettersensor/STATUS11 = {"StatusSTS":{"Time":"2019-02-25T23:52:09","Uptime":"0T13:30:24","SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"Wifi":{"AP":1,"SSId":"Solaris","BSSId":"44:4E:6D:63:34:8C","Channel":1,"RSSI":80}}}
23:52:13 MQT: SmartHome/Wettersensor/STATE = {"Time":"2019-02-25T23:52:13","Uptime":"0T13:30:28","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Solaris","BSSId":"44:4E:6D:63:34:8C","Channel":1,"RSSI":82}}
23:52:13 MQT: SmartHome/Wettersensor/SENSOR = {"Time":"2019-02-25T23:52:13","ANALOG":{"A0":1}}
```

My weather sensor including both a TSL2561 and a BME280 is working fine until the device somehow  performs a restart. I am using a precompiled binary (sonoff-sensors.bin). After the restart all sensor configuration is gone (GPIO mapping for SDA and SCL). 
Funnily enough the device seems to do the restart every morning at around 10:20am. That is what I get in the logfile after the restart:

```
00:00:00 Project sonoff WettersensorV2 Version 6.4.1.17(b4b1ee9-sensors)-2_4_2
00:00:00 WIF: Connecting to AP1 Solaris in mode 11N as Wettersensor-2952...
00:00:08 WIF: Connected
00:00:08 HTP: Web server active on Wettersensor-2952 with IP address 192.168.60.155
10:21:55 MQT: Attempting connection...
10:21:55 MQT: Connected
10:21:55 MQT: SmartHome/Wettersensor/LWT = Online (retained)
10:21:55 MQT: SmartHome/Wettersensor/cmnd/POWER = 
10:21:55 MQT: SmartHome/Wettersensor/INFO1 = {"Module":"Generic","Version":"6.4.1.17(b4b1ee9-sensors)","FallbackTopic":"cmnd/DVES_058B88_fb/","GroupTopic":"sonoffs"}
10:21:55 MQT: SmartHome/Wettersensor/INFO2 = {"WebServerMode":"Admin","Hostname":"Wettersensor-2952","IPAddress":"192.168.60.155"}
10:21:55 MQT: SmartHome/Wettersensor/INFO3 = {"RestartReason":"Blocked Loop"}
10:22:03 MQT: SmartHome/Wettersensor/STATE = {"Time":"2019-02-25T10:22:03","Uptime":"0T00:00:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Solaris","BSSId":"44:4E:6D:63:34:8C","Channel":1,"RSSI":66}}
10:22:03 MQT: SmartHome/Wettersensor/SENSOR = {"Time":"2019-02-25T10:22:03","ANALOG":{"A0":1}}
10:24:03 MQT: SmartHome/Wettersensor/STATE = {"Time":"2019-02-25T10:24:03","Uptime":"0T00:02:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Solaris","BSSId":"44:4E:6D:63:34:8C","Channel":1,"RSSI":80}}
10:24:03 MQT: SmartHome/Wettersensor/SENSOR = {"Time":"2019-02-25T10:24:03","ANALOG":{"A0":1}}
10:26:03 MQT: SmartHome/Wettersensor/STATE = {"Time":"2019-02-25T10:26:03","Uptime":"0T00:04:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Solaris","BSSId":"44:4E:6D:63:34:8C","Channel":1,"RSSI":78}}
10:26:03 MQT: SmartHome/Wettersensor/SENSOR = {"Time":"2019-02-25T10:26:03","ANALOG":{"A0":1}}
10:28:03 MQT: SmartHome/Wettersensor/STATE = {"Time":"2019-02-25T10:28:03","Uptime":"0T00:06:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Solaris","BSSId":"44:4E:6D:63:34:8C","Channel":1,"RSSI":84}}
10:28:03 MQT: SmartHome/Wettersensor/SENSOR = {"Time":"2019-02-25T10:28:03","ANALOG":{"A0":1}}
10:30:03 MQT: SmartHome/Wettersensor/STATE = {"Time":"2019-02-25T10:30:03","Uptime":"0T00:08:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Solaris","BSSId":"44:4E:6D:63:34:8C","Channel":1,"RSSI":80}}
10:30:03 MQT: SmartHome/Wettersensor/SENSOR = {"Time":"2019-02-25T10:30:03","ANALOG":{"A0":1}}
10:31:45 APP: Serial logging disabled
```

I am totally confused because the devices worked for some days without any restart.... The problem occurred suddenly.

After reconfiguring the device I get what I want to get:

```
00:00:00 Project sonoff WettersensorV2 Version 6.4.1.17(b4b1ee9-sensors)-2_4_2
00:00:00 WIF: Connecting to AP1 Solaris in mode 11N as Wettersensor-2952...
00:00:11 WIF: Connected
00:00:11 HTP: Web server active on Wettersensor-2952 with IP address 192.168.60.155
00:01:43 MQT: Attempting connection...
00:01:43 MQT: Connected
00:01:43 MQT: SmartHome/Wettersensor/LWT = Online (retained)
00:01:43 MQT: SmartHome/Wettersensor/cmnd/POWER = 
00:01:43 MQT: SmartHome/Wettersensor/INFO1 = {"Module":"Generic","Version":"6.4.1.17(b4b1ee9-sensors)","FallbackTopic":"cmnd/DVES_058B88_fb/","GroupTopic":"sonoffs"}
00:01:43 MQT: SmartHome/Wettersensor/INFO2 = {"WebServerMode":"Admin","Hostname":"Wettersensor-2952","IPAddress":"192.168.60.155"}
00:01:43 MQT: SmartHome/Wettersensor/INFO3 = {"RestartReason":"Software/System restart"}
00:02:01 MQT: SmartHome/Wettersensor/UPTIME = {"Time":"2019-02-26T00:02:01","Uptime":"0T00:00:22"}
00:02:10 MQT: SmartHome/Wettersensor/STATE = {"Time":"2019-02-26T00:02:10","Uptime":"0T00:00:31","SleepMode":"Dynamic","Sleep":50,"LoadAvg":116,"Wifi":{"AP":1,"SSId":"Solaris","BSSId":"44:4E:6D:63:34:8C","Channel":1,"RSSI":78}}
00:02:10 MQT: SmartHome/Wettersensor/SENSOR = {"Time":"2019-02-26T00:02:10","ANALOG":{"A0":1},"BME280":{"Temperature":4.2,"Humidity":69.4,"Pressure":1033.9},"TSL2561":{"Illuminance":0.024},"PressureUnit":"hPa","TempUnit":"C"}
```

Maybe someone could give me a hint?!

Jason2866, 2019-02-26:
Use latest dev build from http://thehackbox.org/tasmota/sonoff-sensors.bin
There where problems with WDT resets (introduced with 6.4.1.16). Solved in 6.4.1.18

stephan1980, 2019-02-26:
Thank you, I have uploaded the latest bin. If everything is working fine I will give you feedback!

ascillato2, 2019-02-27:
Hi,

Any news on this?

stephan1980, 2019-02-27:
Unfortunately the same problem with 6.4.1.19.
After uploading everything worked fine for some hours. This morning at 10:30am the devices made a restart because of a blocked loop and again lost all GPIO configuration

ascillato, 2019-02-27:
Let's try something else. Try to do a reset 5 in the Tasmota console. That will erase all settings but will leave the wifi credentials.

If that does not solve your issue, try erase all flash with esptool.py as explained in the wiki and flash again with the precompiled bins from development branch.

stephan1980, 2019-02-28:
I am currently not at home. I am able to try after coming weekend.

agsochi, 2019-03-01:
Is it related to this issue? https://github.com/arendst/Sonoff-Tasmota/issues/5323#issuecomment-468575163

stephan1980, 2019-03-05:
Back home I tried both: Reset 5 and complete flash erasing. Only erasing the complete flash worked for me.

ascillato2, 2019-03-06:
Hi,

Great to know that erasing solve your issue.

Closing this issue as it has been solved.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-06 by ascillato2

pesone, 2019-04-18:
> Unfortunately the same problem with 6.4.1.19.
> After uploading everything worked fine for some hours. This morning at 10:30am the devices made a restart because of a blocked loop and again lost all GPIO configuration

I'm pretty sure that this behaviour is related to the TSL2561.
I can reproduce it  a 100%.

If lux goes any higher about  ~30000 
(this was the highest value i've mesured so far and i have no idea how i could get the actual value. It's for sure that it must be a sunny day;))
the device(webinterface/teleperiod/..) freezes immediatly 
and as soon i put the sensor to shadow it starts reporting again.
If it's too long exposed to values > 30000 you get blocked loop with all configuration

Tested with Wemos D1-Mini & Sonoff Basic 
6.5.0 and 6.4.x (2.5.0 and 2.3.0)

joba-1, 2019-04-18:
This happened with my tsl2561 lib version 2.0.7
It was fixed with 2.0.8.
Current version is 2.0.10.

Sounds like tasmota still uses the old version.

arendst, 2019-04-18:
I'll update to 2.0.10 now

arendst, 2019-04-18:
Library updated. Should be in Hackbox after two hours.

pesone, 2019-04-18:
17:26:50 MQT: stat/Uvlux/STATUS10 = {"StatusSNS":{"Time":"2019-04-18T17:26:50","VEML6070":{"UvLevel":1968,"UvIndex":2.63,"UvIndexText":"Niedrig","UvPower":0.066},"TSL2561":{"Illuminance":114500.000}}}

114500 Lux
i will report tomorrow if it's stable  

thx

pesone, 2019-04-19:
```
09:48:57 MQT: tele/Uvlux/STATE = {"Time":"2019-04-19T09:48:57","Uptime":"0T16:24:42","Vcc":2.993,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"KBW1HAB","BSSId":"44:4E:6D:0F:B7:F5","Channel":11,"RSSI":54,"LinkCount":13,"Downtime":"0T00:00:53"}}
09:48:57 MQT: tele/Uvlux/SENSOR = {"Time":"2019-04-19T09:48:57","VEML6070":{"UvLevel":642,"UvIndex":0.86,"UvIndexText":"Niedrig","UvPower":0.021},"TSL2561":{"Illuminance":118300.000}}
```
thumbs up

vel2000, 2019-05-05:
> I'm pretty sure that this behaviour is related to the TSL2561.
>

Nope, unfortunately its not only related to the TSL2561
Happened to me today after a power outage.
Device type has than changed from "generic" to "sonoff-basic" and all gpio mappings (SCL/SDA) were gone.
My sensor is a SHT31-D.
Lolin V3, Firmware Tasmota 6.5.0.9