arendst/Tasmota #5345: Relay triggering on power cycle

zymbiosis, 2019-02-26:
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : Arduino IDE
- [X] Hardware used : ESP8266 relay module (https://www.aliexpress.com/item/ESP8266-ESP-01S-5V-WiFi-Relay-Module-Things-Smart-Home-Remote-Control-Switch-for-Arduino-Phone/32890526507.html)
- [ ] If a pre-compiled release or development binary was used, which one? :
- [X] You have tried latest release or development binaries? :
- [X] Provide the output of command ``status 0`` :
```
22:13:10 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["GarageOpener"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:13:10 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:08:19","StartupUTC":"2019-02-26T03:04:51","Sleep":50,"BootCount":28,"SaveCount":64,"SaveAddress":"F4000"}}
22:13:10 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2019-02-25T21:06:25","Boot":6,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
22:13:10 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Symbiosis",""],"TelePeriod":300,"SetOption":["00008008","55C180C0","00000000"]}}
22:13:10 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":563,"Free":440,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
22:13:10 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3195","IPAddress":"192.168.1.106","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.21","Mac":"BC:DD:C2:23:EC:7B","Webserver":2,"WifiConfig":4}}
22:13:10 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_23EC7B7B","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:13:10 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 26 03:13:10 2019","Local":"Mon Feb 25 22:13:10 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"-05:00","Sunrise":"01:40","Sunset":"12:26"}}
22:13:10 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-02-25T22:13:10"}}
22:13:10 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-02-25T22:13:10","Uptime":"0T00:08:19","Vcc":3.482,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":91,"POWER":"Off","Wifi":{"AP":1,"SSId":"-----------","BSSId":"------------","Channel":1,"RSSI":100}}}
```

Hello!  I am attempting to use this generic ESP8266 relay module as a garage opener.  I have tasmota configured as a generic module with GPIO0 set as Relay1i.  Pulsewidth is configured so when I click the button, the relay toggles on then off.  This all works great!  However, when there is a power failure, the relay triggers momentarily immediately after regaining power.  Unfortunately the logs show nothing (even with more debug) in regards to power cycling.  If I set the PowerOnState to 1 it triggers twice, so it almost seems like the relay is triggering before tasmota is fully powered up.  Thoughts?

Thanks for your help!

wongnam, 2019-02-26:
don't use GPIO 0 in this case. i suggest try to use GPIO 5 or something else usable pin,  If you want to know why, Please read Wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs#usable-pins

netpok, 2019-02-26:
Use a pin which not affects the boot sequence as mentioned in the previous comment and use a pull-up resistor on the control line.

zymbiosis, 2019-02-26:
Thanks for the suggestions.  Unfortunately GPIO0 is hardwired to the relay, so using a different GPIO is out of the question.  I will attempt to add an external pull-up (hopefully without damaging any of the board) and see if that helps.

Any other suggestions besides hardware modification?  Much appreciated

netpok, 2019-02-26:
You can try `poweronstate 0`, but if the impulse happens before tasmota boot than your out of luck, you should test the circuit itself by applying power while the esp is removed and the gpio0 is shorted to vcc in the socket.

Jason2866, 2019-02-26:
@zymbiosis  This behaviour cant be changed when using GPIO 0.
It is from hardware design of the ESP chip....

wongnam, 2019-02-26:
@zymbiosis I've just wondered that who/which manufacturer design such wiring?

ascillato2, 2019-02-26:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-26 by ascillato2

original-spaceman, 2022-03-15:
Hi,
the topic closed but I will add a comment anyways as it might help someone.
I've been facing the same problem using a relay board that came together with an ESP-01s. It is hardwired to use GPIO-0. So I thought let's cut the trace on the pc board and connect it to GPIO-2, which is right next to it.
Unfortunately this did not work either as GPIO-2 also goes high for a split moment turing power on.
SOLUTION: Use GPIO-3 instead (GPIO-1 would probably work as well - in case you need to drive two releays -  but I have not tested it). Typically it is not required to use the serial interface (Rx and Tx) on a Tasmota that just drives one or two releays and therefore one can use these pins as alternative GPIOs that do not play up during boot. Configuring Tasmota via Wifi is not compromized when using GPIO-3 (and/or GPIO-1) as outputs.
Cheers, Bernd

mdtdamien, 2022-10-22:
my trick is to use dupont connectors between the ESP01 board and the relay board.
have configured Pin GPIO3 in Tasmota, which I redirect to GPIO0 on the relay board