arendst/Tasmota #5346: Use display - IP address display always 0.0.0.0 after restart.

wongnam, 2019-02-26:
**Describe the bug**
Use display LCD2004, LCD1608, OLED SSD1306 - After restart the module -> IP address was shown 0.0.0.0 on display even SSID was connected and RSSI shown 100%, it's seem the module was connected to WIFI successful. 

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_ Display
- [x] _Hardware used_ :  WEMOS D1 Mini, LCD1608
- [x] _Development/Compiler/Upload tools used_ : VSC + PlatfromIO IDE
- [x] _If a pre-compiled release or development binary was used, which one?_ : Self-compiled
- [x] _You have tried latest release or development binaries?_ : Yes, 6.4.1.18 Core 2.4.2
- [x] _Provide the output of command_``status 0`` :
```
00:00:00 Project sonoff Energy Monitor Version 6.4.1.18(sonoff)-2_4_2
00:00:00 SNS: Hardware Serial
00:00:00 WIF: Connecting to AP2 Wong in mode 11N as p-0307...
00:00:07 WIF: Connected
00:00:07 HTP: Web server active on p-0307 with IP address 192.168.137.57
10:43:26 MQT: Attempting connection...
10:43:31 MQT: Connect failed to 192.168.12.155:1883, rc -2. Retry in 10 sec
10:43:31 RSL: tele/p/STATE = {"Time":"2019-02-26T10:43:31","Uptime":"0T00:00:14","Vcc":2.965,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":519,"POWER":"ON","Wifi":{"AP":2,"SSId":"Wong","BSSId":"22:B9:A5:1C:B0:E4","Channel":11,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:07"}}
10:43:31 RSL: tele/p/SENSOR = {"Time":"2019-02-26T10:43:31","ENERGY":{"TotalStartTime":"2019-02-26T10:14:20","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":222,"Current":0.000}}
10:43:41 CMD: status 0
10:43:41 RSL: stat/p/STATUS = {"Status":{"Module":0,"FriendlyName":["Energy Monitor"],"Topic":"p","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:43:41 RSL: stat/p/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:24","StartupUTC":"2019-02-26T03:43:17","Sleep":50,"CfgHolder":4617,"BootCount":13,"SaveCount":30,"SaveAddress":"F6000"}}
10:43:41 RSL: stat/p/STATUS2 = {"StatusFWR":{"Version":"6.4.1.18(sonoff)","BuildDateTime":"2019-02-25T00:47:21","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
10:43:41 RSL: stat/p/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"192.168.12.155","LogPort":514,"SSId":["iot-2","Wong"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
10:43:41 RSL: stat/p/STATUS4 = {"StatusMEM":{"ProgramSize":497,"Free":504,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0F282794","240003A0","000000CE","000013C0"]}}
10:43:41 RSL: stat/p/STATUS5 = {"StatusNET":{"Hostname":"p-0307","IPAddress":"192.168.137.57","Gateway":"192.168.137.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.137.1","Mac":"DC:4F:22:58:41:33","Webserver":2,"WifiConfig":4}}
10:43:41 RSL: stat/p/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.155","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_584133","MqttUser":"admin","MqttType":1,"MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:43:41 RSL: stat/p/STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 26 03:43:41 2019","Local":"Tue Feb 26 10:43:41 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+07:00","Sunrise":"06:09","Sunset":"18:02"}}
10:43:41 RSL: stat/p/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
10:43:41 RSL: stat/p/STATUS10 = {"StatusSNS":{"Time":"2019-02-26T10:43:41","ENERGY":{"TotalStartTime":"2019-02-26T10:14:20","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":222,"Current":0.000}}}
10:43:41 RSL: stat/p/STATUS11 = {"StatusSTS":{"Time":"2019-02-26T10:43:41","Uptime":"0T00:00:24","Vcc":2.965,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":2,"SSId":"Wong","BSSId":"22:B9:A5:1C:B0:E4","Channel":11,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:07"}}}
10:43:41 RSL: tele/p/STATE = {"Time":"2019-02-26T10:43:41","Uptime":"0T00:00:24","Vcc":2.965,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"POWER":"ON","Wifi":{"AP":2,"SSId":"Wong","BSSId":"22:B9:A5:1C:B0:E4","Channel":11,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:07"}}
10:43:41 RSL: tele/p/SENSOR = {"Time":"2019-02-26T10:43:41","ENERGY":{"TotalStartTime":"2019-02-26T10:14:20","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":222,"Current":0.000}}
10:43:41 MQT: Attempting connection...
```

**To Reproduce**
- Restart the Wemos, wait info shown on display, you can see IP is 0.0.0.0

**Expected behavior**
It should be shown IP address at this time.

**Screenshots**
[image]

andrethomas, 2019-02-26:
I'm guessing the display is not updating because you're using `DisplayMode 2`

I2C based 16x01 should be `DisplayMode 1` if i recall correctly.

wongnam, 2019-02-26:
It's not related to DisplayMode 2. I try difference(1-5) mode is still the same
The info only appear only 1 time at booting.

arendst, 2019-02-26:
At the time it shows IP address the wifi is not yet connected hence no IP address. It's a feature!

wongnam, 2019-02-26:
> At the time it shows IP address the wifi is not yet connected hence no IP address. It's a feature!

I have ever see it show IP in boot process which is useful if it can appear at startup. so I just came up to see if there is any way to improve.

wongnam, 2019-02-26:
I have a screen shot to compare them. FYR.
[image]

arendst, 2019-02-26:
The startup info is  fed in a buffer at startup. The buffer is shown a line at a time. At the time you see ip 0.0.0.0 it's at least 5 seconds past the time the buffer was fed.

wongnam, 2019-02-26:
I see, I got your point now. Thanks @arendst .

closed 2019-02-26 by wongnam