arendst/Tasmota #5348: NODEMCU LOLIN ESP-12E reseting when configure buttons

hwarang098tkd, 2019-02-26:
I have a NODEMCU LOLIN ESP-12E and I have set on D2 and D7 2 relays [https://www.google.com/search?biw=1920&bih=888&tbm=isch&sa=1&ei=69t0XJrMLcyalwS2_YCACQ&q=slv-05++relay+arduino&oq=slv-05++relay+arduino&gs_l=img.3...4252.5496..5737...0.0..0.131.938.0j8......1....1..gws-wiz-img.VGaDdY8rv6A#imgrc=MRTtmck9cdllaM:](url).
The problem is that when I set buttons to GPIOs in " Module parameters " after a while it reset to default configuration
PINS
VIN-->DC+ relay1
VIN-->DC+ relay2
GND-->DC- relay1
GND-->DC-relay2
D2-->IN relay1(with 220R)
D7-->IN relay2(with 220R)

It works also for a while if i connect the button. I use this example [https://www.arduino.cc/en/tutorial/button](url) but I connect the 
3.3vpin not 5v. and the resistors are 330R. 
If i dont apply buttons (in software and hardware) it runs perfectly.
What is wrong?
thanx in advance

netpok, 2019-02-26:
Can you submit a schematic for your design?

hwarang098tkd, 2019-02-26:
In design i  have 4 buttons . but in configuration fron now i set only 2. In future i want to have 2 more virtual buttons for different use( All relays OFF, all ON, set with rules)

[image]

Status 0. 
`09:50:13 MQT: stat/bathroom_heat_relay/STATUS = {"Status":{"Module":18,"FriendlyName":["bathroom_heat_relay","2"],"Topic":"bathroom_heat_relay","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:50:13 MQT: stat/bathroom_heat_relay/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"""","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:18","StartupUTC":"2019-02-26T08:48:55","Sleep":50,"BootCount":6,"SaveCount":24,"SaveAddress":"F8000"}}
09:50:13 MQT: stat/bathroom_heat_relay/STATUS2 = {"StatusFWR":{"Version":"6.4.1.14(sonoff)","BuildDateTime":"2019-02-26T08:14:01","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
09:50:13 MQT: stat/bathroom_heat_relay/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["David",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000020"]}}
09:50:13 MQT: stat/bathroom_heat_relay/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":3,"Features":["00000809","0FDAE794","240383A0","23B617CC","00003BC0"]}}
09:50:13 MQT: stat/bathroom_heat_relay/STATUS5 = {"StatusNET":{"Hostname":"bathroom_heat_relay","IPAddress":"192.168.1.16","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"2C:3A:E8:06:A0:E0","Webserver":2,"WifiConfig":4}}
09:50:13 MQT: stat/bathroom_heat_relay/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.43","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_06A0E0","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:50:13 MQT: stat/bathroom_heat_relay/STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 26 08:50:13 2019","Local":"Tue Feb 26 09:50:13 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:38","Sunset":"18:28"}}
09:50:13 MQT: stat/bathroom_heat_relay/STATUS10 = {"StatusSNS":{"Time":"2019-02-26T09:50:13","ANALOG":{"A0":1}}}
09:50:13 MQT: stat/bathroom_heat_relay/STATUS11 = {"StatusSTS":{"Time":"2019-02-26T09:50:13","Uptime":"0T00:01:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"David","BSSId":"A4:B1:E9:9C:17:B0","Channel":6,"RSSI":92}}}`

I erased chip with python and flash it again.

netpok, 2019-02-26:
Let's assume it's a mistake in your schematic, but the first 3 button is miswired (the input is connected to the 3v3 pin instead of the pullled down one), only the forth one is correct. The 330R resistors are a little overkill, you should be fine with a 4k7 one, but it should also work with just the switch between GND and the pin (using the internal pullup resistor).

That aside if the schematic is mistaken and not your device, then it should work. Please set debug level to maximum before enabling the buttons, then copy the log from the console to here. Also use code blocks (` ``` ` newline content newline ` ``` `) instead of quotes for better readability

hwarang098tkd, 2019-02-26:
schematic is wrong. So i have to change resistors to see what happen...

netpok, 2019-02-26:
No the resistors shouldn't matter in this case, it's just a general problem which can lead to overload of power supply when you press multiple buttons at the same time.

The logs are needed for further information.

hwarang098tkd, 2019-02-26:
3 hours now working fine without reset. I didn;t change anything.
NOW I want the 3rd button to 
(`` ``` `` 14:05:15 MQT: stat/bathroom_heat_relay/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":438,"Rules":"on button3#state=2 do power2 1 endon on button3#state=2 do power1 1 endon"}
14:06:11 MQT: stat/bathroom_heat_relay/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":438,"Rules":"on button3#state=3 do power2 2 endon on button3#state=3 do power1 2 endon"}

 `` ``` ``)

but on single press execute rule2 . It cant distinguish the single from double press.
{"SetOption13":"OFF"}
 {"SetOption32":"20"}
{"SetOption11":"OFF"}
 {"SetOption1":"ON"}
{"SwitchMode3":0}
 {"ButtonDebounce":50}

ascillato2, 2019-02-26:
Hi,

Sorry `on button3#state=3` means holding button3. 

Please, check the wiki at rules. There are examples on how using rules. If you need further assistant with rules, please, just ask here on in the Tasmota Support Chat. 

Closing issue as the original thread is solved. 

Thanks.

closed 2019-02-26 by ascillato2