arendst/Tasmota #5350: Problem after energy reset

quorle, 2019-02-26:
I have a problem with my Sonoff POW R2: I've done an EnergyReset1, EnergyReset2, EnergyReset3 on all my sonoffs. However, there are problems now. I will now be transferring 3 new topics via MQTT (EnergyReset1, EnergyReset2, EnergyReset3), if I delete them, they will re-appear at the next update. In addition, the consumption is no longer counted properly. That is, total and day are the same. Is there a guide here to undo this again, so that the energy consumption is properly recorded again?

[image]

Jason2866, 2019-02-26:
Issue template missing. Please post `status 0` Output

quorle, 2019-02-26:
```
18:52:59 CMD: status 0
18:52:59 MQT: stat/sonoff/STATUS = {"Status":{"Module":43,"FriendlyName":["Sonoff_POW_Waschmaschine"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":5,"SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:52:59 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T18:38:39","StartupUTC":"","Sleep":50,"BootCount":15,"SaveCount":38,"SaveAddress":"FA000"}}
18:52:59 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018.12.24 14:55:04","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
18:52:59 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Klitsch-Klatsch",""],"TelePeriod":10,"SetOption":["00008009","558180C0","00000001"]}}
18:52:59 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":522,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000407","0FDAE794","000383A0","23B617CE","00003BC0"]}}
18:52:59 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3537","IPAddress":"192.168.2.140","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"84:F3:EB:4A:CD:D1","Webserver":2,"WifiConfig":4}}
18:52:59 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.102","MqttPort":1883,"MqttClientMask":"Sonoff_POW_Waschmaschine","MqttClient":"Sonoff_POW_Waschmaschine","MqttUser":"Sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:52:59 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 18:52:59 1970","Local":"Thu Jan 01 18:52:59 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
18:52:59 MQT: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:52:59 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T18:52:59","ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":19.537,"Yesterday":0.000,"Today":19.537,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
18:52:59 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T18:52:59","Uptime":"0T18:38:39","Vcc":3.442,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Klitsch-Klatsch","BSSId":"00:24:FE:05:50:8A","Channel":5,"RSSI":72}}}
18:53:02 MQT: tele/sonoff/STATE = {"Time":"1970-01-01T18:53:02","Uptime":"0T18:38:43","Vcc":3.442,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Klitsch-Klatsch","BSSId":"00:24:FE:05:50:8A","Channel":5,"RSSI":72}}
18:53:02 MQT: tele/sonoff/SENSOR = {"Time":"1970-01-01T18:53:02","ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":19.537,"Yesterday":0.000,"Today":19.537,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
```

Jason2866, 2019-02-26:
Your device hasnt correct time. This causes issues. If you manage to get correct time device will work as expected

quorle, 2019-02-26:
I have now corrected the time. Now I have to wait and see if something changes. Could you possibly change the time server via MQTT command, or change it in the WebUI? Since I do not like it, when Tasmota nac outside calls. Why should energy counting be a problem if time is not right?

quorle, 2019-02-26:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._
**I have a problem with my Sonoff POW R2: I've done an EnergyReset1, EnergyReset2, EnergyReset3 on all my sonoffs. However, there are problems now. I will now be transferring 3 new topics via MQTT (EnergyReset1, EnergyReset2, EnergyReset3), if I delete them, they will re-appear at the next update. In addition, the consumption is no longer counted properly. That is, total and day are the same. Is there a guide here to undo this again, so that the energy consumption is properly recorded again?**

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : _**Sonoff POW R2**_
- [X] _Development/Compiler/Upload tools used_ : **_ATOM with PlatformIO_**
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [X] _You have tried latest release or development binaries?_ : **_6.4.1_**
- [ ] _Provide the output of command_``status 0`` :
```
21:56:51 CMD: status 0
21:56:51 MQT: stat/sonoff/STATUS = {"Status":{"Module":43,"FriendlyName":["Sonoff_POW_Waschmaschine"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":5,"SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:56:51 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T01:42:33","StartupUTC":"2019-02-26T19:14:18","Sleep":50,"BootCount":16,"SaveCount":39,"SaveAddress":"F9000"}}
21:56:51 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018.12.24 14:55:04","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
21:56:51 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Klitsch-Klatsch",""],"TelePeriod":10,"SetOption":["00008009","558180C0","00000001"]}}
21:56:51 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":522,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000407","0FDAE794","000383A0","23B617CE","00003BC0"]}}
21:56:51 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3537","IPAddress":"192.168.2.140","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"84:F3:EB:4A:CD:D1","Webserver":2,"WifiConfig":4}}
21:56:51 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.102","MqttPort":1883,"MqttClientMask":"Sonoff_POW_Waschmaschine","MqttClient":"Sonoff_POW_Waschmaschine","MqttUser":"Sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:56:51 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 26 20:56:51 2019","Local":"Tue Feb 26 21:56:51 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:38","Sunset":"18:28"}}
21:56:51 MQT: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:56:51 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-02-26T21:56:51","ENERGY":{"TotalStartTime":"2019-02-26T20:14:09","Total":19.537,"Yesterday":0.000,"Today":19.537,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
21:56:51 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-02-26T21:56:51","Uptime":"0T01:42:33","Vcc":3.441,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Klitsch-Klatsch","BSSId":"00:24:FE:05:50:8A","Channel":5,"RSSI":66}}}
21:56:53 MQT: tele/sonoff/STATE = {"Time":"2019-02-26T21:56:53","Uptime":"0T01:42:35","Vcc":3.442,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Klitsch-Klatsch","BSSId":"00:24:FE:05:50:8A","Channel":5,"RSSI":66}}
21:56:53 MQT: tele/sonoff/SENSOR = {"Time":"2019-02-26T21:56:53","ENERGY":{"TotalStartTime":"2019-02-26T20:14:09","Total":19.537,"Yesterday":0.000,"Today":19.537,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
21:57:03 MQT: tele/sonoff/STATE = {"Time":"2019-02-26T21:57:03","Uptime":"0T01:42:45","Vcc":3.442,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Klitsch-Klatsch","BSSId":"00:24:FE:05:50:8A","Channel":5,"RSSI":66}}
21:57:03 MQT: tele/sonoff/SENSOR = {"Time":"2019-02-26T21:57:03","ENERGY":{"TotalStartTime":"2019-02-26T20:14:09","Total":19.537,"Yesterday":0.000,"Today":19.537,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._
[image]

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-02-27:
Are you running the commands like:

`Energyreset1 0`

Can you post the output of the console when you do that?

ascillato2, 2019-02-27:
Also, please post the output of the console when your restart your device. Remember to use weblog 4 for more debug information.

wongnam, 2019-03-01:
@quorle Perhaps your mqtt message is retained in MQTT broker, try to clear mqtt retained message for ```EnergyReset1```, ```EnergyReset1```, ```EnergyReset1```.

ascillato2, 2019-03-06:
Closing this issue as there is no feedback.

Please, when you have time post the output of the console when your restart your device. Remember to use weblog 4 for more debug information.

Thanks

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-06 by ascillato2