arendst/Tasmota #5351: Unable to connect to webserver

DaveGalbraith, 2019-02-26:
HI,

I flashed a 3 sonoff basics using ESPEASY and sonoff.bin (v6.4.1). Each device is able to connect to my wifi and gets an IP address. However, I can't access the web server. I can see the device in my router and can ping from there but can't ping from anywhere else. I have read the various other issues that have been raised but none seem applicable to me (e.g. AP isolation is off, connecting to network via ethernet does not help).

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Development/Compiler/Upload tools used :
- [ x] Hardware used : 
- [ x] If a pre-compiled release or development binary was used, which one? :
sonoff.bin v6.4.1
- [ x] You have tried latest release or development binaries? :
- [ x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```
09:50:40 CMD: status 0
09:50:40 RSL: sonoff/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:50:40 RSL: sonoff/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:26","StartupUTC":"2019-02-26T08:50:14","Sleep":50,"BootCount":25,"SaveCount":73,"SaveAddress":"FB000"}}
09:50:40 RSL: sonoff/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":7,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
09:50:40 RSL: sonoff/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["my-ssid","my-ssid2"],"TelePeriod":300,"SetOption":["000A8009","558180C0","00000200"]}}
09:50:40 RSL: sonoff/stat/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
09:50:40 RSL: sonoff/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3427","IPAddress":"192.168.1.21","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":2}}
09:50:40 RSL: sonoff/stat/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_922D63","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:50:40 RSL: sonoff/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 26 08:50:40 2019","Local":"Tue Feb 26 09:50:40 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:38","Sunset":"18:28"}}
09:50:40 RSL: sonoff/stat/STATUS10 = {"StatusSNS":{"Time":"2019-02-26T09:50:40"}}
09:50:40 RSL: sonoff/stat/STATUS11 = {"StatusSTS":{"Time":"2019-02-26T09:50:40","Uptime":"0T00:00:26","Vcc":3.307,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"my-ssid","BSSId":"xx:xx:xx:xx:xx:xx","Channel":11,"RSSI":100}}}

Any help or pointers are much appreciated.

Jason2866, 2019-02-26:
Power of all sonoff devices. Power off router and all Accesspoints. Wait 5 seconds. Power on router and AP
Now power on sonoff devices.

DaveGalbraith, 2019-02-26:
Brilliant, that seemed to do the trick ta.

closed 2019-02-26 by DaveGalbraith