arendst/Tasmota #5352: Control sonoff url Comand outside network

georgia088, 2019-02-26:
This may be a stupid question, but I don’t know the answer. I have basic Sonoff flashed with tasmota (awesome btw!). I use them to control lights in my house. I understand how to toggle these switches from any browser using the locally assigned ip for each device. (Ex. http://192.168.1.13/cm?cmnd=Power%20On). However, I can only do this if my browser is connected to my local network. What is the easiest way to be able to toggle this same device from outside my network? 
Thanks!

netpok, 2019-02-26:
Easiest way probably is to set up a home automation software (like OpenHAB) and use that. Without that your options are quite limited if you want even a minimal security over your things.

georgia088, 2019-02-26:
Thanks for the quick response! Doesn’t that require hardware (raspberry pi etc.) to be on on the home network at all times? I don’t have that hardware. Is there no way to do it without this?

blakadder, 2019-02-26:
Tasmota isn't designed for external access due to weak security in web ui. If you expose the device to external access there might be someone else playing with your switch or doing even worse things.

netpok, 2019-02-26:
The only way that might be possible is via mqtt over tls, but that version of the firmware cannot accomodate the web interface. And you need a secure mqtt hosting.

georgia088, 2019-02-26:
It wouldn’t be possible to setup port forwarding or a dns etc.?

Also, out of curiousit what would be possible for someone to do besides messing with switches; if they could access it due to lack of security?

Thanks for enlightening me!

jziolkowski, 2019-02-26:
they can access your wifi password, for starters

netpok, 2019-02-26:
And if they can access the device it can be reflashed to act as gateway to your home network (this is an extreme example). In theory setting the webinterface to user mode would protect against this, but the firmware was never tested against this and there could be bugs allowing messing with everything. Long story short it is not advised.

meingraham, 2019-02-26:
- openHAB/Home Assistant, ... free
- Raspberry Pi ZeroW ($5-$10) is sufficient... but probably want to go with 3B+ ($35)
- Your time... free :wink:

Worth the investment. You'll end up with a more secure configuration and it will open up a ton of automation options.

Mike

georgia088, 2019-02-26:
Ok. So, maybe I am going about this wrong. Maybe y’all can point me in the right direction. 
I am using tasmota simply to be able to use an existing switch (switching rx to ground with existing wall switch). I disable Mqtt. And control the sonoff via echo (Alexa). I understand tasmota isnt really designed to work without using mqtt, but i don’t have a raspberry pi or similar to make use of the other tasmota benefits. 

So, my question is there something else I could flash the sonoff with to enable using existing switch (rx  to ground) that also has app support or url support that I could use ifttt with?

Thanks again!

netpok, 2019-02-26:
If you are connected it to alexa maybe you are able to control it from the alexa app?

georgia088, 2019-02-26:
Yes, I can. But the Alexa app has limited capabilities. No widgets for phone. And you can’t use Alexa app to perform the action in ifttt (at least not that I can figure out). I may be a bit of a control freak, but I would like to be able to utilize the shortcut app to enable Siri to control the sonoff devices and utilize the shortcut widget. I may be asking for too much, but though it wouldn’t hurt to ask. Keep the ideas coming!
Thanks!

DavinKD, 2019-02-26:
Sounds like you would have been better off leaving it stock and using the eWeLink app.

georgia088, 2019-02-26:
That won’t allow you to use existing switch to switch from ground to rx and toggle the switch....

netpok, 2019-02-26:
What kind of router do you have, if it can run openwrt, then you can set up an https protected website on it with a proper webserver (like nginx or lighttpd and forward the remote calls from there).

Jason2866, 2019-02-26:
If you want a flexible smart solution https://github.com/arendst/Sonoff-Tasmota/issues/5352#issuecomment-467456936

andrethomas, 2019-02-26:
I would not recommend exposing the http interface of any IOT device to a public IP. The only safer way to solve this would be following the recommendations above...

netpok, 2019-02-26:
I mean't he can implement an interface for himself, not proxying the original one.

andrethomas, 2019-02-26:
@netpok I understood... your solution is viable and included in the "recommendations above" :)

ascillato2, 2019-02-26:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-26 by ascillato2

joba-1, 2019-02-26:
@georgia088 : the recommendations above are all good. 

But if you insist and can compile your own firmware (with Arduino IDE), then this might be something you like:

It is a quick mix of webserver and web online updater example with added switch capability. Just edit the configuration section to your needs. 

- http://{yourhostname}/on switches on in your wlan
- http://{yourhostname}/off switches off
- http://{yourhostname}/update allows ota update in case you want to go back to tasmota (don't use /update over the internet!)
- any other URL shows the current switch status

Now you only have to define a forwarding rule from a router port (e.g. 1234) to {yourhostname}:80
and you (and anyone else) can switch your relay over the internet using http://{yourexternaldnsname}:1234/on etc. ( let me know your dns name to test :) ).

Updates are only possible with the update password. So choose a good one and don't use it over the internet, only in your local safe wlan, since it will be sent in clear text.

Disclaimer: not tested, just compiled.

```
/*
  OTA update: 
  To upload firmware file firmware.bin through terminal you can use: 
    curl -F "image=@firmware.bin" {yourhostname}.local/update
  DO NOT USE UPDATE PASSWORD OVER INTERNET (it can be sniffed and your firmware replaced by hostile stuff)
  Author: Joachim Banzhaf, License: GPL V2
*/

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPUpdateServer.h>

#define SWITCH_PIN D2                   // adjust to gpio pin that switches your relay

const char* host = "joba1-switch";      // adjust hostname to your wishes
const char* ssid = "........";          // replace dots with your wlan ssid
const char* password = "........";      // replace dots with your wlan password
const char* update_user = "updater";    // adjust name to your wishes
const char* update_pass = "........";   // replace dots with your update password (don't use over internet)
const char* update_path = "/update";

ESP8266WebServer httpServer(80);
ESP8266HTTPUpdateServer httpUpdater;

void setup(void){

  Serial.begin(115200);
  Serial.println();
  Serial.println("Booting Sketch...");
  pinMode(SWITCH_PIN, OUTPUT|INPUT);
  
  WiFi.mode(WIFI_AP_STA);
  WiFi.begin(ssid, password);

  while(WiFi.waitForConnectResult() != WL_CONNECTED){
    WiFi.begin(ssid, password);
    Serial.println("WiFi failed, retrying.");
  }

  MDNS.begin(host);

  httpUpdater.setup(&httpServer, update_path, update_user, update_pass);
  
  httpServer.on("/on", []() { 
    digitalWrite(SWITCH_PIN, HIGH); 
    httpServer.send (200, "text/plain", "->ON"); });

  httpServer.on("/off", []() {
    digitalWrite(SWITCH_PIN, LOW); 
    httpServer.send(200, "text/plain", "->OFF"); });
  
  httpServer.onNotFound( []() {
    if( digitalRead(SWITCH_PIN) == HIGH )
      httpServer.send(200, "text/plain", "ON");
    else
      httpServer.send(200, "text/plain", "OFF"); });
  
  httpServer.begin();

  MDNS.addService("http", "tcp", 80);
  Serial.printf("HTTPUpdateServer ready! Open http://%s.local/update in your browser\n", host);
}

void loop(void){
  httpServer.handleClient();
}

```

joba-1, 2019-02-26:
> Keep the ideas coming!

Ok, another idea - feeling a bit stupid not having thought of this before, because I use it all the time: 

My router (fritzbox) has a VPN feature. If I activate it on my mobile I can access all ip devices as if I were at home in my WLAN. Maybe your router can do this, too?

georgia088, 2019-02-27:
I like this idea about the vpn! Not sure if my router supports it. I will have to check. To be honest I think I’m too stupid to do the first suggestion. I don’t know anything about creating firmware.... I don’t know much about a vpn but I’m willing to look in to it! I will look at my routers settings. It’s a tplink archer c7..... if that helps.

georgia088, 2019-02-27:
> > Keep the ideas coming!
> 
> Ok, another idea - feeling a bit stupid not having thought of this before, because I use it all the time:
> 
> My router (fritzbox) has a VPN feature. If I activate it on my mobile I can access all ip devices as if I were at home in my WLAN. Maybe your router can do this, too?

Ok, I have been looking into it, but I'm not sure that I understand how I can utilize it.  If I run the VPN feature through my router, will I be able to create a link (maybe wrong terminology) that will allow me to access/utilize my sonoffs local ip adresses from a outside my private network? Also, which VPN service would you suggest I use?

andrethomas, 2019-02-27:
If you have a router which supports openvpn/l2tp/pptp you can use it directly without using a vpn provider... preferably in that order from a security perspective. If not you can configure a raspberry pi on your lan to serve an openvpn endpoint and route an ip internal to your network to a device connecting to the pi.

georgia088, 2019-02-27:
Ok, my router (Linksys archer c7) has the l2tp option under the wan setting, but I don't know how to configure it.  How do I know if it supports it?

andrethomas, 2019-02-27:
google :)

joba-1, 2019-02-27:
what i use is ipsec. It is the most secure and the fastest approach. No
need for external providers or additional hardware in your network. For how
to set this up on your router, only your router manual will know.
But just to give an idea: i create a new user on the fritzbox and give that
user vpn authorisation. Then the router displays settings to be entered on
the phone, and that‘s it. Not too complicated, I think.

-- 
iphone so i typo

netpok, 2019-02-27:
Just for information: l2tp and ipsec is the same (well technically no but they are generally used on the same thing in this context, l2tp provides the transport protocol and ipsec provides the encryption).

Also just some nitpicking while Ipsec is pretty secure, openvpn is more secure, so its not the most secure (and we probably never heard of whatever counts as the most secure in the next ten minutes, before something succeeds it).

lalo-uy, 2019-02-27:
That is usually for vpn forwarding, not the same beast.
You want your router to act as a VPN server.
Many Linux based router has this option.

glowinthedark, 2019-06-26:
The simplest possible way is to set up an SSH tunnel to a jump server (e.g. VPS) and then forward the ports so that you can basically access your home LAN securely. Once configured you'll be able to for example to control the switch with curl via ssh with `curl` HTTP requests. That's a similar setup to what the original ewelink mobile app does, except that you will have a much more secure setup since you connect over a secure tunnel.